meshtastic/firmware #2874: [Bug]: Fixed Position doesn't survive a node reboot.

kodiakastu, 2023-10-08:
### Category

Other

### Hardware

T-Lora v2 1.6

### Firmware Version

2.2.10

### Description

Hi. I've seen #2717 and meshtastic/Meshtastic-Android#681. I still cannot figure out to make the fixed position survive a reboot. I'm using the Android app (V2.2.8) to set it up.. 

* Smart position is off
* Use fixed position is on (the 3 fields after are always zeroed after a node reboot.)
* GPS is disabled because there is none onboard.

I had the same issue on 2.2.9, tried to update but I still have the same issue. Let me know if I can provide more details/information.

I don't know if it's the right place to mention it, but there is no fields in the web client to specify coordinates for the fixed position.

pdxlocations, 2023-10-08:
https://github.com/meshtastic/firmware/assets/117498748/aa71a22d-a849-4e2f-91ea-b7b83552b7c4

FW 2.2.10 TLora V2116

pdxlocations, 2023-10-08:
> I don't know if it's the right place to mention it, but there is no fields in the web client to specify coordinates for the fixed position.

https://github.com/meshtastic/web/issues/87

kodiakastu, 2023-10-08:
https://github.com/meshtastic/firmware/assets/15963021/e84cbf0e-6a78-4418-a183-d6f69ca0b0cb

I'm doing the same thing :/

thebentern, 2023-10-08:
> 
> 
> https://github.com/meshtastic/firmware/assets/15963021/e84cbf0e-6a78-4418-a183-d6f69ca0b0cb
> 
> 
> 
> I'm doing the same thing :/
> 
> 
> 
> 

It's possible you have file system corruption issues going on. See if you can grab the serial log with meshtastic --noproto when you save from the app.

kodiakastu, 2023-10-08:
I grabbed the logs from flasher.meshtastic.org. If it's really necessary, I'll install python.

I tried to grab a little before/after to make sure everything is in there.

```
DEBUG | 15:17:41 141 [Power] Battery: usbPower=1, isCharging=1, batMv=4352, batPct=0
DEBUG | 15:17:41 141 [Screen] Screen: Powered...
DEBUG | 15:17:42 142 [ServerAPI] PACKET FROM PHONE (id=0x5db8182b fr=0x00 to=0x60, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 priority=10)
DEBUG | 15:17:42 142 [ServerAPI] Enqueued local (id=0x5db8182b fr=0x00 to=0x60, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 rxtime=1696778262 priority=10)
INFO  | 15:17:42 142 [ServerAPI] getFromRadio=STATE_SEND_PACKETS
DEBUG | 15:17:42 142 [ServerAPI] encoding toPhone packet to phone variant=11, 12 bytes
INFO  | 15:17:42 142 Telling client we have new packets 10
DEBUG | 15:17:42 142 [Router] Add packet record (id=0x5db8182b fr=0x00 to=0x60, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 rxtime=1696778262 priority=10)
DEBUG | 15:17:42 142 [Router] handleReceived(REMOTE) (id=0x5db8182b fr=0x00 to=0x60, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 rxtime=1696778262 priority=10)
DEBUG | 15:17:42 142 [Router] Module 'position' wantsPacket=1
INFO  | 15:17:42 142 [Router] Received position from=0x0, id=0x5db8182b, portnum=3, payloadlen=15
DEBUG | 15:17:42 142 [Router] Incoming update from MYSELF
INFO  | 15:17:42 142 [Router] POSITION node=08ac8b60 l=15 latI=457945600 lonI=-737270667 msl=0 hae=0 geo=0 pdop=0 hdop=0 vdop=0 siv=0 fxq=0 fxt=0 pts=0 time=1696778263
INFO  | 15:17:42 142 [Router] updatePosition REMOTE node=0x8ac8b60 time=1696778263, latI=457945600, lonI=-737270667
DEBUG | 15:17:42 142 [Router] Node status update: 2 online, 2 total
DEBUG | 15:17:42 142 [Router] Module 'position' considered
DEBUG | 15:17:42 142 [Router] Module 'routing' wantsPacket=1
INFO  | 15:17:42 142 [Router] Received routing from=0x0, id=0x5db8182b, portnum=3, payloadlen=15
DEBUG | 15:17:42 142 [Router] Routing sniffing (id=0x5db8182b fr=0x00 to=0x60, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 rxtime=1696778262 priority=10)
DEBUG | 15:17:42 142 [Router] Module 'routing' considered
INFO  | 15:17:46 146 [PositionModule] Sending pos@0:6 to mesh (wantReplies=1)
INFO  | 15:17:46 146 [PositionModule] Providing time to mesh 1696778266
INFO  | 15:17:46 146 [PositionModule] Position reply: time=1696778266, latI=457945600, lonI=--737270667
DEBUG | 15:17:46 146 [PositionModule] Update DB node 0x8ac8b60, rx_time=1696778266, channel=0
DEBUG | 15:17:46 146 [PositionModule] handleReceived(LOCAL) (id=0x7c9b05c0 fr=0x60 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 WANTRESP rxtime=1696778266 priority=10)
DEBUG | 15:17:46 146 [PositionModule] No modules interested in portnum=3, src=LOCAL
DEBUG | 15:17:46 146 [PositionModule] localSend to channel 0
DEBUG | 15:17:46 146 [PositionModule] Add packet record (id=0x7c9b05c0 fr=0x60 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 WANTRESP rxtime=1696778266 priority=10)
DEBUG | 15:17:46 146 [PositionModule] Using AES256 key!
DEBUG | 15:17:46 146 [PositionModule] ESP32 crypt fr=8ac8b60, num=7c9b05c0, numBytes=21!
DEBUG | 15:17:46 146 [PositionModule] enqueuing for send (id=0x7c9b05c0 fr=0x60 to=0xff, WantAck=0, HopLim=3 Ch=0xdd encrypted rxtime=1696778266 priority=10)
DEBUG | 15:17:46 146 [PositionModule] txGood=2,rxGood=3,rxBad=0
DEBUG | 15:17:46 146 [PositionModule] Using channel 0 (hash 0xdd)
DEBUG | 15:17:46 146 [PositionModule] Using AES256 key!
DEBUG | 15:17:46 146 [PositionModule] ESP32 crypt fr=8ac8b60, num=7c9b05c0, numBytes=21!
DEBUG | 15:17:46 146 [PositionModule] decoded message (id=0x7c9b05c0 fr=0x60 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 WANTRESP rxtime=1696778266 priority=10)
INFO  | 15:17:46 146 Telling client we have new packets 12
DEBUG | 15:17:46 146 [RadioIf] Starting low level send (id=0x7c9b05c0 fr=0x60 to=0xff, WantAck=0, HopLim=3 Ch=0xdd encrypted rxtime=1696778266 priority=10)
DEBUG | 15:17:46 146 [RadioIf] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=37, time 673 ms
DEBUG | 15:17:46 146 [RadioIf] AirTime - Packet transmitted : 673ms
INFO  | 15:17:46 146 [ServerAPI] getFromRadio=STATE_SEND_PACKETS
DEBUG | 15:17:46 146 [ServerAPI] encoding toPhone packet to phone variant=11, 12 bytes
INFO  | 15:17:46 146 [ServerAPI] getFromRadio=STATE_SEND_PACKETS
DEBUG | 15:17:46 146 [ServerAPI] phone downloaded packet (id=0x7c9b05c0 fr=0x60 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 WANTRESP rxtime=1696778266 priority=10)
DEBUG | 15:17:46 146 [ServerAPI] encoding toPhone packet to phone variant=2, 49 bytes
DEBUG | 15:17:46 146 [Screen] LastScreenTransition exceeded 10890ms transitioning to next frame
```

andrekir, 2023-10-08:
could be related #2826

kodiakastu, 2023-10-08:
> could be related #2826

The position seems okay seen from an other device on the mesh
[image]

thebentern, 2023-10-10:
Should be fixed in 2.1.11. Please re-open if it continues to occur.

pdxlocations, 2023-10-11:
> Just updated. It's doing the exact same thing. :/
> 
> Edit: @thebentern I cannot reopen since i'm not the one whom closed it I think.

Two things I would try:
1. clearing app data
2. setting these options from client.meshtastic.org or an apple app.

kodiakastu, 2023-10-11:
> Two things I would try:
> 
>     1. clearing app data
> 
>     2. setting these options from client.meshtastic.org or an apple app.

Both of those things didn't work. Also, there is no field to enter a position on the web app ( and I don't have an apple device). I tried a wipe and install on the device; nothing new. 

I then tried to put settings one by one from a fresh wipe. I remarked that if I just turn on the toggle for fixed position, I would receive a success message, but if I only update the position, it would go back to 0/0/0 (Will file a bug on the android repo.) 

Thank you all for your great help! I hope I'll be able one day to contribute :D

closed 2023-10-11 by kodiakastu

pdxlocations, 2023-10-11:
> > Two things I would try:
> > ```
> > 1. clearing app data
> > 
> > 2. setting these options from client.meshtastic.org or an apple app.
> > ```
> 
> Both of those things didn't work. Also, there is no field to enter a position on the web app ( and I don't have an apple device). I tried a wipe and install on the device; nothing new.
> 
> I then tried to put settings one by one from a fresh wipe. I remarked that if I just turn on the toggle for fixed position, I would receive a success message, but if I only update the position, it would go back to 0/0/0 (Will file a bug on the android repo.) It now survives a reboot.
> 
> Thank you all for your great help! I hope I'll be able one day to contribute :D

This is interesting because I have the same device and use Android too.  Have you tried setting GPS enabled and fixed position off, then setting your location by using the "provide phone location to mesh" checkbox.  Then setting fixed position to on and GPS to off (and then disable provide phone location)?

I haven't tried manually entering Lat/Lon in my app.  I've only used my phone to share location.  I'm wondering if the android app doesn't save manually entered locations.

kodiakastu, 2023-10-11:
> This is interesting because I have the same device and use Android too. Have you tried setting GPS enabled and fixed position off, then setting your location by using the "provide phone location to [...]

So: 
- Manually updating the position doesn't cause a reboot. New position doesn't survive.

- Switch Fixed off, GPS on, SEND. Rebooted by itself. Provide phone position, Fixed ON/GPS off, SEND. Rebooted by itself. Provide off, reboot. The position saved from the phone is surviving.

Seems like if you only update the position it doesn't trigger a save, only a new packet and update the value on the ram. 

I don't actually know it's done here, but I assume how values are stored/updated; the value is sent and updated in the eeprom and that's why the device reboots (to reload the new values from eeprom to ram). In the case of only updating the position, it only puts it in ram, like it would with a GPS module, and that's why the new value doesn't survive.

At that point, I don't think it's an android isssue, but it stems from how the fixed position is stored/updated. What's should I do now ?

pdxlocations, 2023-10-11:
> At that point, I don't think it's an android isssue, but it stems from how the fixed position is stored/updated. What's should I do now ?

I feel like I've encountered this bug in past.  Setting fixed position Lat/Lon alone doesn't stick (no reboot) but if you change some combination of the toggles (use fixed position, GPS enabled) the fixed position does stick (and device reboots).

Try just turning Use Fixed Position to off  and GPS on then hit send (reboots)
Then turn Use Fix Position to on, set your position and turn GPS off, then hit send.

The more I play with it, the more it looks like I have to flip the GPS enabled toggle for the device to save/reboot.

kodiakastu, 2023-10-11:
>The more I play with it, the more it looks like I have to flip the GPS enabled toggle for the device to save/reboot.

If I toggle fixed off, send (wait for reboot) toggle fixed on AND set new position it survives.
If I just toggle GPS ( either on or off) and change position, it saves. 

Seems like the change in the fixed positions doesn't trigger a save. Depending on how those coordinates are treated in the background it could make sense. You would not want to trigger a save every time your phone sends a new position to a fixed position device.

myano, 2025-03-13:
Anyone else still experiencing this? I'm having trouble saving a "fixed position" with given coordinates.

I'll save it, and it'll show correctly on the map for a few minutes after a reboot, then it changes GPS coordinates to a location a mile away. (And it looks like there are a few other nodes at that other location.)

I've tried different combinations of ideas suggested in this thread, including doing a factory reset of my Meshtastic device. The only thing that seems to "lock" the position to my current location is enabling "Provide phone location to mesh" but I want my Meshtastic device to show the correct location when I'm not near it.

I'm running 2.5.19 on my Android phone, and 2.5.20 on my RAK4631 device (it's the WisMesh Repeater Mini).

fifieldt, 2025-03-15:
@myano sounds like position precision settings

myano, 2025-04-17:
> [@myano](https://github.com/myano) sounds like position precision settings

That was it! I had to go into the precision settings on the channel in the radio settings. That fixed it. Thank you!