meshtastic/firmware #2877: GPS power enable pin not defined in latest versions for Heltec v3

Juzzle1, 2023-10-09:
### Category

Hardware Compatibility

### Hardware

Heltec V3

### Firmware Version

2.2.10 / Master

### Description

For Heltec v3, the PIN_GPS_EN define has been removed from variant.h in 
https://github.com/meshtastic/firmware/pull/2799

Without this, now my GPS modules remain powered even when I shutdown the device (I can see the LEDs are still active on on the module), causing the battery to drain.

I also noticed that this pin definition has _not_ been removed for the Heltec WSL in this same PR.

For reference my GPS modules are switched via a 2N2222 transistor with base connected to pin 46.

Juzzle1, 2023-10-09:
Great, thanks!

jp-bennett, 2023-10-09:
Looks like it's already present in the latest Python client. So 
`meshtastic --set position.gps_en_gpio 46`
should get you back.

Juzzle1, 2023-10-09:
Ah, probably need to update my client to see it. 

Thanks for looking!