meshtastic/firmware #2885: [Bug]: Device becomes unresponsive or powers off if Bluetooth is disablled

cordelster, 2023-10-12:
### Category

BLE

### Hardware

Rak4631

### Firmware Version

2.1.23.04bbdc6

### Description

Tested across multiple RAK devices to expedite and confirm.
Desired effect is to have a Repeater or Router with USB Serial an Bluetooth disabled.
All devices set to US region, each given two PSK channels with one being "admin" slot 2:
In each case, setting Bluetooth disabled seems to create a condition where the device becomes unresponsive to admin commands first, later followed by becoming completely unresponsive and later appears to be powered off (no LED activity). The time to become unresponsive can take up to 72 hours.
Tested with/without power (battery only).

Equipment:
RAK4631
RAK19007
RAK18001 - PWM Buzzer module (on some devices which don't appear to have any bearing on issue/full disclosure in case it becomes relevant)

Tested each of the following modes with Bluetooth disabled, eventually becomes unresponsive regardless of USB serial setting:
Client/router - because issues in router were previously known
Router
Repeater

All devices where Bluetooth is enabled have not exhibited issues. Snip from serial debug shows only Power status when and after the device faults. No other clues are obvious.

### Relevant log output

```Shell
DEBUG | ??:??:?? 8781 [Router] Incoming message was filtered 0x5185b8ba
DEBUG | ??:??:?? 8788 [Power] Battery: usbPower=0, isCharging=0, batMv=4186, batPct=98
DEBUG | ??:??:?? 8808 [Power] Battery: usbPower=0, isCharging=0, batMv=4172, batPct=96
DEBUG | ??:??:?? 8828 [Power] Battery: usbPower=0, isCharging=0, batMv=4188, batPct=98
DEBUG | ??:??:?? 8848 [Power] Battery: usbPower=0, isCharging=0, batMv=4190, batPct=98
DEBUG | ??:??:?? 8868 [Power] Battery: usbPower=0, isCharging=0, batMv=4162, batPct=95
DEBUG | ??:??:?? 8886 [RadioIf] (bw=250, sf=9, cr=4/8) packet symLen=2 ms, payloadSize=39, time 205 ms
DEBUG | ??:??:?? 8886 [RadioIf] Lora RX (id=0x4b25c3d2 fr=0xfd to=0xff, WantAck=0, HopLim=5 Ch=0xb9 encrypted rxSNR=10.5 rxRSSI=-73)
DEBUG | ??:??:?? 8886 [RadioIf] AirTime - Packet received : 205ms
DEBUG | ??:??:?? 8886 [Router] Add packet record (id=0x4b25c3d2 fr=0xfd to=0xff, WantAck=0, HopLim=5 Ch=0xb9 encrypted rxSNR=10.5 rxRSSI=-73)
DEBUG | ??:??:?? 8886 [Router] Using channel 0 (hash 0xb9)
DEBUG | ??:??:?? 8886 [Router] Using AES256 key!
DEBUG | ??:??:?? 8886 [Router] Software encrypt fr=2c1985fd, num=4b25c3d2, numBytes=23!
DEBUG | ??:??:?? 8886 [Router] decoded message (id=0x4b25c3d2 fr=0xfd to=0xff, WantAck=0, HopLim=5 Ch=0x0 Portnum=67 rxSNR=10.5 rxRSSI=-73)
DEBUG | ??:??:?? 8886 [Router] handleReceived(REMOTE) (id=0x4b25c3d2 fr=0xfd to=0xff, WantAck=0, HopLim=5 Ch=0x0 Portnum=67 rxSNR=10.5 rxRSSI=-73)
DEBUG | ??:??:?? 8886 [Router] Module 'DeviceTelemetry' wantsPacket=1
INFO  | ??:??:?? 8886 [Router] Received DeviceTelemetry from=0x2c1985fd, id=0x4b25c3d2, portnum=67, payloadlen=19
DEBUG | ??:??:?? 8886 [Router] Module 'DeviceTelemetry' considered
DEBUG | ??:??:?? 8886 [Router] Module 'routing' wantsPacket=1
INFO  | ??:??:?? 8886 [Router] Received routing from=0x2c1985fd, id=0x4b25c3d2, portnum=67, payloadlen=19
DEBUG | ??:??:?? 8886 [Router] Routing sniffing (id=0x4b25c3d2 fr=0xfd to=0xff, WantAck=0, HopLim=5 Ch=0x0 Portnum=67 rxSNR=10.5 rxRSSI=-73)
INFO  | ??:??:?? 8886 [Router] Rebroadcasting received floodmsg to neighbors
DEBUG | ??:??:?? 8886 [Router] Using AES256 key!
DEBUG | ??:??:?? 8886 [Router] Software encrypt fr=2c1985fd, num=4b25c3d2, numBytes=23!
DEBUG | ??:??:?? 8886 [Router] enqueuing for send (id=0x4b25c3d2 fr=0xfd to=0xff, WantAck=0, HopLim=4 Ch=0xb9 encrypted rxSNR=10.5 rxRSSI=-73)
DEBUG | ??:??:?? 8886 [Router] txGood=100,rxGood=229,rxBad=0
DEBUG | ??:??:?? 8886 [Router] rx_snr found. hop_limit:4 rx_snr:10.500000
DEBUG | ??:??:?? 8886 [Router] rx_snr found in packet. As a router, setting tx delay:294
DEBUG | ??:??:?? 8886 [Router] Delivering rx packet (id=0x4b25c3d2 fr=0xfd to=0xff, WantAck=0, HopLim=5 Ch=0x0 Portnum=67 rxSNR=10.5 rxRSSI=-73)
DEBUG | ??:??:?? 8886 [Router] Update DB node 0x2c1985fd, rx_time=0, channel=0
DEBUG | ??:??:?? 8886 [Router] Forwarding to phone (id=0x4b25c3d2 fr=0xfd to=0xff, WantAck=0, HopLim=5 Ch=0x0 Portnum=67 rxSNR=10.5 rxRSSI=-73)
WARN  | ??:??:?? 8886 [Router] ToPhone queue is full, discarding oldest
DEBUG | ??:??:?? 8886 [Router] Module 'routing' considered
DEBUG | ??:??:?? 8886 [RadioIf] Starting low level send (id=0x4b25c3d2 fr=0xfd to=0xff, WantAck=0, HopLim=4 Ch=0xb9 encrypted rxSNR=10.5 rxRSSI=-73 priority=64)
DEBUG | ??:??:?? 8886 [RadioIf] (bw=250, sf=9, cr=4/8) packet symLen=2 ms, payloadSize=39, time 205 ms
[... 231 lines cut ...]
DEBUG | ??:??:?? 9888 [Power] Battery: usbPower=0, isCharging=0, batMv=4173, batPct=97
DEBUG | ??:??:?? 9908 [Power] Battery: usbPower=0, isCharging=0, batMv=4158, batPct=95
DEBUG | ??:??:?? 9928 [Power] Battery: usbPower=0, isCharging=0, batMv=4163, batPct=96
DEBUG | ??:??:?? 9948 [Power] Battery: usbPower=0, isCharging=0, batMv=4162, batPct=95
...
DEBUG | ??:??:?? 10808 [AirTime] Rotating airtimes to a new period = 3
DEBUG | ??:??:?? 10809 [Power] Battery: usbPower=0, isCharging=0, batMv=4161, batPct=95
DEBUG | ??:??:?? 10829 [Power] Battery: usbPower=0, isCharging=0, batMv=4176, batPct=97
DEBUG | ??:??:?? 10849 [Power] Battery: usbPower=0, isCharging=0, batMv=4166, batPct=96
DEBUG | ??:??:?? 10869 [Power] Battery: usbPower=0, isCharging=0, batMv=4158, batPct=95
DEBUG | ??:??:?? 10889 [Power] Battery: usbPower=0, isCharging=0, batMv=4163, batPct=96
DEBUG | ??:??:?? 10909 [Power] Battery: usbPower=0, isCharging=0, batMv=4162, batPct=95
DEBUG | ??:??:?? 10929 [Power] Battery: usbPower=0, isCharging=0, batMv=4162, batPct=95
...
DEBUG | ??:??:?? 14408 [AirTime] Rotating airtimes to a new period = 4
DEBUG | ??:??:?? 14409 [Power] Battery: usbPower=0, isCharging=0, batMv=4159, batPct=95
DEBUG | ??:??:?? 14429 [Power] Battery: usbPower=0, isCharging=0, batMv=4159, batPct=95
DEBUG | ??:??:?? 14449 [Power] Battery: usbPower=0, isCharging=0, batMv=4161, batPct=95
DEBUG | ??:??:?? 14469 [Power] Battery: usbPower=0, isCharging=0, batMv=4167, batPct=96
DEBUG | ??:??:?? 14489 [Power] Battery: usbPower=0, isCharging=0, batMv=4157, batPct=95
DEBUG | ??:??:?? 14509 [Power] Battery: usbPower=0, isCharging=0, batMv=4180, batPct=97
DEBUG | ??:??:?? 14529 [Power] Battery: usbPower=0, isCharging=0, batMv=4178, batPct=97
DEBUG | ??:??:?? 14549 [Power] Battery: usbPower=0, isCharging=0, batMv=4177, batPct=97
...
```

thebentern, 2023-10-13:
I have seen this happen as well, and I believe @RicInNewMexico has also. I have not been able to pinpoint what's happening yet. I have a theory that there is some SoftDevice construct shuts down the device after a certain period of BLE inactivity, but I can't seem to find any documentation about this behavior. Not sure what else it could be because, as you found, grabbing logs of what's happening doesn't seem to yield anything suspicious.

cordelster, 2023-10-15:
I have ordered a rakdap, though I believe China is on holiday till the first week of November. Either way it will be a while before I can check and see if it will be of any help.

HookdomPonix, 2023-10-25:
I too am encountering the same issue.

HookdomPonix, 2023-10-26:
@cordelster There is a place in the US that ships fast and has them in stock:  

https://store.rokland.com/collections/rakwireless-products/products/rakdap1-debug-and-flash-tool-nanodap-v-2-3

HookdomPonix, 2023-11-21:
I have the RAKDap hooked up to a 4631 and am monitoring the terminal.   I have just disabled Bluetooth and will see if I can replicate with some data coming out of it.

cordelster, 2023-11-22:
Finally just got my order from Ali a couple days ago and I about gave up,
though I haven't got it all setup yet. Will be doing so in the next few
days.

Since others have told me they have RAKdap here, I presume this has
already  been done with no definitive results. I'll try anyway as well.

jpedlowCDW, 2023-12-29:
I think I'm encountering something similar on my TTGO Tbeam 1.1 as well, if my phone goes out of range (bluetooth disconnect) the device becomes responsive and will not work until I entirely reflash the device (issue persists between attempts to reboot etc)

HookdomPonix, 2023-12-31:
Updates:   I found another RAK powered off entirely and it wouldn't restart/reboot/anything.  Had to pull power and go that way which sucks.   

Using the RAKdap has been helpful but not conclusive in this issue.  It almost acts like flushing the bluetooth cache is resetting some interrupt in radiolib.   I can confirm it will also do this if a phone is removed from the unit for a long time - like no device nearby even.   

Still troubleshooting more.

David-Woodward, 2024-03-22:
I'm pretty sure I'm having this same issue with my RAK  4631.  I enabled bluetooth and it's been online for a full day now.  Seemed to start having problems within 6-12 hours for me with bluetooth disabled.

I wonder if setting the device up as a repeater with rebroadcast mode set to ALL_SKIP_DECODING might resolve this as well.  Not saying that would be a solution or even a workaround, but it might help troubleshoot it.  I'm guessing that mode doesn't produce the bluetooth event notifications that @HookdomPonix theorized about earlier since the device is supposed to be just forwarding packets along without even decoding them.  If that's sounds feasible and you'd like me test it, just let me know.  Though it may be next week before I have chance to do that.

olivierfaber, 2024-03-22:
Does this bug apply in router mode or only when bluetooth is specifically turned off?
I have a remote node that I think is suffering from this issue and I don't know if I should just turn on the bluetooth toggle or switch it to router_client role..

dustingrady, 2024-04-12:
I've had this happen to me both in CLIENT_ROUTER and REPEATER modes recently.

thebentern, 2024-04-12:
Role does not change the behavior. For NRF52, you have to explicitly set bluetooth.enabled = false to turn off the BLE radio and cause this issue.

thatSFguy, 2024-04-12:
I found the same issue and I'm happy the router node still accessible.  As a workaround, I'll change the BLE pass code to something difficult and enable it.  Thanks!

HookdomPonix, 2024-04-12:
I had worked this issue extensively with a RakDAP.   It...was never evident what's causing it.  I suspect the softdevice gets into a bad state when it gets lonely because BT is off and maybe a buffer overflows and starts kerflooiing.  

The FreeRTOS threads are still running but not all of them when it does this.  

I did confirm it looks like flushing the Bluetooth message buffer IS working and its not an out of control message buffer doing this.   It starts pruning itself on time and per spec.  

I also confirmed that power utilization with a very sensitive system for measuring it was not impacted in a notable way by disabling bluetooth.  The NRF52 is just plain efficient at power management regarding bluetooth.  

Suggestions -  a) more troubleshooting of this issue, maybe by someone who knows this hardware platform better than I do.  b). we should implement a lightweight security layer on the BT pin code if we can.   like: 

if 10 failed pin tries
   check if admin channel present
   if present, enable managed node and prevent pairing of BT until disabled by an admin controller
   if not present, erase key material and other sensitive data.   factory reset, wipe flash

thatSFguy, 2024-04-13:
Another option could be reducing the effectiveness of the bluetooth antenna, thereby reducing the transmission distance, by replacing the antenna with a 50 Ohm resistor.  This would basically be a dummy load.  Not sure by how much the distance would be reduced by...

thatSFguy, 2024-05-10:
Actually the simplest solution to enhance the security is to just enable the random BLE Pin via the admin channel.  Since I'm using it as a router, it's virtually impossible to brute force the pin to connect to BLE.

SpudGunMan, 2024-05-10:
The functionality of BT to turn it off prevents anyone from seeing it at all. Pin isn't the issue. 

What about removing the Bluetooth wake value from power settings in yaml config?

xlfe, 2024-05-16:
I think I'm seeing this too, on a RAK4361 with bluetooth disabled, version 2.2.24.e6a2c06 - worked for 3 days 1 hour and then didn't wake again for a telemetry send and unresponsive on the mesh

mike6789, 2024-07-15:
I am not sure if this is my issue as I have not inspected logs but I have a Wisblock device acting as ROUTER that freezes randomly and needs to be manually power cycled.

fifieldt, 2024-09-16:
Workaround has been in place for a couple months and no further issues reported.

closed 2024-09-16 by fifieldt