meshtastic/firmware #2895: [Feature Request]: Traceroute - label mqtt links

mrekin, 2023-10-18:
### Platform

NRF52, ESP32

### Description

Hi!
In my opinion, it will be useful to mark connections via mqtt when displaying a trace. 
Like _"Node A" -> "Node B [mqtt]" -> "Node C"_
This can help with understanding the mesh network topology and planning it.

It can probably be implemented by adding node labeling to payload when sending (or recieving) packet to mqtt.
Thanks.

tavdog, 2023-10-20:
So the suggestion is to label whenever the route goes through an mqtt connection or just hits a node that might also be connected via mqtt ?

mrekin, 2023-10-20:
Suggestion is to label route parts, which goes throw mqtt.
Then node re-transmits route packet - it add own id to payload and sends it to the air and, if connected, to mqtt.
Before sending to mqtt to the payload can be added additional label, so if target node recieved packet over the air - there no label, in case mqtt - label will be here.

mrekin, 2023-10-20:
As result I want know which part of trace  was over the air, which over mqtt
Like this :)
[image]

GUVWAF, 2023-10-20:
Currently there's no information added to the packet to know whether it originated from MQTT or not, but it is one of the goals for 3.0: https://github.com/orgs/meshtastic/projects/20?pane=issue&itemId=39847626

thebentern, 2023-12-02:
This is not feasible until 3.0, where we can trace packet transport metadata. I have an issue on the 3.0 board to add this information, because I have big plans for BLE / TCP only transmissions going forward.