meshtastic/firmware #2896: [Board]: reconsideration of the chatter 2.0

rimwulf, 2023-10-19:
### SOC

ESP32

### Lora IC

DL-CC68-S-868

### Product Link

https://www.samsclub.com/p/circuitmess-chatter-stem-project-build-and-code-walkie-talkies/PL980418145

### Description

Requesting reconsideration of Chatter 2.0 support as product is still being produced.

Confirmed product is in-fact open source with hardware, schematics, and code.  I included photos of mine board for reference and in contact with CircuitMess to see if original PCB files can be obtained. 

Latest interest I've found was less than 30 days ago so produce cannot be EOL.

Refrence:
https://github.com/meshtastic/firmware/issues/2707#issue-1845692213

Original code
https://github.com/CircuitMess/Chatter2-Firmware

Fork interest
https://github.com/bestander/Chatter2-Firmware-for-Chatter1

Hardware
  LoRa module DL-CC68-S-868

  ESP-WROOM-32

Hardware schematics:
[image]

Reference photos.
[image]
[image]
[image]

If permitted this will be updated with know info I just ask for consideration and not closing this issue

code8buster, 2023-10-19:
It looks like you have everything you need to port meshtastic to it and make a pull request. PRs are generally accepted.

The link you provided states that they're no longer available, as does CircuitMess' own website. Their only mention of what _might_ be available is a projection; to be shipped with their subscription boxes in 2025.

I don't think any devs are volunteering to secondhand-purchase and support what likely is a deprecated device. Unqualified statements like "Latest interest I've found was less than 30 days ago so produce cannot be EOL." do not function as proof that a device is not EOL.
They sold just over 17k units to Sams Club a year ago, and there's no indication that they will sell more:
https://hr.linkedin.com/posts/dorotearudec_so-stoked-about-our-biggest-collaboration-activity-6996443037845344256-aqs0

sheehanisageek, 2023-11-01:
I have over 20 of these chatter 2 units and would love to see a meshtastic port for them.

rimwulf, 2023-11-03:
Yeah.

code8buster, 2023-11-04:
> Yeah.
> […](#)

Do either of you have any interest in creating the variant.h files and testing? You could use the DIY variant as a jumping off point.

rimwulf, 2023-11-05:
I do! But I'm not sure yet I'm trying to learn to code. I know someone who
might as well But I don't want to awful promise or overstate my abilities.

code8buster, 2023-11-06:
I found that they actually still sell some variant of this device on etsy. Any idea if this is the same hardware, or old stock of an earlier revision?
https://www.etsy.com/listing/1357363727/circuitmess-chatter-diy-lora

fang64, 2023-11-29:
I just acquired one of these, I'll look at porting it, these are pretty interesting. @rimwulf @code8buster

caveman99, 2023-12-03:
The reset pin is connected to 3v3 that's perfectly ok. Have a look at the picomputer S3 definition, it's wired up in a similar way.

rimwulf, 2023-12-17:
> I found that they actually still sell some variant of this device on etsy. Any idea if this is the same hardware, or old stock of an earlier revision? https://www.etsy.com/listing/1357363727/circuitmess-chatter-diy-lora

Oh sorry haven't been getting updates. 
No that's the original Chatter awa Chatter 1.0 though it has much of the same hardware but that is in fact overpriced legacy. The chatter 2.0 has a few changes and it costs less than $30 US from eBay. Bought mine from Walmart but the seller decided to raise the price after selling 10 of these kits.

Someone did make code that allowed the original to talk to the 2.0 I didn't find it relevant but I can share it if it helps. 

Spoke to the company they hinted that they are planning on making another run so we can't have the PCB files.

garthvh, 2023-12-18:
Seems unlikely that the Walmart version is more capable, what are the hardware differences?

fang64, 2023-12-19:
> Seems unlikely that the Walmart version is more capable, what are the hardware differences?

I have no idea honestly, I've seen them on ebay for as little as $15 btw

GPSFan, 2023-12-25:
I'll add my 2cents in here, after seeing fang64's posts on Discord, I ordered one from e-bay. Got it, and was able to fiddle enough with variant.h and a few other files to get it compiled and loaded onto the hardware. It worked to the extent that I could send messages on Medium-Slow.
The display looked nice, almost as nice as my t-deck. So I got that running. I wanted to add GPS (of course) so it was fairly easy to re-purpose the small 4 pin connector on the side.
The only remaining issue I had was the crappy LLCC68. Browsing around the Internet, I found the Heltec RA-62 
https://heltec.org/project/ht-ra62/
https://www.aliexpress.us/item/3256805259034353.html?spm=a2g0o.productlist.main.1.17c950fekrb5ss&algo_pvid=af290929-42b6-4ffa-9c56-e01322af7474&algo_exp_id=af290929-42b6-4ffa-9c56-e01322af7474-0&pdp_npi=4%40dis%21USD%214.80%213.22%21%21%214.80%21%21%40210318bb17021821910745339e0e5e%2112000033108798513%21sea%21US%21132573136%21&curPageLogUid=RgrSoazZeWtr

The RA-62 is an almost pin compatible replacement for the LLCC68, but uses an SX1262.
I ordered a couple and swapped them out on my Chatter2s, I had to cut and jumper the Busy signal as that was the only pin not compatible. If you are replacing the Lora module, a single cut & jumper should be a non-issue.
Voila a Chatter2 that is fully Meshtastic compatible.
I was also able to get one button (Wake, lower right corner) to serve as ther user button. It wakes the device up out of deep sleep just fine (with my software mods).
I could do a pull request if there is any interest.
[image]

GPSFan, 2023-12-25:
More Pix ;>)
[image]
[image]

fang64, 2023-12-27:
@GPSFan I haven't had time to get the dial pad keyboard working, I mean the radio on it doesn't support SF12 but it's still useful I think for those wanting to use it as is. That's pretty awesome though.

GPSFan, 2023-12-27:
Replacing the Lora module is not that hard, and without replacing it you can still use the Medium-Slow preset with other SX1262 based devices, The actual data rate is faster than Long-Fast and the link budget is down only 2.5db. A better antenna will more than make up for that loss. (IMHO all those springy coil antennas are nothing but crap, yeah you can make one look nice on a VNA, but as soon as anything gets near them the resonant point can shift by 20-30Mhz, and there goes what little gain you had)

brewtide, 2024-01-01:
Would either of you like to share your modifications to variant.h and other changes that you've managed to get the 'stock' profile of this hardware to roughly work on? I would be appreciated!

GPSFan, 2024-01-02:
After the holidays are over (a day or two), and after I un-bork my local repo, I'll submit a PR, I got behind in updating stuff and sheep has hit the can. I'm also working on an audit of all the gnss init code for the u-blox receivers, I've found a bunch of typos and incorrect messages. Lots of testing to do before I go live with that PR though.

RebootLoop, 2024-01-02:
I picked up three of these kits (one for each of my boys, one for myself) for Christmas for a fun little family project. Didn't realize that they actually communicate using LoRa til we started playing around with them. Not a big coder/programmer, but would happily test anything you might put together for the stock units. Really hope we can get the keypads to work so we can use them as self-contained standalone devices.

Goocherrific, 2024-01-07:
I bought a dozen and would love to test as well

GPSFan, 2024-01-07:
Sorry, I haven't gotten the PR out yet, been busy with the GPS audit, it's grown larger than I thought it would.
Also been doing some power measurements on the Chatter 2. Here is a screenshot of the point when the backlight turns off.
[image]

GamingNJncos, 2024-01-14:
@GPSFan Would very much be interested in your PR!  Had similar idea on radio replacement to port this over and just stumbled across this.

GPSFan, 2024-01-14:
I have to apologize for not getting the PR out. RealLife(tm) and DayJob(tm) have not left much time for fiddling with Meshtastic.  Hopefully tomorrow or Monday. 
Replacing the Lora Radio takes a bit of SMD rework knowledge, but is not beyond too many folks skill level. Even without it the Chatter 2 can communicate with the Medium-Slow preset.

fang64, 2024-01-14:
@GPSFan @GamingNJncos same boat here, I still need to get the keypad working as well.

code8buster, 2024-01-31:
https://github.com/meshtastic/firmware/releases/
The chatter2 firmware is in, starting with 2.2.20
Enjoy! Keep in mind your unmodified devices must use Medium Slow or faster presets.

closed 2024-01-31 by code8buster

n2hyo, 2024-03-01:
Ok so how the heck do i get the bin file uploaded to the chatter, i have been playing around with no luck.

Thanks

GPSFan, 2024-03-02:
Use the cli method with esptool:
https://meshtastic.org/docs/getting-started/flashing-firmware/esp32/cli-script/
Or clone the repo and build the firmware yourself using Platformio:
https://meshtastic.org/docs/development/firmware/build/

The web flasher has not been updated to support the Chatter 2 as yet.

n2hyo, 2024-03-02:
Thanks i got the firmware loaded and can send messages by using the web client but only one
key on the keyboard works as fang64 mentioned. Guess it is still in the works.

Going to order some ht-ra62 now.

GPSFan, 2024-03-03:
Yep, only one key until someone does more coding than I have time for. The RA-62 works great, only one cut & jumper necessary, if you  have any issues with the swap let me know, I used a small hot plate to remove the old module. Of course you now have a UFL antenna connector instead of that springy thingy that doesn't resonate anywhere near 915MHz.

eureekasigns, 2024-04-27:
> Yep, only one key until someone does more coding than I have time for. The RA-62 works great, only one cut & jumper necessary, if you have any issues with the swap let me know, I used a small hot [...]

Great stuff, thank you. How small of a hot plate? 

Sadly, I have ruined pads using just hot air and lifting the chip. Heat seems to dissipate too fast with just hot air on these, and a hot plate sounds like it's possibly better in this case but would welcome advice on that for sure.

That said, I have been able to successfully get one working! It's very cool to see one of these become a node but I'd like to do a few more for kicks without ruining any more of them in the process.

GPSFan, 2024-04-27:
I have a Thermolyne HP-2305B. it is rated at 325 watts and I set the temp with a thermocouple to a bit above the melting point of Lead free solder. It takes about 30 seconds for the module to heat up enough to pop it off.

eureekasigns, 2024-04-27:
That sounds good, thank you. 
The one I had working flaked out this am, then the power switch came off of it somehow. None of the others I had partially working seem salvageable at least yet. If nothing else it's a learning experience at this stage. I had higher hopes that I'd be better able to get more working, but alas, my luck has not been the greatest with the ones I had. 

I may order some more and some more chips to experiment with, but they're certainly fragile little things. In any case, it's still very cool that such a thing works (at least, when done nice and proper).

GPSFan, 2024-04-27:
Yeah, it is, the Chatter 2 is a great Meshtastic platform for testing all sorts of stuff, if you hose one up, "so what" they are cheap enough. Best of luck with the rest of your boards.

eureekasigns, 2024-04-29:
Agree. Thanks!

One other really basic question. To clarify, are you cutting the pin on the CPU or the pad where the LoRa chip goes? I can't quite tell from the picture, but it looks like the CPU pin is no longer making contact with the board, is that correct?

GPSFan, 2024-04-29:
Sorry, I never uploaded the picture of where the cut was.
[image]

eureekasigns, 2024-04-29:
I'll. Be. ... 
Wow. That was the ticket. 
There's no hope for some of them that I badly messed up, but this was the ticket. 

It was still beneficial to learn from my mistakes on these. Thrilled to have at least some working now! Hoping to figure out a case, or project box next. 

Thank you so much for the great responses, the help, and for the work on getting these working with Meshtastic!
[image]

scott1627, 2024-05-09:
Just wanted to throw this out here in case someone might like to use it. I designed a shell in the style of a handheld transceiver. Uses two 18650 batteries, a 12mm latching toggle switch to relocate the power switch to the top, an SMA pigtail and an inexpensive 915mhz whip antenna. It is by no means perfect, but feel free to use it if you would like.

https://www.thingiverse.com/thing:6614717

[image]
[image]
[image]
[image]
[image]

AughtCool, 2024-05-10:
Just thought I would chime in.  I was able to load [firmware-2.2.20.af5ac32.zip] on both my chatter 2's using the BIN file and espwebtool (https://esp.huhn.me/).  Seems to be working fine.  I changed to medium-slow as suggested but will try longfast as well.  Now if we can get the keyboard to work!  Thanks!

eureekasigns, 2024-05-10:
That is awesome! Thank you for sharing. 

I will have to see if I can ask to have some made by those near me with a printer. 
Have been struggling with a project box that's either too small or too large. 

One thing I want to add to this is a momentary button for the firmware update unless there's some better way than wiring to short pin 1 to ground while doing command line install. Easy enough to add that somewhere. 

Is there an opening for the USB port by any chance?

Also curious how you wired the power button. I haven't played with relocation of the power yet, but have got it to charge via the port using a wire to the top pin of the power switch. Using same little charge controller board setup. 

Looks really cool. Hopefully can get someone to print me some up!

AughtCool, 2024-05-10:
Very nice case, will have to print some.  Question, how did you modify the antenna.  Can you post a closeup?  Thanks!

scott1627, 2024-05-10:
Here is a very rough little schematic I threw together. I pulled 5v power from the type c port and fed it into a tp4056 lithium charging circuit. The image shows the points I soldered to. As for the antenna, I did the radio swap as documented by GPSfan in the earlier posts. If you wanted to use the stock radio with a different antenna, it can be done with an SMA pigtail. Coax conductor will need to be soldered to the pad where the springy antenna was after removal and the shield can be soldered below it.
[image]

eureekasigns, 2024-05-11:
Nice move on power to the battery input from the output of the charging board. I had thought about going to the switch somehow but this is easier and makes sense, thanks!

Gnu-Bricoleur, 2024-05-17:
I'm just getting into Meshtastic and I happen to have two Chatter V1 lying around, I would love to be able to use them to connect to my node and send messages.
Hardware seems to be similar to Chatter V2.0 so I should be able to just flash the latest FW ?
Would it be possible to use the key pad to send messages with a Meshtatic plugin ?
Would this be a good starting point ?
https://meshtastic.org/docs/development/device/module-api/
I'm willing to give it a go but I'm not sure where to start so I might need a nudge in the right direction !

GPSFan, 2024-05-17:
You could try flashing the latest firmware, but IMHO it probably won't work. Unless you change the Lora module you will only be able to talk to other Meshtastic nodes on one channel that is NOT Long-Fast.
I've never looked at the differences to the Chatter 1, don't have any and am tied up on other projects.
Have a go at figuring out the keyboard.

Gnu-Bricoleur, 2024-05-21:
Both Chatter V1.0 and V2.0 uses the same ESP32-WROM-32 module and DL-LLCC68-S-868 Lora module, and after looking at the schematics, there doesn't seem to be any significant differences in the modules connections.
So I tried flashing the Chatter V1.0 firmware and it works in Medium-Slow just as GPSFan said it would and I can exchange messages with my other node !
[image]
Thanks @GPSFan 😄

tomhanax, 2024-05-22:
> Would it be possible to use the key pad to send messages with a Meshtatic plugin ?

Also my point, having several Chatters 2, there will be more fun to have them "meshtasticized" also with the keypad. I asked similar question on https://github.com/meshtastic/firmware/discussions/3927. I am eager to start to code, but I do not know, where is the good place to start digging :(

Gnu-Bricoleur, 2024-05-22:
@tomhanax Chatter V1.0 and V2.0 use the same keypad circuit based on two 74HC165 PISO shift registers. 
After looking at Meshtastic code, core modules are not the way to go to add keypad support
Meshtastic actually already support on device message input through the canned message module (https://meshtastic.org/docs/configuration/module/canned-message/) 
We just need to add support for Chatter keypad. 
I started working on it yesterday without results yet. 
I can start a PR so we can try to make that work together if you want!

tomhanax, 2024-05-23:
@Gnu-Bricoleur so some good news after all :) I have T-deck and although message writing is not as nice and friendly as you expect from hardware for messaging, I thought that extending some classes/code using similar principle will do the job. But although I consider myself as good coder, I am lame in C++, can read it but that's all. I will gladly offer any help that I could, but I doubt that actual coding will do any good. But who knows... Anyway, I can see that you already considered this "extend the core" approach and abandoned it. I am curious and want to learn: can you explain, why? What is missing in core classes and why this is not good approach? What about to refactor some core code to be able more flexibly plug in more keypads (touch for example)? 

Also, as the second goal (if I was a C++ guy), I will surely upgrade that message writing interface. Some bigger font, recipient list or whatever else. To be first class citizen.

Anyway, I am glad that Chatter 2 exists, for the price you can pick it on Amazon (or similar) you have complete standalone device that you can tweak more and more. T-deck is nicer, sure, touch screen, music and so... but it cost me 4 times more. For the meshtastic quite pricey and overkill I'd say....

eureekasigns, 2024-05-23:
This is really cool to see. Hoping it can be figured out. Would it be possible or at all helpful to borrow any of the source code for the original?
https://github.com/CircuitMess/Chatter2-Firmware/

Side notes on the 3d printed case - the design is interesting but I may look at something else or changing some things about it, though I don't know if the costs are worth it yet to print more than a couple. 
First, the SMA adapter hole on the inside is extremely tight. It's difficult to get the end though at all. Had to shave off a bit. Related to that, I'm not sold on it being recessed. This makes it difficult to tighten the nut on top and to fit a regular antenna. The rest is ok, but I'd have done a few things a little differently. 

Maybe if I can figure out how to change things or just design based on it I'll go for it. It's still very, very cool that this is available and I thank the designer for making and sharing!

eureekasigns, 2024-06-01:
> @tomhanax Chatter V1.0 and V2.0 use the same keypad circuit based on two 74HC165 PISO shift registers. After looking at Meshtastic code, core modules are not the way to go to add keypad support [...]

Excited someone's looking into possible use of the buttons on this. Looks like there's some work on the linked issue with an alpha version that might resolve. Look forward to the possibility of more buttons doing more things!

Gnu-Bricoleur, 2024-06-29:
@GPSFan Which GPS module do you recommend to use with the Chatter ?
I looked through the discussion but couldn't find the reference.

dogboyaa1, 2024-06-30:
> @GPSFan Which GPS module do you recommend to use with the Chatter ? I looked through the discussion but couldn't find the reference.

Also had the same question. Thank you all for the great work. I love these little things

GPSFan, 2024-06-30:
I've used the Chatter 2 as a test platform to test many GPS receiver types.
I am partial to u-blox modules, but not Neo-6 or Neo-7.

For a good price/performance I would suggest:
ATGGM336   ... cheaper... but good...
u-blox M10   
u-blox F10
tau1202
u-blox M8N
u-blox M9
u-blox F9   ... more expensive... IMHO outside most (all) Meshtastic use cases.

Each has its good & bad points, some have better power management, some are a better receiver.
None (without hardware hacking) support power supply control. So with defaults the GPS is "always on" or in PMREQ standby.

I added the Chatter 2 to Meshtastic because it is a great hacking platform, cheap and lends it self to hacking.
The folks working on adding keyboard support have my greatest respect as that is beyond my pay grade WRT programming skills.

dogboyaa1, 2024-06-30:
> Just wanted to throw this out here in case someone might like to use it. I designed a shell in the style of a handheld transceiver. Uses two 18650 batteries, a 12mm latching toggle switch to relocate the power switch to the top, an SMA pigtail and an inexpensive 915mhz whip antenna. It is by no means perfect, but feel free to use it if you would like.
> 
> https://www.thingiverse.com/thing:6614717
> 
> ![IMG_2934](https://private-user-images.githubusercontent.com/169350348/329201958-f84a278a-631d-4bec-bc6e-879ec24306bf.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk3MTAyODQsIm5iZiI6MTcxOTcwOTk4NCwicGF0aCI6Ii8xNjkzNTAzNDgvMzI5MjAxOTU4LWY4NGEyNzhhLTYzMWQtNGJlYy1iYzZlLTg3OWVjMjQzMDZiZi5KUEc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYzMFQwMTEzMDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02MTk5MjU1YzM2MGY1ZGQ3OGIxYmI4MzViM2I5M2I4ZjVkNGIzNTUzOGExM2ZiOWY2YTAxOTcyODdmNjI1MDllJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.kVloqL2BBa5KOcQQRAM20_YAjHyjCCefpTQcjWicfmw) ![IMG_2936](https://private-user-images.githubusercontent.com/169350348/329201968-dd246a19-7bfe-403f-8a37-98c5e2a62b64.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk3MTAyODQsIm5iZiI6MTcxOTcwOTk4NCwicGF0aCI6Ii8xNjkzNTAzNDgvMzI5MjAxOTY4LWRkMjQ2YTE5LTdiZmUtNDAzZi04YTM3LTk4YzVlMmE2MmI2NC5KUEc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYzMFQwMTEzMDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kNjAyY2QzYTJkYzM4MTBkOGI4ZGZlYjkwMmNiNzVhMmJiZDk1ZmM0ZDAyNDhiYzM3ZGRlNGJjYjMyMzYyMTNiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.hAwwYSKLnBj48zTycVPI5pKR2a7aHrWl2OIhdH7oaOg) ![IMG_2932](https://private-user-images.githubusercontent.com/169350348/329201980-1af5396c-a66f-4e4a-a479-1db51f5d8773.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk3MTAyODQsIm5iZiI6MTcxOTcwOTk4NCwicGF0aCI6Ii8xNjkzNTAzNDgvMzI5MjAxOTgwLTFhZjUzOTZjLWE2NmYtNGU0YS1hNDc5LTFkYjUxZjVkODc3My5KUEc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYzMFQwMTEzMDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xYWM2MGU2YmUxNjdlZTNiMGQxNTgzYjNhMGJjZjA3MmEzMzE1MDVjNWQ2NzE0NGQyOGQ1YTg4MTY3OGI0YTkxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.zVGEqFk-ERZLKWKhI8GK3S2-6x5VOrt8UlNYQEeF-Lc) ![IMG_2933](https://private-user-images.githubusercontent.com/169350348/329201998-68e2f998-18a6-4962-bcb4-c9362e5c7a7d.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk3MTAyODQsIm5iZiI6MTcxOTcwOTk4NCwicGF0aCI6Ii8xNjkzNTAzNDgvMzI5MjAxOTk4LTY4ZTJmOTk4LTE4YTYtNDk2Mi1iY2I0LWM5MzYyZTVjN2E3ZC5KUEc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYzMFQwMTEzMDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xYjE2YWMwOGY0NmI1YmQ4ZTA5YzU2NmMzYzhmZTAyNzRlYjlkMDEzZGQ5YmY5MWYwYWQ1MjAxZDRjN2I3NWQ1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.5ACs6ecSl1zGYgNTtVfW43iNwhC-cR2gCIgNlnxRT3c) ![IMG_2935](https://private-user-images.githubusercontent.com/169350348/329202008-68a04dc0-3b65-468f-bcad-cd15b4d5b064.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk3MTAyODQsIm5iZiI6MTcxOTcwOTk4NCwicGF0aCI6Ii8xNjkzNTAzNDgvMzI5MjAyMDA4LTY4YTA0ZGMwLTNiNjUtNDY4Zi1iY2FkLWNkMTViNGQ1YjA2NC5KUEc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYzMFQwMTEzMDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05OTNlYTQwNGQ5NGZiZWNkMzViODFiZTliZDAxNDkyYzllMGUzZmViMTlmODhkMmZlODNhZDBkZjY1YTQxZTk4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.lgR57y_xelTMryV4gYChKLYRpHiEd09negdF0VP6KdY)

Hi i wanted to ask you real quick to confirm. the lithium charger you mentioned it is going to output 5v? Would that be too much for the unit since it only runs on 3 x aaa batteries. I noticed new batteries are like 4.7v but figured I will ask before damaging my unit. thx!

dogboyaa1, 2024-06-30:
> I've used the Chatter 2 as a test platform to test many GPS receiver types. I am partial to u-blox modules, but not Neo-6 or Neo-7.
>
> For a good price/performance I would suggest: ATGGM336 ... [...]

Hi thank you for the suggestions! just curious so i dont make a mistake do you know the pinout. i thought i found on the chatter forum but wanted to double check with you.

scott1627, 2024-06-30:
> Hi i wanted to ask you real quick to confirm. the lithium charger you mentioned it is going to output 5v? Would that be too much for the unit since it only runs on 3 x aaa batteries. I noticed new [...]

The charge circuit outputs 4.2v which is slightly below a fully charged set of AAA batteries. The device will show around 80% charge with the stock firmware with a fully charged lithium cell however I am sure the firmware could be adjusted to expect 4.2v instead. Long story short, the voltage of the charger/ lithium cell will not damage the device as long as it is wired up correctly.

dogboyaa1, 2024-06-30:
> The charge circuit outputs 4.2v which is slightly below a fully charged set of AAA batteries. The device will show around 80% charge with the stock firmware with a fully charged lithium cell however [...]

Got it thank you

dogboyaa1, 2024-07-01:
> Hi thank you for the suggestions! just curious so i dont make a mistake do you know the pinout. i thought i found on the chatter forum but wanted to double check with you.

I was able to figure out the pin out. If looking at it from the side:

Pin Layout
Pin 1: Ground (Left Pin)
Pin 2: RX (Connects to GPS modules TX Pin)
Pin 3: TX (Connectd to GPS modules RX Pin)
Pin 4: 3.3v (Power)
[image]

tomhanax, 2024-07-01:
As very good work on keypad support was done by @Gnu-Bricoleur in #4022, this little devices are ALMOST field-usable - at least for me.

As ALMOST I mean I really miss one, very important feature: sound notification on message arrival. In the real world usage this is crucial, you can surely imagine. On T-Deck you will hear some beeping, we call it "frogs" :) (meshtastic is green you know... ;)

I made some other tweaks like font sizes and I am overall happy with them. But I can not find where to turn this sound on. I am sure that the Chatter is sound capable, since we can already here the "startup" sound, so this should be theoretically possible and easy. But where is it hidden?

Can anybody please help?

scott1627, 2024-07-02:
> As very good work on keypad support was done by @Gnu-Bricoleur in #4022, this little devices are ALMOST field-usable - at least for me.
>
> As ALMOST I mean I really miss one, very important feature: [...]

I was able to get the audible notifications working by setting "Output pin buzzer GPIO" to Pin 31 under the external notification config. I also enabled the toggles for alert GPIO buzzer when receiving bell and message. On a side note, Has anyone made a guide for flashing the newest firmware with keypad support to the device, or has this been added to the official builds?

tomhanax, 2024-07-02:
Thanks @scott1627, it works perfectly! Now I am finally able to use them on the field! 
Anyway, I think that the buzzer pin value should be changed in `variant.h`, too:
```
#define PIN_BUZZER 31
```
I tested it locally and it works, so maybe somebody with github workflow experience could change this, so user won't have to in the GUI.

[73 more replies]