meshtastic/firmware #2900: [Feature Request]: RTTTL tones over DAC devices

thebentern, 2023-10-21:
### Platform

ESP32

### Description

Seems like we should be able to utilize devices with built in speakers connected to native DAC / amps to use with Ext. Notification plugin for instance and play notifications tones. A couple of devices like the T-Watch S3 and T-Deck have hardware out of the box for this. 
Example for library usage:
https://github.com/earlephilhower/ESP8266Audio/blob/0abcf71012f6128d52a6bcd155ed1404d6cc6dcd/examples/PlayRTTTLToI2SDAC/PlayRTTTLToI2SDAC.ino

thebentern, 2023-10-21:
Even SAM for text to voice is an option in this library family. For example:
https://github.com/meshtastic/firmware/commit/df110f3ec4865fda586b2b846cec28c0840fd0ea

mverch67, 2023-10-25:
Unfortunately the ESP8266Audio library seems outdated. It does not support esp32-S3 and the newer esp-idf 5.1 I2S packages.

thebentern, 2023-10-26:
> Unfortunately the ESP8266Audio library seems outdated. It does not support esp32-S3 and the newer esp-idf I2S packages.

I have it working on the T-Watch S3 and T-Deck locally, out of the box. In terms of the esp-idf specific i2s packages, we've typically followed the "fix and fork" model for adding what we want to a library that is otherwise good for our use-case. Have you found another library you have in mind?

thebentern, 2023-12-14:
Closed with #2917

closed 2023-12-14 by thebentern

ulrichard, 2023-12-18:
What do I have to do to get a tone on a t-deck when receiving messages?
* I upgraded to the 2.2.16.1c6acfd Alpha firmware -> no sound
* I enabled the Ext Notif "Module Enabled" and "Alert Message" -> no sound
* Additionally enabled "Alert Message Buzzer" -> still no sound

Do I have to upload and configure a sound file to the device somehow? 
Or do I have to enable the Audio module?

ulrichard, 2023-12-18:
Ah, found it by reading the source code. I had to also enable "Use I²S Pin as Buzzer"