meshtastic/firmware #2901: [Feature Request]: Improved granularity of MQTT topics

crypto-smoke, 2023-10-21:
### Platform

NRF52, ESP32

### Description

Currently MQTT is utilized by nodes sending protobuf messages to a channel topic, for example `msh/2/c/LongFast`. This works and makes sense when using the MQTT broker to link networks via the internet. We could do better for telemetry however. With JSON enabled, nodes could publish sensor data to specific topics, such as `msh/2/telemetry/<node id>/battery-voltage`. This allows clients to subscribe to the sensor readings they care about without having to parse every message that the node has sent on that channel. 

Along with this, an improvement to the telemetry module could be made that enables labelling of sensors. This way if there is multiple temperature sensors hooked up, the user can name them things like `internal-temp` or `cpu-temp` and then these can be used in the MQTT topic names.

thebentern, 2023-12-16:
Moving to 3.0 board for tracking there

closed 2023-12-16 by thebentern

alphabbett, 2024-02-07:
How do I remove the extra topic names? ( /2/c ).  I use the topic names to automatically sort my data into mysql database, so I can't have these extra topics and I can't figure out how to get rid of them.