meshtastic/firmware #2902: [Feature Request]: LED Status mode options

reubenbijl, 2023-10-21:
### Platform

ESP32

### Description

I recently stumbled across this project when I ordered some Tlora-s3 boards.

I've been really impressed with the documentation and setup for the project overall, but one thing that didn't make sense to me was the LED_PIN blinking. I've had a brief look through the source code, and it looks like that other than while in deep sleep the LED blinks on and off based on CPU availability which feels to me more like a debug mode for the LED than a production feature.

One thought I had was that the LED could be used for showing when there is a new message or alert, and then be turned off again when the button was pressed.

Potentially this could be a new mode, but I wonder what other behaviours could be useful? I'm happy to implement this as a change but wanted to know the best way to share/discuss this feedback and whether it would be valuable.

mverch67, 2023-10-23:
> One thought I had was that the LED could be used for showing when there is a new message or alert, and then be turned off again when the button was pressed.

This exact function is already available through the external notification module.

garthvh, 2023-10-23:
The main led indicates activity (flash, cpu, lora) if the board has a second addressable led like the RAK wisblock it can be set to show the external notifications indication

closed 2023-10-23 by garthvh