meshtastic/firmware #2905: [Bug]: heltec wireless tracker 1.1, no display.

megabyte2010, 2023-10-23:
### Category

Hardware Compatibility

### Hardware

Heltec Wireless Tracker

### Firmware Version

2.2.12

### Description

I have a heltec wireless tracker v1.1. There is no display. The log shows that this board is being detected as v1.0.

### Relevant log output

```Shell
//\ E S H T /\ S T / C

INFO  | ??:??:?? 3 Booted, wake cause 0 (boot count 1), reset_reason=reset
DEBUG | ??:??:?? 3 HELTEC Detect Tracker V1.0
DEBUG | ??:??:?? 3 Filesystem files (483328/1048576 Bytes):
DEBUG | ??:??:?? 3  /static/.gitkeep (0 Bytes)
DEBUG | ??:??:?? 3  /static/Logo_Black.svg.gz (592 Bytes)
DEBUG | ??:??:?? 3  /static/Logo_White.svg.gz (600 Bytes)
DEBUG | ??:??:?? 3  /static/apple-touch-icon.png.gz (3162 Bytes)
DEBUG | ??:??:?? 3  /static/favicon.ico.gz (2221 Bytes)
DEBUG | ??:??:?? 3  /static/icon.svg.gz (842 Bytes)
DEBUG | ??:??:?? 3  /static/index-85eb6f51.css.gz (17110 Bytes)
DEBUG | ??:??:?? 3  /static/index-aaac80c4.js.gz (204247 Bytes)
DEBUG | ??:??:?? 4  /static/index-fc2e9253.js.gz (347 Bytes)
DEBUG | ??:??:?? 4  /static/index.html.gz (546 Bytes)
DEBUG | ??:??:?? 4  /static/maplibre-gl-f0d3274b.js.gz (207264 Bytes)
DEBUG | ??:??:?? 4  /static/robots.txt.gz (42 Bytes)
DEBUG | ??:??:?? 4  /static/site.webmanifest.gz (196 Bytes)
DEBUG | ??:??:?? 4 Using analog input 1 for battery level
INFO  | ??:??:?? 4 ADCmod: ADC characterization based on default reference voltage
INFO  | ??:??:?? 4 Scanning for i2c devices...
DEBUG | ??:??:?? 4 Scanning for i2c devices on port 1
```

RicInNewMexico, 2023-10-31:
For the dev that tackles this bug:

https://resource.heltec.cn/download/Wireless_Tracker

There are the schematics for the v1 and v1.1 Heltec Wireless Tracker. Heltec support confirms that there were some pin reassignments that took place with the v1.1 as well as changes to the power supply circuit. Of course they don't specify what the changes were, we'll have to compare schematics.

GPSFan, 2023-11-12:
And an "oh by the way..." the GPS chip used is a Unicore UC6580. there are at least 2 references to UC6850 in GPS.cpp.

megabyte2010, 2023-11-14:
Someone has the factory_test.ino for 1.1? I only find for 1.0

megabyte2010, 2023-12-01:
> @megabyte2010 please measure resistance between I2C's SCL and SDA lines, make sure they aren't shorted

No shorts between pins 41 and 42 and ground. The board was working with the default factory test when i boughted. The log above is saying that the board doesn't have any power source connected, but she was connected thru usb.

megabyte2010, 2023-12-01:
> Is there still no display?

Yes I. Checked between them and ground. No shorts. Like I said, the board was working with the default firmware that came with her. Then I flashed meshtastic and the display stop working. The board seems to freeze for 30 seconds or so, then the white led start flashing. I can connect with the android app but there is no display and maybe no gps.

GPSFan, 2023-12-01:
On the Wireless tracker can you check the voltage on the Ve pin, it is on the bottom left row of pins below the display.
That pin is Vext and needs to be 3.3V as it supplies the GNSS chip and the Display.

jp-bennett, 2023-12-01:
The problem is a mis-detect of the board version. That results in a pin not getting powered, and that pin-off means that the I2C scan works by timing out each probe, rather than getting a nak. The root problem is the mis-detect, but it's not at all clear why some of the 1.1 boards detect properly, and some don't.

megabyte2010, 2023-12-01:
One stupid question... All boards came with factory test installed by default. I have two boards 1.1. All came with that program working. All boards worked the first time I installed meshtastic. One of them, I erased and flashed again. Since then, there is no display. Erasing all eeprom doesn't do nothing. Installing with browser or with cli it's the same. It's "possible" that when we install meshtastic the first time, remains some of the factory code? That's why I want to find the factory_ test for 1.1. To try to reproduce the problem.

GPSFan, 2023-12-01:
So the real questions are, how does the firmware tell a v1.0 from a v1.1 and how/why is it getting confused.

megabyte2010, 2023-12-01:
Anyway, even supplying with 3.3v, the display not always starts. I'm giving up for now.

GPSFan, 2023-12-01:
that could be the interim solution, but now there is only one Wireless Tracker firmware flavor, and only one build variant. Looking at the GPS.cpp, GPS.h files there are lots of ifdefs for v03 and v05, but isn't clear to me (yet) how the decision is made as to v03 or v05.

GPSFan, 2023-12-01:
and maybe this is a problem with your one board. mine flashed ok with .13 and .14, although I could not get the web flasher to work. command line flashing worked three times so far on my board.

megabyte2010, 2023-12-01:
One of them flash and works ok with web flasher, the other only works when she wants. I need the factory file to find out if it is hardware fault or not.

jp-bennett, 2023-12-01:
We've tossed around the idea of saving the results of boot-time probes, to speed up boots. This issue might be best solved with that approach, too.

GPSFan, 2023-12-02:
I'm fairly new to Meshtastic, but not to esp32 nor GPS/GNSS... Mostly u-blox. I'm very interested in the unicore 6580, and the wireless tracker seems to be a good avenue to explore its possibilities. I have several t-beams, a t-deck, a wireless tracker and a Heltec lora v0, or v1. My use case is tracking fast off road vehicles where there is poor line of sight and no internet. I've run several in the field test over the summer (US) with promising results, low operator interaction is important, while situational awareness is key.

megabyte2010, 2023-12-02:
Another thing.. it seems that the integrated gps isn't getting data. I tried a external antenna but with no results. The working board that I have is always saying no sats.

thebentern, 2023-12-02:
> Another thing.. it seems that the integrated gps isn't getting data. I tried a external antenna but with no results. The working board that I have is always saying no sats.

All signs point to the device being a lemon.

closed 2023-12-02 by thebentern

GPSFan, 2023-12-02:
with the original firmware from Heltec, did the GNSS work?

caveman99, 2023-12-03:
that GNSS const may be a typo, but i don't see how changing it oin every plave it occurs wil magically make the chip work.

luiscamalmeida, 2023-12-03:
The typo is not everywhere so there is some code that is not executed. I believe it's the configuration to use GPS + GLONASS.

GPSFan, 2023-12-03:
After finding that typo, I dove a bit deeper into the code and didn't think it would really matter, so I didn't follow it up any more.
I think that this commit:
https://github.com/meshtastic/firmware/commit/2544733ad40769b8dec528309ae7d3c2459f8c2a
might have more of an effect on GNSS operation. and is not in a released version yet.
GPS locking can be a funny thing, only with a good view of the sky and a descent antenna can you do real TTFF analysis.