meshtastic/firmware #2906: [Bug]: MQTT Proxy to Client reboot cycle when client app not connected

vinodcombat, 2023-10-24:
### Category

Other

### Hardware

Rak4631

### Firmware Version

v2.2.11.10265aa

### Description

When mqtt client proxy is enabled and device is not connected to client app, the device crashes and reboots whenever it is trying to send a mqtt message through client app.

### Relevant log output

```Shell
INFO  | ??:??:?? 30 [NodeInfoModule] JSON publish message to msh/2/json/LongFast/!ba1a41d3, 206 bytes: {"channel":0,"from":3122282963,"id":1720167599,"payload":{"hardware":9,"id":"!ba1a41d3","lon
DEBUG | ??:??:?? 30 [NodeInfoModule] Sending mqtt message on topic 'msh/2/json/LongFast/!bINFO  | ??:??:?? 35 [NodeInfoModule] Sending our nodeinfo to mesh (wantReplies=1)
INFO  | ??:??:?? 35 [NodeInfoModule] sending owner !ba1a41d3/Meshtastic 41d3/41d3
DEBUG | ??:??:?? 35 [NodeInfoModule] Initial packet id 153613851, numPacketId 4294967295
DEBUG | ??:??:?? 35 [NodeInfoModule] Update DB node 0xba1a41d3, rx_time=0, channel=0
DEBUG | ??:??:?? 35 [NodeInfoModule] handleReceived(LOCAL) (id=0x0927f61d fr=0xd3 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 WANTRESP priority=10)
DEBUG | ??:??:?? 35 [NodeInfoModule] No modules interested in portnum=4, src=LOCAL
DEBUG | ??:??:?? 35 [NodeInfoModule] localSend to channel 0
DEBUG | ??:??:?? 35 [NodeInfoModule] Add packet record (id=0x0927f61d fr=0xd3 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 WANTRESP priority=10)
INFO  | ??:??:?? 35 [NodeInfoModule] Should encrypt MQTT?: 0
DEBUG | ??:??:?? 35 [NodeInfoModule] MQTT onSend - Publishing portnum 4 message
DEBUG | ??:??:?? 35 [NodeInfoModule] MQTT Publish msh/2/c/LongFast/!ba1a41d3, 94 bytes
DEBUG | ??:??:?? 35 [NodeInfoModule] Sending mqtt message on topic 'msh/2/c/LongFast/!ba1a41d3' to client for proxying to server
INFO  | ??:??:?? 35 [NodeInfoModule] serialized json message: {"channel":0,"from":3122282963,"id":153613853,"payload":{"hardware":9,"id":"!ba1a41d3","longname":"Meshtastic 41d3","shortname":"41d3
INFO  | ??:??:?? 35 [NodeInfoModule] JSON publish message to msh/2/json/LongFast/!ba1a41d3, 205 bytes: {"channel":0,"from":3122282963,"id":153613853,"payload":{"hardware":9,"id":"!ba1a41d3","long
DEBUG | ??:??:?? 35 [NodeInfoModule] Sending mqtt message on topic 'msh/2/json/LongFast/!ba1a41d3' to clie��@INFO  | ??:??:?? 1
```

garthvh, 2023-10-24:
Why do you have both json and the client proxy on? Pretty likely that the json is crashing.

closed 2023-11-09 by garthvh

arne182, 2024-01-27:
I also have this rebooting problem. What I was hoping is to send the data to homeassistant unencrypted and json formated from my mobile connection. @garthvh why is this a bad config idea?