meshtastic/firmware #2907: [Bug]: Not sure it's a bug but I cannot get messaging to work on rak4631/19007

feh123, 2023-10-24:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.2.11

### Description

I have a pair of nodes I use for hiking. A rak4631/19007 with OLED, gps (12500) and accelormeter. The second is a rak5631/19007 with rak eink screen. My idea was to get good battery life from the oled device using a short screen on time and start on motion - this node has to have a gps - I cannot guarantee it will have a phone connected. I have got them to message using 2.1.23 after quite struggle - they seem very sensitive to how I install the FW and do the device setup. I described this in Discord. But I wanted to get the improvements that 2.2.x give. I have just set the nodes up again today - starting from a factory reset using the uf2 file. Then I installed 2.2.11 using drag and drop with the uf2 files. I then configure the lora, bluetooth enabled (no pin), wake on motion etc that I need using the meshtastic cli (I have upgraded to the latest version). Everything else is default. If Iook at the nodes all looks fine - position on map (my iphone gives the eink node a position), timestamp etc. However I cannot send a message from the eink to oled node with the iOS app. It never gets acknowledged. I attach the noproto output logs. One from the eink and one from the oled as I send a message from the eink node. If anyone can see why messaging is failing that would be great. Thanks a lot! 
[Terminal Saved rakoled noproto when receiving message runing 2.2.11.txt](https://github.com/meshtastic/firmware/files/13113421/Terminal.Saved.rakoled.noproto.when.receiving.message.runing.2.2.11.txt)
[Terminal Saved rakeink noproto sending message running 2.2.11.txt](https://github.com/meshtastic/firmware/files/13113423/Terminal.Saved.rakeink.noproto.sending.message.running.2.2.11.txt)

GUVWAF, 2023-10-24:
The RAK OLED node is getting multiple position packets **per second** from the phone. Eventually the queue gets full: 
`DEBUG | 08:31:40 1250 NOTE: tophone queue status queue is full, discarding oldest
INFO  | 08:31:40 1250 Telling client we have new packets 2260`

Please check the settings of the position packets.

feh123, 2023-10-24:
Hi @GUVWAF thanks for checking the logs. And I have messaging working again! I connected my iphone to the rakeink node. I couldn't see much wrong with smart position or minimum broadcast interval. I have on the app share location to mesh enabled and that seems to be the problem - its set to five seconds and I tried one minute but that just crashed the app. So I disabled share location to mesh and that seems to be the answer. Messaging is now working. It's odd but the iOS app still won't change items on the nodes - I have to use meshtastic CLI but messaging is working! Thanks a lot for your help.

closed 2023-10-24 by feh123