meshtastic/firmware #2936: [Bug]: Reset to Defaults when Battery Dies

pdxlocations, 2023-11-10:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.2.7 & 2.2.4

### Description

Using RAK4631 with 19007 and 19003, the node will sometimes reset to defaults (region UNSET) when the battery dies during operation. 
Discussion on Discord:
https://discord.com/channels/867578229534359593/1150915176657326130

### Relevant log output

```Shell
//\ E S H T /\ S T / C

DEBUG | ??:??:?? 2 Filesystem files:
DEBUG | ??:??:?? 2 Using analog input 5 for battery level
INFO  | ??:??:?? 2 Scanning for i2c devices...
DEBUG | ??:??:?? 2 Scanning for i2c devices on port 1
INFO  | ??:??:?? 2 No I2C devices found
DEBUG | ??:??:?? 2 acc_info = 0
INFO  | ??:??:?? 2 Meshtastic hwvendor=9, swver=2.2.4.3bcab0e
DEBUG | ??:??:?? 2 Reset reason: 0x0
DEBUG | ??:??:?? 2 Setting random seed 274540578
INFO  | ??:??:?? 2 Initializing NodeDB
INFO  | ??:??:?? 2 No /prefs/db.proto preferences found
INFO  | ??:??:?? 2 Installing default DeviceState
DEBUG | ??:??:?? 2 Initial packet id 306687037, numPacketId 4294967295
INFO  | ??:??:?? 2 No /prefs/config.proto preferences found
INFO  | ??:??:?? 2 Installing default LocalConfig
INFO  | ??:??:?? 2 Setting default channel and radio preferences!
DEBUG | ??:??:?? 2 Expanding short PSK #1
INFO  | ??:??:?? 2 Wanted region 0, using UNSET
INFO  | ??:??:?? 2 No /prefs/module.proto preferences found
INFO  | ??:??:?? 2 Installing default ModuleConfig
INFO  | ??:??:?? 3 No /prefs/channels.proto preferences found
INFO  | ??:??:?? 3 Installing default ChannelFile
INFO  | ??:??:?? 3 No /oem/oem.proto preferences found
INFO  | ??:??:?? 3 Setting default channel and radio preferences!
DEBUG | ??:??:?? 3 Expanding short PSK #1
INFO  | ??:??:?? 3 Wanted region 0, using UNSET
DEBUG | ??:??:?? 3 region=0, NODENUM=0xf54919c2, dbsize=1
INFO  | ??:??:?? 3 Saving /prefs/db.proto
INFO  | ??:??:?? 4 Saving /prefs/channels.proto
DEBUG | ??:??:?? 4 Using MSL altitude model
INFO  | ??:??:?? 4 Setting GPS power=1
DEBUG | ??:??:?? 4 WANT GPS=1
DEBUG | ??:??:?? 4 Probing for GPS at 9600 
INFO  | ??:??:?? 5 Found a UBlox Module using baudrate 9600
DEBUG | ??:??:?? 5 Module Info : 
DEBUG | ??:??:?? 5 Soft version: 1.00 (59842)
DEBUG | ??:??:?? 5 Hard version: 00070000
DEBUG | ??:??:?? 5 Extensions:2
DEBUG | ??:??:?? 5   PROTVER 14.00
DEBUG | ??:??:?? 5   GPS;SBAS;GLO;QZSS
DEBUG | ??:??:?? 5 Protocol Version:14.00
DEBUG | ??:??:?? 5 ProtVer=14
DEBUG | ??:??:?? 5 Setting GPS+SBAS
INFO  | ??:??:?? 5 GNSS configured for GPS+SBAS. Pause for 0.75s before sending next command.
INFO  | ??:??:?? 6 GNSS module configuration saved!
DEBUG | ??:??:?? 6 GxGSA NOT available
DEBUG | ??:??:?? 6 NeighborInfoModule is disabled
INFO  | ??:??:?? 6 No /prefs/ringtone.proto preferences found
INFO  | ??:??:?? 6 Initializing External Notification Module
INFO  | ??:??:?? 6 Using Pin 36 in digital mode
WARN  | ??:??:?? 6 GPS FactoryReset requested
INFO  | ??:??:?? 7 Saving /prefs/db.proto
WARN  | ??:??:?? 8 SX126xInterface(cs=42, irq=47, rst=38, busy=46)
INFO  | ??:??:?? 8 Starting meshradio init...
DEBUG | ??:??:?? 8 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=0, time 231 ms
DEBUG | ??:??:?? 8 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=253, time 3246 ms
INFO  | ??:??:?? 8 Radio freq=906.875, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 8 Set radio: region=UNSET, name=LongFast, config=0, ch=19, power=30
INFO  | ??:??:?? 8 Radio myRegion->freqStart -> myRegion->freqEnd: 902.000000 -> 928.000000 (26.000000 mhz)
INFO  | ??:??:?? 8 Radio myRegion->numChannels: 104 x 250.000kHz
INFO  | ??:??:?? 8 Radio channel_num: 20
INFO  | ??:??:?? 8 Radio frequency: 906.875000
INFO  | ??:??:?? 8 Slot time: 42 msec
INFO  | ??:??:?? 8 Set radio: final power level=22
INFO  | ??:??:?? 8 SX126x init result 0
INFO  | ??:??:?? 8 Frequency set to 906.875000
INFO  | ??:??:?? 8 Bandwidth set to 250.000000
INFO  | ??:??:?? 8 Power output set to 22
DEBUG | ??:??:?? 8 Current limit set to 140.000000
DEBUG | ??:??:?? 8 Current limit set result 0
DEBUG | ??:??:?? 8 SX126X_E22 mode enabled. Setting DIO2 as RF Switch
INFO  | ??:??:?? 8 Set Rx Boosted Gain mode; result: 0
INFO  | ??:??:?? 8 SX1262 Radio init succeeded, using SX1262 radio
INFO  | ??:??:?? 8 Not using Ethernet
DEBUG | ??:??:?? 8 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=237, time 3057 ms
DEBUG | ??:??:?? 8 LoRA bitrate = 77.526985 bytes / sec
INFO  | ??:??:?? 8 PowerFSM init, USB power=1
DEBUG | ??:??:?? 8 Enter state: BOOT
DEBUG | ??:??:?? 8 [Power] Battery: usbPower=0, isCharging=0, batMv=2795, batPct=0
DEBUG | ??:??:?? 8 [Power] Low voltage counter: 1/10
WARN  | ??:??:?? 8 [GPS] Warning, 1 new GPS checksum failures, for a total of 1.
DEBUG | ??:??:?? 8 [GPS] publishing pos@0:2, hasVal=0, GPSlock=0
DEBUG | ??:??:?? 8 [GPS] No GPS lock
DEBUG | ??:??:?? 8 [GPS] onGPSChanged() pos@0, time=0, lat=0, lon=0, alt=0
INFO  | ??:??:?? 8 [GPS] updatePosition LOCAL pos@0, time=0, latI=0, lonI=0, alt=0
DEBUG | ??:??:?? 8 [GPS] Node status update: 1 online, 1 total
INFO  | ??:??:?? 8 [RangeTestModule] Range Test Module - Disabled
INFO  | ??:??:?? 11 [PowerFSM] Loss of power in Powered
INFO  | ??:??:?? 11 [PowerFSM] Loss of power in Powered
INFO  | ??:??:?? 11 [PowerFSM] Initialize the Bluefruit nRF52 module
INFO  | ??:??:?? 12 [PowerFSM] Bluetooth pin set to '123456'
INFO  | ??:??:?? 12 [PowerFSM] Configuring the Device Information Service
INFO  | ??:??:?? 12 [PowerFSM] Configuring the Battery Service
INFO  | ??:??:?? 12 [PowerFSM] Configuring the Mesh bluetooth service
INFO  | ??:??:?? 12 [PowerFSM] Setting up the advertising payload(s)
INFO  | ??:??:?? 12 [PowerFSM] Advertising
DEBUG | ??:??:?? 12 [PowerFSM] Enter state: ON
WARN  | ??:??:?? 12 [GPS] Warning, 1 new GPS checksum failures, for a total of 2.
INFO  | ??:??:?? 13 BLE Connected to BEN's S22
INFO  | ??:??:?? 14 BLE Disconnected, reason = 0x13
INFO  | ??:??:?? 16 BLE Connected to BEN's S22
INFO  | ??:??:?? 19 BLE pairing process started with passkey 123 456
INFO  | ??:??:?? 19 BLE passkey pairing: match_request=0
DEBUG | ??:??:?? 28 [Power] Battery: usbPower=0, isCharging=0, batMv=2807, batPct=0
DEBUG | ??:??:?? 28 [Power] Low voltage counter: 2/10
INFO  | ??:??:?? 36 [NodeInfoModule] Sending our nodeinfo to mesh (wantReplies=1)
INFO  | ??:??:?? 36 [NodeInfoModule] sending owner !f54919c2/Meshtastic 19c2/19c2
DEBUG | ??:??:?? 36 [NodeInfoModule] Update DB node 0xf54919c2, rx_time=0, channel=0
DEBUG | ??:??:?? 36 [NodeInfoModule] handleReceived(LOCAL) (id=0x1247ac40 fr=0xc2 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 WANTRESP priority=10)
DEBUG | ??:??:?? 36 [NodeInfoModule] No modules interested in portnum=4, src=LOCAL
DEBUG | ??:??:?? 36 [NodeInfoModule] localSend to channel 0
DEBUG | ??:??:?? 36 [NodeInfoModule] Add packet record (id=0x1247ac40 fr=0xc2 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 WANTRESP priority=10)
DEBUG | ??:??:?? 36 [NodeInfoModule] Expanding short PSK #1
DEBUG | ??:??:?? 36 [NodeInfoModule] Using AES128 key!
DEBUG | ??:??:?? 36 [NodeInfoModule] nRF52 encrypt fr=f54919c2, num=1247ac40, numBytes=50!
WARN  | ??:??:?? 36 [NodeInfoModule] send - lora tx disable because RegionCode_Unset
```

mrekin, 2023-11-15:
The same behavior with heltec wsl

HookdomPonix, 2023-11-23:
Off the cuff guess:  This could be due to the deepsleep voltage being very close to the "end of my little world" voltage for the 4630 board.

garthvh, 2023-12-03:
@pdxlocations  was this with unprotected 18650 batteries?  Seems to have been the case with several discord users.

pdxlocations, 2023-12-03:
> @pdxlocations was this with unprotected 18650 batteries? Seems to have been the case with several discord users.

No, this was using flat packs.  I found it pretty easy to replicate: drain a battery dead, plug into usb for 30 seconds or so to charge.  Let it die again.  The radio resets maybe 25%? of the time.

garthvh, 2023-12-14:
Seems like we are probably writing to the flash still below the point where we should be.

closed 2023-12-16 by thebentern

cpkirkpat, 2024-09-16:
This just happened to my Heltec V3. The battery died today and when I got home and plugged it in, my settings and keys were gone and its name was changed back to the default. I’m running 2.5.0.

PR #3016 (2023-12-16): Set NRF cpu brownout at 2.4V instead of running down to the limit
Closes #2936 hopefully.
@pdxlocations can you try to repro with the build assets from this change on a RAK?