meshtastic/firmware #2938: [Feature Request]: Add role to NodeInfo or broadcast device metadata

garthvh, 2023-11-14:
### Platform

NRF52, ESP32

### Description

Add the device role to NodeInfo so clients can better display nodes on maps, or broadcast the device metadata packet.

Having all of this functionality locked up so it only works when there is an admin channel is unfortunate.

thebentern, 2023-11-14:
@andrekir @sachaw @ajmcquilkin thoughts? I'm leaning towards adding role to NodeInfo, since it's just one additional byte

ajmcquilkin, 2023-11-14:
Yeah I'm open to adding this to `NodeInfo`, would we also have to add it to `NodeInfoLite` to persist it to the DB?

thebentern, 2023-11-14:
Yes, and I think we'd want it pull in on the want_config.

> Yeah I'm open to adding this to `NodeInfo`, would we also have to add it to `NodeInfoLite` to persist it to the DB?