meshtastic/firmware #2945: Online/Offline maps w/ optional SD card storage

garthvh, 2023-11-17:


garthvh, 2023-11-17:
@mverch67 this was in the design repo cross platform project, I converted it to a firmware issue

RicInNewMexico, 2023-11-24:
Funny, we were just discussing this on the discord yesterday. I didn't even realize an issue had been opened on the topic. 

I came across this which might be useful as a starting point: https://github.com/mryndzionek/esp32_offline_osm/tree/main

I think the most difficult part (from my perspective anyway) will be getting the SD card functioning. I've tried in the past and had issues with the SPI locking.

danumbaman, 2023-12-30:
perhaps the most valuable feature of the t-deck. would love to see this happen in some form of OSM

kovar-ursa, 2024-01-19:
I'd like to be able to log position data locally to a TRACKER device. SD card support is clearly a necessary first step.

This YouTube video shows how the problem was solved for a LilyGo device: https://www.youtube.com/watch?v=hLCFNR3jp8Y

KyleMaas, 2024-04-15:
I would love to see this feature as well.

jgauchia, 2024-04-20:
Hi, 
I'm working in a project with Offline OSM maps (PNG tiles stored in SD card) as well (under development) vector maps.

[https://github.com/jgauchia/IceNav-v3](url)

My programming skills are fairly simple, and it's a personal hobby project. Perhaps something from it could be useful to you."

thebentern, 2024-10-07:
This is being tracked in Device-UI effort. Closing

closed 2024-10-07 by thebentern