meshtastic/firmware #2958: [Board]: Makerfabs SenseLora

samuk, 2023-11-22:
### SOC

ESP32

### Lora IC

RFM92_93W

### Product Link

https://www.makerfabs.com/senselora-industrial-grade-air-monitor.html

### Description

Looking at this for repeater/infrastructure use

It's open-hardware https://github.com/Makerfabs/SenseLoRa-LoRa-Receiver/tree/main/hardware

Includes Solar charging. Fairly affordable. Available to buy off the shelf.

I'll see what leaving the sensors out of the packing list does to the price.

thebentern, 2023-11-25:
Purchased a couple of the solar kits as well as the LoRA receiver to work on support for these. Good find!

samuk, 2023-11-25:
Great! Apparently the price drops to $32 without the sensors..

RicInNewMexico, 2023-11-28:
My dev unit left their warehouse in China on Saturday and should be here sometime in the next two weeks. If we're going to support this unit it only makes sense to add support for it's sensors as well. I went ahead and ordered BH1750 Light Sensor and SGP30 Gas Sensor breakouts from Amazon that should be here Thursday so I can start working on getting those supported and test their functionality on a known working device.

garthvh, 2024-07-03:
Disappinting hardware, closing

closed 2024-07-03 by garthvh