meshtastic/firmware #2965: [Bug]:

pavelw, 2023-11-26:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.2.1.2

### Description

1. static ip, gateway, subnet dns are set
2. config settings yaml file exported via cli command
3. importing settings  through cli: `meshtastic --configure settings.yaml`
EXPECTED: 
config settings imported
ACTUAL
- output ends with error message:

```
Connected to radio
INFO file:node.py beginSettingsTransaction line:535 Telling open a transaction to edit settings
Setting device owner to s5
Setting device owner short to s5
Setting channel url to https://meshtastic.org/e/#redacted-channel-name
Fixing altitude at 141 meters
Fixing latitude at 53.4591455 degrees
Fixing longitude at 15.948628699999999 degrees
Setting device position
Set bluetooth.enabled to True
Set bluetooth.fixed_pin to 123456
Set bluetooth.mode to FIXED_PIN
Set device.node_info_broadcast_secs to 10800
Set device.role to ROUTER_CLIENT
Set device.serial_enabled to True
Set display.screen_on_secs to 600
Set lora.hop_limit to 3
Set lora.modem_preset to VERY_LONG_SLOW
Set lora.override_duty_cycle to True
Set lora.region to EU_868
Set lora.sx126x_rx_boosted_gain to True
Set lora.tx_enabled to True
Set lora.tx_power to 27
Set lora.use_preset to True
Set network.address_mode to STATIC
Aborting due to: Assignment not allowed to message, map, or repeated field "ipv4_config" in protocol message object.
```

additional note:
in yaml, in the mqtt: address is stored in normal human readable form: `192.168.1.1`
the ipv4Config settings are not in this notation:

```
ipv4Config:
      dns: 134744072
      gateway: 16885952
      ip: 671197376
      subnet: 16777215
```

### Relevant log output

```Shell
# start of Meshtastic configure yaml
channel_url: https://meshtastic.org/e/#url-redacted
config:
  bluetooth:
    enabled: true
    fixedPin: 123456
    mode: FIXED_PIN
  device:
    nodeInfoBroadcastSecs: 10800
    role: ROUTER_CLIENT
    serialEnabled: true
  display:
    screenOnSecs: 600
  lora:
    hopLimit: 3
    modemPreset: VERY_LONG_SLOW
    overrideDutyCycle: true
    region: EU_868
    sx126xRxBoostedGain: true
    txEnabled: true
    txPower: 27
    usePreset: true
  network:
    addressMode: STATIC
    ipv4Config:
      dns: 134744072
      gateway: 16885952
      ip: 671197376
      subnet: 16777215
    ntpServer: 0.pool.ntp.org
  position:
    broadcastSmartMinimumDistance: 100
    broadcastSmartMinimumIntervalSecs: 30
    gpsAttemptTime: 900
    gpsEnabled: true
    gpsUpdateInterval: 120
    positionBroadcastSecs: 900
    positionBroadcastSmartEnabled: true
    positionFlags: 779
  power:
    lsSecs: 300
    minWakeSecs: 10
    sdsSecs: 4294967295
    waitBluetoothSecs: 60
location:
  alt: 141
  lat: 43.4691455
  lon: 16.848628699999999
module_config:
  ambientLighting:
    blue: 60
    current: 10
    green: 161
    red: 193
  detectionSensor:
    detectionTriggeredHigh: true
    minimumBroadcastSecs: 45
  mqtt:
    address: 192.168.1.1
    jsonEnabled: true
    root: msh
  neighborInfo:
    updateInterval: 900
  telemetry:
    airQualityInterval: 900
    deviceUpdateInterval: 900
    environmentUpdateInterval: 900
owner: s5
owner_short: s5
```

pavelw, 2023-11-26:
I have tried to import the yaml file again, modyfing the IPs to normal notation, suspecting that it may be the reason for error:

   ipv4Config:
      dns: 8.8.8.8
      gateway: 192.168.1.1
      ip: 192.168.1.40
      subnet: 255.255.255.0

the error message was the same:
```
Set network.address_mode to STATIC
Aborting due to: Assignment not allowed to message, map, or repeated field "ipv4_config" in protocol message object.
```

ajmcquilkin, 2023-11-26:
This sounds like this could be an issue with the Python CLI, not sure this is a firmware issue. I would recommend filing this issue there. Yaml config import isn't something that's natively supported on devices.

pavelw, 2023-11-26:
> This sounds like this could be an issue with the Python CLI, not sure this is a firmware issue. I would recommend filing this issue there. Yaml config import isn't something that's natively supported on devices.
> 
> https://github.com/meshtastic/python

made issue there: https://github.com/meshtastic/python/issues/466

closed 2023-11-26 by garthvh