meshtastic/firmware #2967: [board] "tbeam" doesn't cover all LILYGO® TTGO T-Beam Devices

chunis, 2023-11-28:
TTGO T-Beam Devices contains below 3 "T-Beam" boards:
- T-Beam v1.1
- T-Beam with M8N
- T-Beam with M8N & SX1262

However, there is only one firmware named **tbeam** supports SX1262, Which means another 2 SX1276-based boards are not covered.

So, is there any easy solution for the 2 boards, or should we add another firmware named like **tbeam-sx1276** for them? Thanks.

code8buster, 2023-11-28:
The hardware docs page for tbeams tells you which firmware to install for all of them, which happens to be the same firmware for all the Tbeam variants you've listed.

Indeed, for every variant and combination of GNSS/LoRa radio/PMU on a tbeam, it's already covered.

https://meshtastic.org/docs/hardware/devices/tbeam/

closed 2023-11-28 by code8buster

lora-gateway, 2023-11-29:
@code8buster Thanks for your reply.

I have 3 devices all are "T-Beam with M8N" and all gave me "**Critical fault #3**"; since it uses sx1276 while the latest T-Beam already switches to sx1262, that's why I thought maybe the reason is because "tbeam" firmware doesn't support this old device anymore.

Now that confirmed by you that the firmware still supports it, then maybe the problem is caused by other reasons (I just powered with USB and no battery; or the devices are all broken for unknown reasons).

I have the log (one line says "**RF95 init result -13**") but I guess this is not the right place for pasting it and asking for more help? If so, can you guide me for the right forum or somewhere else for help? 

Thanks again.

chunis, 2023-11-29:
@code8buster Thanks for your reply.

I have 3 devices all are "T-Beam with M8N" and all gave me "**Critical fault #3**"; since it uses sx1276 while the latest T-Beam already switches to sx1262, that's why I thought maybe the reason is because "tbeam" firmware doesn't support this old device anymore.

Now that confirmed by you that the firmware still supports it, then maybe the problem is caused by other reasons (I just powered with USB and no battery; or the devices are all broken for unknown reasons).

I have the log (one line says "**RF95 init result -13**") but I guess this is not the right place for pasting it and asking for more help? If so, can you guide me for the right forum or somewhere else for help?

Thanks again.