meshtastic/firmware #2978: [Bug]: False positives in detection module with RAK12013 radar module

Kollensperger, 2023-12-01:
### Category

Hardware Compatibility

### Hardware

Rak4631

### Firmware Version

2.2.14

### Description

**Setup:**
Mounted RAK12013 module on RAK19007 base board with RAK4631 core.
Enabled Detection Sensor Config in iOS on GPIO pin 21 and selected pullup resistor active (seems theres a 100 Ohm resistor on the OUT line of the 12013 module). Min time between broadcasts set to one minute.

**Result:**
Detection module seems to work, however there seems to be quite some interference, possibly from the LoRa signals themselves. False positives increase markedly the more nodes are in the mesh. I assume this is because more background messages are sent. There are some articles both on the RAK forum as well as discussing many false positives in noisy RF environments for the RCWL-9196 chip.

**Potential Solutions:**
Suggestions range from installing caps on the powerline to adding some kind of debouncing / filtering code to ensure detection is an actual event. I'm trying to get one of the RAK sensor extention cables to see if that might help, but it would be nice to be able to mount the sensor directly on the  base board, provided someone can come up with a filter to prevent the false positives.

garthvh, 2023-12-01:
Do you have it in an enclosure outside? Not seeing a lot of false positives.

Kollensperger, 2023-12-01:
Yes, its in one of Tropho's 3D printed enclosures. I've tried it indoors as well as outdoors (in a waterproof enclosure and placed on lawn on virgin snow on a day with no wind to avoid twigs triggering the radar and ~10 m from house to reduce any other signals.  No paw prints in the snow so pets can be excluded.
The very best I can get, with just two devices, is about one false positive every ten minutes, and then suddenly it'll get a many one after the other. The times I had several nodes on the mesh, the detection frequency dropped to watever the minimum time between broadcasts is. Hence my thoughts about interference - also see the second comment by Bernd [here at the RAK forum](https://forum.rakwireless.com/t/phantom-interrupts/4799) - I've also seen mentions of other approaches as referenced in my previous post.

Have you run your radar setup for an extended period of time? what frequency of false positives do you see?

garthvh, 2023-12-01:
Yeah I would use a larger enclosure without so much stuff right next to the sensor.

garthvh, 2023-12-01:
[image]

garthvh, 2023-12-01:
The latest version of the app in TestFlight and this firmware version have some detection sensor updates

https://github.com/meshtastic/artifacts/commit/59051576f7211c4b3e3f0e4386a7894f24b2b762

closed 2023-12-01 by garthvh

thebentern, 2023-12-02:
This fix will be bundled in the upcoming alpha release.