meshtastic/firmware #2990: [Feature Request]: Long-term INA___ current/voltage sampling and averaging

KodinLanewave, 2023-12-05:
### Platform

NRF52, ESP32

### Description

Currently INA219 devices sample once to provide an instantaneous reading of current and voltage for telemetry.  The Adafruit INA219 library (I haven't checked the other INA variants for comparison yet), defaults to 128 x 12-bit bus samples averaged together over a total read/sample time of 69ms.  I'd like to see longer term averaging if this won't cause too large a performance hit to various platforms.  Maybe some sort of configurable sample frequency and total number of samples to average?  IE: maybe once a second over 1 minute?  As an example.  The application for this is that I have a node with battery and solar, and I'm running i2c current and voltage sensor(s) to watch current and voltage in/out of the battery as well as off the solar panel.  With sleep modes being what they are, a meshtastic node consumes power inconsistently, and having a longer-term average is much more useful for getting a better picture of power consumption or solar charging performance.  Thanks!

crypto-smoke, 2023-12-05:
I'd like to see more of a sum than an average. It would probably need tweaks to the telemetry packet, or perhaps we just add a new field "power" that is calculated as `amps * volts * seconds since last reading` and expressed as milliwatt-seconds over the mesh. Or watt hours but that will make for some small numbers.

mc-hamster, 2023-12-06:
> ### Platform
> NRF52, ESP32
>
> ### Description
> Currently INA219 devices sample once to provide an instantaneous reading of current and voltage for telemetry. The Adafruit INA219 library (I haven't [...]

Hmmm... This may increase battery usage since the CPU will need to be left on. This could be done as an opt-in feature, but should not be implemented as a general purpose feature.

I think the solution should be on the client side rather on the node. On the client (phone, web, where ever) it can be configured to average the readings over the longer period and it won't affect the battery life of the node.

mc-hamster, 2023-12-06:
> I'd like to see more of a sum than an average. It would probably need tweaks to the telemetry packet, or perhaps we just add a new field "power" that is calculated as `amps * volts * seconds since [...]

I'd advice against adding more to the telemetry packet that has to be included by default as it'll use up airtime for people who don't need this information.

KodinLanewave, 2023-12-06:
Would this affect battery life on NRF52 platforms? I wasn't aware that those chips went into any sort of sleep phase...

KodinLanewave, 2023-12-06:
For what it's worth, the current readings of a node are too point-in-time at the moment due to the nature of polling vs. when the node itself might be using power.  The G1 Station is a good example of this:
[image]
as you can see, it's luck of the draw as to whether you sample during a window where the radio is powered up.  This doesn't give a great picture if you're trying to monitor power consumption of the node over time.

garthvh, 2023-12-16:
The summing needs to be done on the client side, this is how all of the telemetry works.

closed 2023-12-16 by garthvh