meshtastic/firmware #3005: [Bug]: TDeck audio rebooting issue

jharvey, 2023-12-09:
### Category

Hardware Compatibility

### Hardware

T-Deck

### Firmware Version

V2.2.14.57542ce beta

### Description

Under Module --> Audio if you enable the Codec it will go into endless rebooting making the TDeck unusable. I was able to reprogram it by booting into the bootloader, and flashing a fresh copy of the Meshtastic firmware. Then I was able to reconfigure. This is likely caused by a GPIO pin shorting when this is enabled. The speaker on this is probably not implemented in software yet. See zone C3 for the speaker connection in the below schematic. Looks like I2S driver needs to be implemented to enable the audio. 
https://github.com/Xinyuan-LilyGO/T-Deck/blob/master/schematic/schematic.pdf

thebentern, 2023-12-10:
Sorry, audio is currently for 2.4GHz radio devices only. We'd like to revisit a version of this at some point in the force of voice memos, but PTT audio over 915MHz lora is not good enough in terms of bandwidth to accommodate reasonable quality.

closed 2023-12-10 by thebentern

jharvey, 2023-12-10:
Could the audio be enabled enough for making beep and alerts? It hadn't occurred to me that the codec was used for voice compression. I was trying to get alert beeps on the T-Deck. 

If the codec is enabled, can it be made so that you don't need to re-flash to get it back?