meshtastic/firmware #3006: [Feature Request]: T-Deck canned messages last menu option

jharvey, 2023-12-09:
### Platform

ESP32

### Description

Create return to previous menu option on the canned messages of the T-Deck 

On the T-Deck when you enable canned messages, if you enter the canned menu list, you have to wait about 20 seconds for it to automatically return to the previous menu. It would be helpful if you could some how return to the last menu on command. Perhaps by default the bottom item in the list could be a selectable option to return to last menu. This way when the roller is rolled slightly down, you don't have to wait long periods of time to get back to the menu which you were originally in. When in this canned message menu, the keyboard is mostly not used. Perhaps when in this menu the up arrow could allow someone to return to the previous menu.

garthvh, 2023-12-09:
Duplicate of https://github.com/meshtastic/firmware/issues/2042

closed 2023-12-09 by garthvh

mverch67, 2024-02-17:
A long press on the screen gets you out.