meshtastic/firmware #3008: [Feature Request]: Add SAMD51 MCU support

tropho23, 2023-12-10:
### Platform

NRF52

### Description

Note 1: I chose NRF52 platform since it's a requirement to submit a feature request, but this is *not* for NRF52.
Note 2: SAMD51 is not the same as STM32

Request adding support for the SAMD51 MCU, as it would open up many dev boards for use with Meshtastic.  SAMD51 has 256KB of RAM and up to 1MB flash memory, same as nrf52840.  Both SAMD51 and nrf5284 use ARM® Cortex®-M4 32-bit processor with FPU, though SAMD51 can be clocked up to 200MHz vs. nrf52840 64MHz.  We run all MCUs at 40MHz anyways for Meshtastic.

It looks like it is just as capable for Meshtastic as nrf52840, but I have no idea if/how it can be supported.  I do know PlatformIO supports it: https://docs.platformio.org/en/latest/boards/atmelsam/sparkfun_samd51_thing_plus.htm.

Regarding filesystem(s), I've found evidence on the internet that claim SAM D5x/E5x Flash can only be written when executing from SRAM" so SRAM may need to be present.  However, there is a FlashStorage_SAMD library in Arduino that provides a way to store and retrieve user's data using the non-volatile flash memory of SAMD21/SAMD51, which is useful if the EEPROM is not available or too small.  No idea if this is relevant to Meshtastic but it does prove there is addressable flash storage for filesystem use.

thebentern, 2023-12-14:
@tropho23 closing this after our discussion the other day

closed 2023-12-14 by thebentern

hayschan, 2025-01-02:
I saw this issue status as completed. Is this supported or not?

If supported, it would be fun to run Meshtastic on Seeed's Wio Terminal.

[image]

[image]