meshtastic/firmware #3009: [Bug]: Admin Command Not Working on Multi-Hop Nodes

pdxlocations, 2023-12-11:
### Category

Other

### Hardware

Rak4631

### Firmware Version

Multiple up to 2.2.14

### Description

Admin Channel is no longer working on nodes more than one hop away.  My network is a mix of firmware.  I've tried reverted the sending node back to 2.2.9 with no improvements.

Possibly related to https://github.com/meshtastic/firmware/issues/3003 ?

andrekir, 2023-12-13:
@pdxlocations if your results were using the Android app, please try release [2.2.16](https://github.com/meshtastic/Meshtastic-Android/releases/tag/2.2.16)

pdxlocations, 2023-12-14:
> @pdxlocations if your results were using the Android app, please try release [2.2.16](https://github.com/meshtastic/Meshtastic-Android/releases/tag/2.2.16)

Yup, all fixed.  Thank you @andrekir !

closed 2023-12-14 by pdxlocations