meshtastic/firmware #3052: [Bug]: Build Failure - two issues

rabarar, 2024-01-04:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

latest pull

### Description

The firmware doesn't build: the  `c_cpp_properties.json` generated file is corrupted with escape `\` characters for one of the filenames

`"~/.platformio/packages/framework-arduinoststm32/variants/STM32WLxx/WL54CCU_WL55CCU_WLE4C\(8-B-C\)U_WLE5C\(8-B-C\)U",`

The build fails with errors:

```
In file included from src/DisplayFormatters.h:2,
                 from src/DisplayFormatters.cpp:1:
src/mesh/NodeDB.h:114:92: error: 'uint' has not been declared
  114 |     void initConfigIntervals(), initModuleConfigIntervals(), resetNodes(), removeNodeByNum(uint nodeNum);
      |                                                                                            ^~~~
In file included from src/GPSStatus.h:2,
                 from src/mesh/MeshService.h:7,
                 from src/PowerFSM.cpp:11:
src/mesh/NodeDB.h:114:92: error: 'uint' has not been declared
  114 |     void initConfigIntervals(), initModuleConfigIntervals(), resetNodes(), removeNodeByNum(uint nodeNum);
      |                                                                                            ^~~~
In file included from src/Power.cpp:14:
src/mesh/NodeDB.h:114:92: error: 'uint' has not been declared
  114 |     void initConfigIntervals(), initModuleConfigIntervals(), resetNodes(), removeNodeByNum(uint nodeNum);
      |                                                                                            ^~~~
In file included from src/SerialConsole.cpp:2:
src/mesh/NodeDB.h:114:92: error: 'uint' has not been declared
  114 |     void initConfigIntervals(), initModuleConfigIntervals(), resetNodes(), removeNodeByNum(uint nodeNum);
      |                                                                                            ^~~~
In file included from src/airtime.cpp:2:
src/mesh/NodeDB.h:114:92: error: 'uint' has not been declared
  114 |     void initConfigIntervals(), initModuleConfigIntervals(), resetNodes(), removeNodeByNum(uint nodeNum);
      |                                                                                            ^~~~
src/Power.cpp: In member function 'uint16_t AnalogBatteryLevel::getINAVoltage()':
src/Power.cpp:290:13: error: 'nodeTelemetrySensorsMap' was not declared in this scope
  290 |         if (nodeTelemetrySensorsMap[meshtastic_TelemetrySensorType_INA219].first == config.power.device_battery_ina_address) {
      |             ^~~~~~~~~~~~~~~~~~~~~~~
*** [.pio/build/wio-e5/src/DisplayFormatters.cpp.o] Error 1
src/Power.cpp:291:20: error: 'ina219Sensor' was not declared in this scope
  291 |             return ina219Sensor.getBusVoltageMv();
      |                    ^~~~~~~~~~~~
*** [.pio/build/wio-e5/src/SerialConsole.cpp.o] Error 1
*** [.pio/build/wio-e5/src/PowerFSM.cpp.o] Error 1
*** [.pio/build/wio-e5/src/airtime.cpp.o] Error 1
src/Power.cpp:294:20: error: 'ina260Sensor' was not declared in this scope
  294 |             return ina260Sensor.getBusVoltageMv();
      |                    ^~~~~~~~~~~~
In file included from src/GPSStatus.h:2,
                 from src/main.h:3,
                 from src/concurrency/NotifiedWorkerThread.cpp:3:
src/mesh/NodeDB.h:114:92: error: 'uint' has not been declared
  114 |     void initConfigIntervals(), initModuleConfigIntervals(), resetNodes(), removeNodeByNum(uint nodeNum);
      |                                                                                            ^~~~
src/Power.cpp:297:20: error: 'ina3221Sensor' was not declared in this scope
  297 |             return ina3221Sensor.getBusVoltageMv();
      |                    ^~~~~~~~~~~~~
src/Power.cpp: In member function 'bool AnalogBatteryLevel::hasINA()':
src/Power.cpp:307:13: error: 'nodeTelemetrySensorsMap' was not declared in this scope
  307 |         if (nodeTelemetrySensorsMap[meshtastic_TelemetrySensorType_INA219].first == config.power.device_battery_ina_address) {
[... 130 lines cut ...]
                 from src/mesh/generated/meshtastic/admin.pb.h:10,
                 from src/mesh/mesh-pb-constants.h:3,
                 from src/mesh/PhoneAPI.h:4,
                 from src/mesh/StreamAPI.h:3,
                 from src/SerialConsole.h:4,
                 from src/DebugConfiguration.h:26,
                 from src/configuration.h:207,
                 from src/mesh/NodeDB.cpp:1:
src/mesh/NodeDB.cpp: In member function 'void NodeDB::init()':
src/mesh/generated/meshtastic/module_config.pb.h:463:55: warning: narrowing conversion of '(const char*)""' from 'const char*' to 'bool' [-Wnarrowing]
  463 | #define meshtastic_RemoteHardwarePin_init_default {0, "", _meshtastic_RemoteHardwarePinType_MIN}
      |                                                       ^~
src/mesh/NodeDB.cpp:455:55: note: in expansion of macro 'meshtastic_RemoteHardwarePin_init_default'
  455 |         meshtastic_NodeRemoteHardwarePin empty[12] = {meshtastic_RemoteHardwarePin_init_default};
      |                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pio/build/wio-e5/src/mesh/PhoneAPI.cpp.o] Error 1
*** [.pio/build/wio-e5/src/mesh/MeshService.cpp.o] Error 1
*** [.pio/build/wio-e5/src/mesh/NodeDB.cpp.o] Error 1
*** [.pio/build/wio-e5/src/main.cpp.o] Error 1
======================================================================= [FAILED] Took 4.00 seconds =======================================================================

Environment    Status    Duration
-------------  --------  ------------
wio-e5         FAILED    00:00:04.004
```

GUVWAF, 2024-01-04:
The `wio-e5` environment is still a work-in-progress, see #2960. 
Do you need specifically this environment? While this is set as the `default_envs` in `platformio.ini`, you can select the appropriate environment for your device.

GUVWAF, 2024-01-04:
For the `wio-e5` environment or in general? The firmware should build successfully for all supported devices on the `master` branch as can be seen by the CI.
As I said, the `wio-e5` is a work-in-progress in PR 2960 using branch `stm32wl`.

rabarar, 2024-01-04:
I’m confused then. I simply cloned the repo master branch, made no modifications, and attempted to build it. So I don’t see how your CI is successful when I can’t replicate a clean, unmodified build.

jp-bennett, 2024-01-04:
You need to pick your build target before building. That said, it's a problem if we're building a WIP target by default.

jp-bennett, 2024-01-04:
We've switched the default build target back to the T-beam. This should make the firmware build cleanly for that device, without setting the target, so closing this issue. Thanks for pointing out this unintended behavior!

closed 2024-01-04 by jp-bennett

rabarar, 2024-01-04:
My pleasure!