meshtastic/firmware #3056: Encrypted packets sent from MQTT won't decrypt

dstewartgo, 2024-01-04:
On the MQTT Connect project (https://github.com/pdxlocations/Meshtastic-MQTT-Connect) we're trying to send encrypted packets to devices, and they are getting ignored (although they decrypt just fine in the MQTT code itself, and the MQTT code can also receive encrypted packets from the device and decode them fine).

The problem appears to be that in device MQTT firmware code, the channel is only set if the message was decoded:

https://github.com/meshtastic/firmware/blob/e3c768bf10b18746f98b46f2df7cac6efae8761b/src/mqtt/MQTT.cpp#L146C1-L146C1

And therefore, when "perhapsDecode" is called, the "decryptForHash" call will fail:

https://github.com/meshtastic/firmware/blob/e3c768bf10b18746f98b46f2df7cac6efae8761b/src/mesh/Router.cpp#L316

andrekir, 2024-01-05:
`channel` in encrypted packets uses the channel name+PSK hash, not channel index:

https://github.com/meshtastic/protobufs/blob/4a00cafd3e87368e7a113e37374e79ada3f0595a/meshtastic/mesh.proto#L869-L875

```
   * A particular node might know only a subset of channels in use on the mesh.
   * Therefore channel_index is inherently a local concept and meaningless to send between nodes.
   * Very briefly, while sending and receiving deep inside the device Router code, this field instead
   * contains the 'channel hash' instead of the index.
   * This 'trick' is only used while the payload_variant is an 'encrypted'.
   */
  uint32 channel = 3;
```

dstewartgo, 2024-01-05:
Thanks, so it looks like MQTT Connect will need to set this properly -- doing so by hand looks like it is working.

closed 2024-01-05 by dstewartgo