meshtastic/firmware #3058: [Board]: WiFi LoRa 32 (V3)

HORUS-UA, 2024-01-05:
### SOC

ESP32

### Lora IC

SX1262

### Product Link

https://heltec.org/project/wifi-lora-32-v3/

### Description

WiFi LoRa 32 is a classic IoT dev-board designed & produced by Heltec Automation. Since its launch in 2017, it has been loved by developers and makers. The newly launched V3 version has the same pin sequence as the V2 version and retains Wi-Fi, BLE, LoRa, OLED display, and other functions.

It is the best option for smart cities, farms, homes, industrial control, house security, wireless meter reading, and IoT developers.

https://docs.platformio.org/en/latest//boards/espressif32/heltec_wifi_lora_32_V3.html

HORUS-UA, 2024-01-05:
This version is not in the list of configurations for [Meshtastic Firmware](https://github.com/meshtastic/firmware)

[platformio]
default_envs = tbeam
;default_envs = pico
;default_envs = tbeam-s3-core
;default_envs = tbeam0.7
;default_envs = heltec-v1
;default_envs = heltec-v2_0
;default_envs = heltec-v2_1
;default_envs = heltec-wireless-tracker
;default_envs = tlora-v1
;default_envs = tlora_v1_3
;default_envs = tlora-v2
;default_envs = tlora-v2-1-1_6
;default_envs = tlora-t3s3-v1
;default_envs = lora-relay-v1 # nrf board
;default_envs = t-echo
;default_envs = nrf52840dk-geeksville
;default_envs = native # lora-relay-v1 # nrf52840dk-geeksville # linux # or if you'd like to change the default to something like lora-relay-v1 put that here
;default_envs = nano-g1
;default_envs = pca10059_diy_eink
;default_envs = meshtastic-diy-v1
;default_envs = meshtastic-diy-v1_1
;default_envs = meshtastic-dr-dev
;default_envs = m5stack-coreink
;default_envs = rak4631
;default_envs = rak10701
;default_envs = wio-e5

HORUS-UA, 2024-01-05:
That is, I can simply add a line to the list of configurations  ?

default_envs = heltec-v3

GUVWAF, 2024-01-06:
I hope it works now for you.
Since the Heltec V3 is already supported, I'm closing this.

closed 2024-01-06 by GUVWAF

HORUS-UA, 2024-01-06:
Yes, it works
Thank you very much for the advice