meshtastic/firmware #3059: [Bug]: T-Echo changing any position settings results in "no GPS device"

rolf-tx, 2024-01-06:
### Category

Hardware Compatibility, Other

### Hardware

T-Echo

### Firmware Version

2.2.17

### Description

For T-Echo changing any parameter in the Position settings tab results in the GPS not being detected "no GPS device".
Changing the GPS pin definitions from Unset to 41, 40, 37 seems to get the device detected some of the time but unable to get satellite lock.

Mictronics, 2024-01-06:
I can't confirm. I have 4 T-Echo with the actual master branch code running, including changes from 2.2.17. All device get the GPS module detected and having a position lock within a few minutes. Changing GPS setting do not have any other effect than the expected change. And I experiment a lot here at the moment.

Mictronics, 2024-01-06:
Tried also the pre-compiled 2.2.17 on one Echo device. GPS position acquired in less than 2 min on the window sill.

rolf-tx, 2024-01-06:
@Mictronics
Thanks for checking.
I’m using the iOS app, are you using iOS or android?

Mictronics, 2024-01-06:
Android.

thebentern, 2024-01-06:
Can you post a meshtastic --info dump of your position config after saving on the iOS app? That would help us pin down what's getting unset

rolf-tx, 2024-01-06:
@thebentern 
I have attached a bad and a good info dump and start log. 
Note that sometimes I had to do a physical hard reset of the device to trigger "no GPS" after setting Timestamp=on, the soft reset still had the GPS functioning.

[info-2.2.17.bad.txt](https://github.com/meshtastic/firmware/files/13850357/info-2.2.17.bad.txt)
[info-2.2.17.ok.txt](https://github.com/meshtastic/firmware/files/13850359/info-2.2.17.ok.txt)
[log.2.2.17.bad.txt](https://github.com/meshtastic/firmware/files/13850535/log.2.2.17.bad.txt)
[log.2.2.17.ok.txt](https://github.com/meshtastic/firmware/files/13850536/log.2.2.17.ok.txt)

Mictronics, 2024-01-06:
The bad config is missing:
```
"gpsUpdateInterval": 120,
"gpsAttemptTime": 900,
```
I can reproduce this bug when I set in Android app the `GPS Update Interval`and `Fix attempt duration` manually to zero (0). The Echo then shows "No GPS Module".

rolf-tx, 2024-01-06:
@Mictronics Thanks for the quick fix. I built it and it fixed the issue.

Mictronics, 2024-01-07:
@rolf-tx I hope you don't build my branches. These are modified on special purpose for personal use.

rolf-tx, 2024-01-07:
@Mictronics No, not normally, just for this case. 

Any idea when the changes will be merged in bleeding?

Mictronics, 2024-01-07:
There is work in progress, see PR #3064

closed 2024-01-07 by rolf-tx