meshtastic/firmware #3060: [Bug]: Devices wont boot from USB power

GaryR-12, 2024-01-06:
### Category

Other

### Hardware

T-Echo

### Firmware Version

2.2.16

### Description

Both my T-Echo and T-Beam, after the battery dies, wont boot back up when USB cable is plugged in. Started after updating to 2.2.16

thebentern, 2024-01-06:
We do not control this in the firmware.

closed 2024-01-06 by thebentern

GaryR-12, 2024-01-14:
@thebentern  Not convinced it is a hardware problem as it seems to have started after 2.2. Also the T-echo screen updates to the boot screen showing the region and version info and then hangs with a solid blue light. Does this not imply that the firmware has begun execution? The USB can by plugged and unplugged repeatedly but solid blue light stays and no boot. Only way to recover is the reset button. I've had this T-echo for a year and never seen this behavior before.