meshtastic/firmware #3061: [Bug]: User button unreliable on Liligo TTGO T3S3

LanWolf, 2024-01-06:
### Category

Other

### Hardware

Other

### Firmware Version

2.2.17

### Description

User button singe press sometimes not wake the screen, and does not  advance to next screen. 
Sometimes a double or triple click wakes the screen
long press to shutdown seems ok.
After a fresh flash the user button works okay for some time, then stops suddenly working correctly.

GUVWAF, 2024-01-06:
Can you share your configuration with `meshtastic --info` from the CLI (after removing private details)?

dmthepm, 2024-01-07:
Im actually finding my hold to reset button to be the least reliable. I often have to do it twice.

LanWolf, 2024-01-07:
> Can you share your configuration with `meshtastic --info` from the CLI (after removing private details)?

`Connected to radio

Owner: RickZ-Go-d70c (RZGo)
My info: { "myNodeNum": ........ , "rebootCount": 42, "minAppVersion": 30200 }
Metadata: { "firmwareVersion": "2.2.17.dbac2b1", "deviceStateVersion": 22, "canShutdown": true, "hasWifi": true, "hasBluetooth": true, "positionFlags": 939, "hwModel": "TLORA_T3_S3" }

Preferences: { "device": { "serialEnabled": true, "nodeInfoBroadcastSecs": 10800 }, "position": { "positionBroadcastSecs": 900, "positionBroadcastSmartEnabled": true, "positionFlags": 939, "broadcastSmartMinimumDistance": 100, "broadcastSmartMinimumIntervalSecs": 30 }, "power": { "waitBluetoothSecs": 60, "sdsSecs": 4294967295, "lsSecs": 300, "minWakeSecs": 10 }, "network": { "ntpServer": "0.pool.ntp.org" }, "display": { "screenOnSecs": 60, "displaymode": "INVERTED" }, "lora": { "usePreset": true, "region": "EU_868", "hopLimit": 3, "txEnabled": true, "txPower": 27, "sx126xRxBoostedGain": true }, "bluetooth": { "enabled": true, "fixedPin": 123456 } }

Module preferences: { "mqtt": { "address": "mqtt.meshtastic.org", "username": "meshdev", "password": "large4cats", "encryptionEnabled": true, "root": "msh" }, "serial": {}, "externalNotification": {}, "rangeTest": {}, "telemetry": { "deviceUpdateInterval": 900, "environmentUpdateInterval": 900, "airQualityInterval": 900 }, "cannedMessage": { "enabled": true, "allowInputSource": "_any", "sendBell": true }, "audio": {}, "remoteHardware": {}, "neighborInfo": { "updateInterval": 900 }, "ambientLighting": { "current": 10, "red": 223, "green": 215, "blue": 12 }, "detectionSensor": { "minimumBroadcastSecs": 45, "detectionTriggeredHigh": true } }

Channels:
  PRIMARY psk=default { "psk": "AQ==", "name": "Broadcast" }
`

GUVWAF, 2024-01-07:
@LanWolf Thanks. I see you have the CannedMessage module enabled. Maybe this influences the button presses. 
Do you have any input broker connected to the device?

LanWolf, 2024-01-07:
You are totally correct. Turned canned msg off and the button for screen change works again.. turn it on and the button scren change stops working again 
I have a m5stack cardkb 1.1 which I sometimes use with the TTGO, because it has a Qwiic plug, and can easily be attached or detached that way.

GUVWAF, 2024-01-07:
Good to know! 

> Hope there can be a fix so this can work both ?

Try setting `canned_message.allow_input_source` to "cardKB".

LanWolf, 2024-01-07:
tried it but sadly no user button screen change
 "cannedMessage": { "enabled": true, "allowInputSource": "cardkb", "sendBell": true },

GUVWAF, 2024-01-07:
@LanWolf I think I found a weird edge case in the code that caused this to happen. It should be fixed with PR #3067.

GUVWAF, 2024-01-08:
@LanWolf Can you confirm if either the PR or the setting works for you?

LanWolf, 2024-01-08:
I can confim the `canned_message.inputbroker_pin_press` set to 1 works,
flashing is a litte tricky at the moment beacuse of other issue.

GUVWAF, 2024-01-08:
Okay, thanks. Then I'm convinced the PR works as well.

PR #3067 (2024-01-10): Allow button press if CannedMessage `updown1` or `rotary1` are not enabled
`BUTTON_PIN` may be 0 (e.g. on LoRa T3S3), which is equal to `inputbroker_pin_press` by default. `inputbroker_pin_press` is only used when either `updown1` or `rotary1` is enabled.

Should fix #3061.
closed 2024-01-10 by thebentern