meshtastic/firmware #3066: [Bug]: T-Beam v1.2 running in router mode crashes and enters endless reboot cycle

attermann, 2024-01-07:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.2.160-beta

### Description

Meshtastic will run without issue for several hours, but then crashes at some point and enters a loop of endless crash/reboot where the oled only shows the splash screen.

While I can't say for sure, this issue seems to have emerged when I changing configuration of this device from router+client to just router mode. I will be reverting to router+client to see if that improves things.

I also wasn't able to obtain logs from the first crash starts the crash/reboot loop yet, but the below logs are repeated on each crash/reboot cycle that follows.

Resetting the device via the RST button clears the issue and the device operates normally again until the next time the issue appears.

Will update this issue as more details are obtained.

### Relevant log output

```Shell
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
E (864) esp_core_dump_flash: No core dum����ѥѥ���found!
E (865) esp_core_dump_flash: No core dump partition found!
[    12][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   464][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
��@INFO  | ??:??:?? 0 

//\ E S H T /\ S T / C

INFO  | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=reset
DEBUG | ??:??:?? 0 Filesystem files (499712/1048576 Bytes):
DEBUG | ??:??:?? 0  /prefs/channels.proto (53 Bytes)
DEBUG | ??:??:?? 0  /prefs/config.proto (88 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (1493 Bytes)
DEBUG | ??:??:?? 0  /static/.gitkeep (0 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_Black.svg.gz (602 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_White.svg.gz (610 Bytes)
DEBUG | ??:??:?? 0  /static/apple-touch-icon.png.gz (3164 Bytes)
DEBUG | ??:??:?? 0  /static/favicon.ico.gz (2270 Bytes)
DEBUG | ??:??:?? 0  /static/icon.svg.gz (852 Bytes)
DEBUG | ??:??:?? 0  /static/index-H3yz6dTg.js.gz (348 Bytes)
DEBUG | ??:??:?? 0  /static/index-cnVnppgi.css.gz (17208 Bytes)
DEBUG | ??:??:?? 0  /static/index-t3MYXjf0.js.gz (206869 Bytes)
DEBUG | ??:??:?? 1  /static/index.html.gz (545 Bytes)
DEBUG | ??:??:?? 1  /static/maplibre-gl-pS6olApQ.js.gz (208178 Bytes)
DEBUG | ??:??:?? 1  /static/robots.txt.gz (42 Bytes)
DEBUG | ??:??:?? 1  /static/site.webmanifest.gz (197 Bytes)
[  1078][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
[  1079][I][XPowersCommon.tpp:248] begin(): SDA:21 SCL:22
[  1084][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
INFO  | ??:??:?? 1 AXP2101 PMU init succeeded, using AXP2101 PMU
[  1116][D][XPowersAXP2101.tpp:2791] setInterruptImpl(): DISABLE - HEX:0xffffffff BIN:
DEBUG | ??:??:?? 1 =======================================================================
DEBUG | ??:??:?? 1 DC1  : +   Voltage:3300 mV 
DEBUG | ??:??:?? 1 DC2  : -   Voltage:900 mV 
DEBUG | ??:??:?? 1 DC3  : -   Voltage:900 mV 
DEBUG | ??:??:?? 1 DC4  : -   Voltage:1100 mV 
DEBUG | ??:??:?? 1 ALDO1: -   Voltage:1800 mV 
DEBUG | ??:??:?? 1 ALDO2: +   Voltage:3300 mV 
DEBUG | ??:??:?? 1 ALDO3: +   Voltage:3300 mV 
DEBUG | ??:??:?? 1 ALDO4: -   Voltage:2900 mV 
DEBUG | ??:??:?? 1 BLDO1: -   Voltage:1800 mV 
DEBUG | ??:??:?? 1 BLDO2: -   Voltage:2800 mV 
DEBUG | ??:??:?? 1 =======================================================================
DEBUG | ??:??:?? 1 Battery: usbPower=1, isCharging=0, batMv=4163, batPct=100
INFO  | ??:??:?? 1 Scanning for i2c devices...
[  1185][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
DEBUG | ??:??:?? 1 Scanning for i2c devices on port 1
DEBUG | ??:??:?? 1 I2C device found at address 0x34
INFO  | ??:??:?? 1 axp192/axp2101 PMU found
DEBUG | ??:??:?? 1 I2C device found at address 0x3c
INFO  | ??:??:?? 1 ssd1306 display found
INFO  | ??:??:?? 1 ssd1306 display found
DEBUG | ??:??:?? 1 0x4 subtype probed in 2 tries 
INFO  | ??:??:?? 1 2 I2C devices found
DEBUG | ??:??:?? 1 acc_info = 0
INFO  | ??:??:?? 1 Meshtastic hwvendor=4, swver=2.2.16.1c6acfd
DEBUG | ??:??:?? 1 Setting random seed 2108203789
DEBUG | ??:??:?? 1 Total heap: 185496
DEBUG | ??:??:?? 1 Free heap: 151108
DEBUG | ??:??:?? 1 Total PSRAM: 4192107
DEBUG | ??:??:?? 1 Free PSRAM: 4191847
DEBUG | ??:??:?? 1 NVS: UsedEntries 90, FreeEntries 540, AllEntries 630, NameSpaces 3
DEBUG | ??:??:?? 1 Setup Preferences in Flash Storage
DEBUG | ??:??:?? 1 Number of Device Reboots: 21795
DEBUG | ??:??:?? 1 OTA firmware version 0.2.1.2fbe7fc
INFO  | ??:??:?? 1 Initializing NodeDB
INFO  | ??:??:?? 1 Loading /prefs/db.proto
INFO  | ??:??:?? 1 Loaded saved devicestate version 22
INFO  | ??:??:?? 1 Loading /prefs/config.proto
INFO  | ??:??:?? 1 Loaded saved config version 22
[  1362][E][vfs_api.cpp:105] open(): /littlefs/prefs/module.proto does not exist, no permits for creation
INFO  | ??:??:?? 1 No /prefs/module.proto preferences found
INFO  | ??:??:?? 1 Installing default ModuleConfig
INFO  | ??:??:?? 1 Loading /prefs/channels.proto
INFO  | ??:??:?? 1 Loaded saved channelFile version 22
[  1412][E][vfs_api.cpp:105] open(): /littlefs/oem/oem.proto does not exist, no permits for creation
INFO  | ??:??:?? 1 No /oem/oem.proto preferences found
DEBUG | ??:??:?? 1 cleanupMeshDB purged 0 entries
WARN  | ??:??:?? 1 Using nodenum 0x7a6c69c4 
DEBUG | ??:??:?? 1 Number of Device Reboots: 21795
DEBUG | ??:??:?? 1 Expanding short PSK #1
INFO  | ??:??:?? 1 Wanted region 1, using US
DEBUG | ??:??:?? 1 region=1, NODENUM=0x7a6c69c4, dbsize=16
INFO  | ??:??:?? 1 Saving /prefs/db.proto
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x40081b9d  PS      : 0x00060035  A0      : 0x400870dc  A1      : 0x3ffc004c  
A2      : 0x04000000  A3      : 0x3ffbe658  A4      : 0x80084a14  A5      : 0x400980c2  
A6      : 0x3ffda878  A7      : 0x00000804  A8      : 0x3ff44000  A9      : 0x3ffc002c  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x3ffb78f0  
A14     : 0x3ffda878  A15     : 0x0000000c  SAR     : 0x00000006  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x40093504  LEND    : 0x4009350f  LCOUNT  : 0xffffffff  
Core  1 was running in ISR context:
EPC1    : 0x40156843  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

Backtrace: 0x40081b9a:0x3ffc004c |<-CORRUPTED

Core  0 register dump:
PC      : 0x40231f42  PS      : 0x00060835  A0      : 0x801545d4  A1      : 0x3ffbce40  
A2      : 0x00000000  A3      : 0x80000001  A4      : 0x8009bca4  A5      : 0x3ffbcd60  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x80154116  A9      : 0x3ffbce10  
A10     : 0x00000000  A11     : 0x80000001  A12     : 0x8009bca4  A13     : 0x3ffbcd40  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001d  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x40231f3f:0x3ffbce40 0x401545d1:0x3ffbce60 0x4009961c:0x3ffbce80

ELF file SHA256: bb5e6e2f7662bad0

E (2385) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
```

attermann, 2024-01-07:
Here are logs after RST that show a normal boot after clearing the issue for reference.

```Shell
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
E (865) esp_core_dump_flash: No core dum����ѥѥ���found!
E (865) esp_core_dump_flash: No core dump partition found!
[    12][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   464][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
��@INFO  | ??:??:?? 0 

//\ E S H T /\ S T / C

INFO  | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=reset
DEBUG | ??:??:?? 0 Filesystem files (499712/1048576 Bytes):
DEBUG | ??:??:?? 0  /prefs/channels.proto (53 Bytes)
DEBUG | ??:??:?? 0  /prefs/config.proto (88 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (1493 Bytes)
DEBUG | ??:??:?? 0  /static/.gitkeep (0 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_Black.svg.gz (602 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_White.svg.gz (610 Bytes)
DEBUG | ??:??:?? 0  /static/apple-touch-icon.png.gz (3164 Bytes)
DEBUG | ??:??:?? 0  /static/favicon.ico.gz (2270 Bytes)
DEBUG | ??:??:?? 0  /static/icon.svg.gz (852 Bytes)
DEBUG | ??:??:?? 0  /static/index-H3yz6dTg.js.gz (348 Bytes)
DEBUG | ??:??:?? 0  /static/index-cnVnppgi.css.gz (17208 Bytes)
DEBUG | ??:??:?? 0  /static/index-t3MYXjf0.js.gz (206869 Bytes)
DEBUG | ??:??:?? 0  /static/index.html.gz (545 Bytes)
DEBUG | ??:??:?? 1  /static/maplibre-gl-pS6olApQ.js.gz (208178 Bytes)
DEBUG | ??:??:?? 1  /static/robots.txt.gz (42 Bytes)
DEBUG | ??:??:?? 1  /static/site.webmanifest.gz (197 Bytes)
[  1062][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
[  1064][I][XPowersCommon.tpp:248] begin(): SDA:21 SCL:22
[  1068][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
INFO  | ??:??:?? 1 AXP2101 PMU init succeeded, using AXP2101 PMU
[  1100][D][XPowersAXP2101.tpp:2791] setInterruptImpl(): DISABLE - HEX:0xffffffff BIN:
DEBUG | ??:??:?? 1 =======================================================================
DEBUG | ??:??:?? 1 DC1  : +   Voltage:3300 mV 
DEBUG | ??:??:?? 1 DC2  : -   Voltage:900 mV 
DEBUG | ??:??:?? 1 DC3  : -   Voltage:900 mV 
DEBUG | ??:??:?? 1 DC4  : -   Voltage:1100 mV 
DEBUG | ??:??:?? 1 ALDO1: -   Voltage:1800 mV 
DEBUG | ??:??:?? 1 ALDO2: +   Voltage:3300 mV 
DEBUG | ??:??:?? 1 ALDO3: +   Voltage:3300 mV 
DEBUG | ??:??:?? 1 ALDO4: -   Voltage:2900 mV 
DEBUG | ??:??:?? 1 BLDO1: -   Voltage:1800 mV 
DEBUG | ??:??:?? 1 BLDO2: -   Voltage:2800 mV 
DEBUG | ??:??:?? 1 =======================================================================
DEBUG | ??:??:?? 1 Battery: usbPower=1, isCharging=0, batMv=4163, batPct=100
INFO  | ??:??:?? 1 Scanning for i2c devices...
[  1170][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
DEBUG | ??:??:?? 1 Scanning for i2c devices on port 1
DEBUG | ??:??:?? 1 I2C device found at address 0x34
INFO  | ??:??:?? 1 axp192/axp2101 PMU found
DEBUG | ??:??:?? 1 I2C device found at address 0x3c
INFO  | ??:??:?? 1 ssd1306 display found
INFO  | ??:??:?? 1 ssd1306 display found
DEBUG | ??:??:?? 1 0x4 subtype probed in 2 tries 
INFO  | ??:??:?? 1 2 I2C devices found
DEBUG | ??:??:?? 1 acc_info = 0
INFO  | ??:??:?? 1 Meshtastic hwvendor=4, swver=2.2.16.1c6acfd
DEBUG | ??:??:?? 1 Setting random seed 77522584
[... 55 lines cut ...]
INFO  | ??:??:?? 1 Bandwidth set to 250.000000
INFO  | ??:??:?? 1 Power output set to 20
INFO  | ??:??:?? 1 RF95 Radio init succeeded, using RF95 radio
INFO  | ??:??:?? 1 Not using WIFI
DEBUG | ??:??:?? 1 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=237, time 2000 ms
DEBUG | ??:??:?? 1 LoRA bitrate = 118.500000 bytes / sec
INFO  | ??:??:?? 1 PowerFSM init, USB power=0
DEBU���??:??:?? 1 Enter state: BOOT
[  1857][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 320 / 4 = 80 Mhz, APB: 80000000 Hz
DEBUG | ??:??:?? 1 [Power] Battery: usbPower=1, isCharging=0, batMv=4163, batPct=100
DEBUG | ??:??:?? 1 [Screen] Screen: Started...
DEBUG | ??:??:?? 1 [GPS] WANT GPS=0
INFO  | ??:??:?? 1 [GPS] Setting GPS power=0
INFO  | ??:??:?? 1 [RangeTestModule] Range Test Module - Disabled
DEBUG | ??:??:?? 4 [PowerFSM] Enter state: ON
INFO  | ??:??:?? 4 [PowerFSM] Initialise the NimBLE bluetooth module
INFO  | ??:??:?? 6 [PowerFSM] Turning off screen
INFO  | ??:??:?? 8 [PowerFSM] lsEnter begin, ls_secs=86400
INFO  | ??:??:?? 9 [PowerFSM] Disable bluetooth
INFO  | ??:??:?? 22 [PowerFSM] Exit light sleep gpio: btn=0
INFO  | ??:??:?? 22 [PowerFSM] wakeCause2 7
INFO  | ??:??:?? 22 [PowerFSM] Exit state: LS
DEBUG | ??:??:?? 22 [PowerFSM] Enter state: NB
INFO  | ??:??:?? 22 [PowerFSM] Disable bluetooth
```

mverch67, 2024-01-12:
_Interrupt wdt timeout on CPU1_ means that a long-running action may be executed inside an interrupt handler (e.g. serial print) which is bad and can lead to arbitrary errors, in this case a timeout exception is triggered and then panic with reboot.

geeksville, 2024-01-27:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/lilygo-t-bean-supreme-s3-boot-loop/9517/2

donniefitz2, 2024-01-27:
I created the forum post mentioned above. In my case, with the T-Beam Supreme, the device enters the boot loop as soon as it restarts after being set to the Router role. I will get the log output and add it here soon.

donniefitz2, 2024-01-28:
It turns out that I can't post logs because I can't get the device out of the boot loop. The only way to recover is to flash the device.

mverch67, 2024-02-20:
There is hope... I could reproduce this issue!

Felix-Quehl, 2024-05-16:
I have the same issue. The device works fine in client mode, but as soon as I set it into `repeater` or `router` mode it cycles in a crash/boot loop and is not working.
I tried firmware version

- `2.3.9`
- `2.3.7`
- `2.3.5`
- `2.3.4`

MannyLama, 2024-08-14:
Can reproduce here on the T3-S3 LilyGo running 2.4.1
On version 2.1.11 seems to be working fine.