meshtastic/firmware #3071: [Bug]: pool.ntp.org used without vendor pool

anm, 2024-01-09:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

2.2.16

### Description

n.pool.ntp.org server addresses should not be used in products. They should use a suitable subdomain which can be applied for.

See https://www.ntppool.org/en/vendors.html

garethhcoleman, 2024-04-07:
The ntppool folks seem very straightforward - a contact form [here](https://manage.ntppool.org/manage/vendor) asks for a name and email address and a message. A message could be sent saying something like:

"Are you able to assign the meshtastic project a vendor code so that Meshtastic devices in the field can take advantage of your service?

The Meshtastic project is non-commercial opensource https://meshtastic.org/docs/introduction

The ntp client code itself is from another project, https://github.com/arduino-libraries/NTPClient

Thank you for your consideration.

Sent on behalf of Meshtastic LLC."

thebentern, 2024-05-06:
I've submitted an application. Waiting to hear back

PR #3819 (2024-05-07): Default to new vendor ntp pool
We received a vendor pool from ntp.org
Closes #3071
closed 2024-05-07 by thebentern