meshtastic/firmware #3072: [Feature Request]: Shutoff/deep-sleep on low battery

EternityForest, 2024-01-09:
### Platform

ESP32

### Description

Some lithium cells may not be protected, and not all users are going to understand the importance, and on top of that, trusting the protection PCB alone might not be the best idea for something somewhat relevant for safety.  

Most boards also do not offer any kind of power switch, and adding one can be tricky without making the case bulky. Plus, one may wish to conserve battery in case it's needed for an emergency canned message.

On boards that are able to do so, I think it would be very nice to have a setting that automatically goes into a total deep sleep, with the WiFi, Bt, and LoRa off, waking up once every few minutes only to check if the battery voltage is above the programmed minimum.

This way, when the battery got below 10% or so, it could turn off and potentially give you days or weeks to recharge, before the battery would be at risk.  This could also be useful on very small solar repeaters, which might occasionally run out of power in stormy weather.

Perhaps it could send it's battery status and other telemetry once every few days in the low power state, so people can tell what's up with unattended remote nodes for as long as possible, if the solar panel failed, you would have a little more info to work with for a bit longer until you could fix it.

GUVWAF, 2024-01-09:
This is already in place: 
https://github.com/meshtastic/firmware/blob/ea651c2f8fa421ad64809eb9d66b6a466471d643/src/Power.cpp#L526-L538

EternityForest, 2024-01-09:
Was not aware of that! Is it in the documentation anywhere? Maybe it should have a blurb on the power config page?

GUVWAF, 2024-01-09:
Go for it!

closed 2024-01-09 by garthvh

b8b8, 2025-03-08:
It seems the T1000E can get over one month of battery life if set to tracker role and position every 30min. 
Perhaps for NRF devices, this can be reworked so if below 10% battery set to sensor role with telemetry every 12hours until the battery has gotten up to 50% charge. This could only be for nodes initially set to router. 

https://discord.com/channels/867578229534359593/981977696051560458/1345950940695625868

garthvh, 2025-03-09:
10% is going to send way to many notifications on either platform, iOS sends the local node notification at 4% and can still take hours to die.

b8b8, 2025-03-09:
?
i am not talking about an app notification. I am talking about nodes in router mode that have less than 10% battery to automatically go into some sort of deepsleep workaround that was affecting some NRF boards until the solar panel has a chance to up the voltage.