meshtastic/firmware #3074: Heltec Tracker Screen Timeout[Bug]:

BobsBlueNorth, 2024-01-09:
### Category

Hardware Compatibility

### Hardware

Heltec Wireless Tracker

### Firmware Version

2.2.17.dbac2b1 and prior

### Description

I have two Heltec Wireless Tracker devices. The screen timeout is non-functional and the display remains powered up all the time no matter what value I enter in for the time delay.

jp-bennett, 2024-01-09:
The TFT power supply is also the GNSS power supply on these devices. We need to blank the screen, similar to what we do on the Pi if there's no backlight pin defined: https://github.com/meshtastic/firmware/blob/master/src/graphics/TFTDisplay.cpp#L495C9-L495C21

BobsBlueNorth, 2024-01-10:
I fell back to 2.2.13 and the screen timeout seems to be working.

jp-bennett, 2024-01-10:
> I fell back to 2.2.13 and the screen timeout seems to be working.

Is this a v1.1 tracker, or the original 1.0?

BobsBlueNorth, 2024-01-11:
Sorry. V1.1

Gabrielerusso, 2024-02-02:
https://github.com/Gabrielerusso/firmware

i fixed it there, i'll try to push it soon after i test some other things i fixed

closed 2024-02-20 by thebentern