meshtastic/firmware #3076: [Bug]: GPS Latitude value randomly inverts

kingoffoxes, 2024-01-11:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.2.16.1c6acfd

### Description

Occasionally the reported node GPS position looses the negative in front. 
ie. -40.0000 becomes 40.0000 which shows on the map as a position on the other side of the planet. 
using Apple app to view map. Compared on the local app and another node reporting the DUT nodes position on its map.

KodinLanewave, 2024-01-11:
I've seen similar issues but instead of just losing the negative, I lose the entire value of latitude and it goes to zero, as shown here from my Owntracks history for one of my nodes:
[image]

jp-bennett, 2024-01-11:
This looks familiar. @KodinLanewave , what hardware is this? 
@kingoffoxes, do you know which revision of the t-beam you have?

kingoffoxes, 2024-01-11:
> This looks familiar. @KodinLanewave , what hardware is this? @kingoffoxes, do you know which revision of the t-beam you have?

T-Beam is a T-22_v1.1 20210222 (SX1262) with GPS u-blox NEO-M8N-0-10

KodinLanewave, 2024-01-13:
I'm using a RAK with the M8N GPS module.

madeofstown, 2024-04-02:
I see this on my Heltec V3 and Pi Pico W using my phones GPS as well as on my Raspberry Pi 4 (Linux native device) with a LoRA/GNSS Hat.

closed 2024-04-12 by thebentern