meshtastic/firmware #3080: [Board]: Is it possible to add the Lilygo T3 LoRa ESP 32 TCXO?

Burny08, 2024-01-11:
### SOC

ESP32

### Lora IC

LILYGO 868/915Mhz Modell: LORA32 TCXO

### Product Link

https://de.aliexpress.com/item/32872078587.html?spm=a2g0o.order_list.order_list_main.9.36e05c5fprjabY&gatewayAdapt=glo2deu

### Description

I have been informed by the manufacturer that the TCXO variant is not supported. 

The Modell is: T3 V1.6.1 with SX1276

Unfortunately I am not a programmer myself, but he told me that this board can probably be added. 
Here is the message from the manufacturer: 

> Hello, the meshtastic firmware is not adapted to the TXCO version yet.
> You can add the below sentences in the source code to make it work.
> https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/examples/RadioLibExamples/SX1276/SX1276_Receive_Interrupt/boards.h
> 
> Or to test hardware function, you can upload the below firmware to  check :
> https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/firmware/hardware_test/T3_V1.6.1_TXCO
> https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/firmware#to-write-firmware-please-check-the-corresponding-link-below
[image]

marclura, 2024-01-13:
Hi, I managed to make the LilyGo T3 V1.6.1 TXCO version work, by modifying the existing _tlora_v2_1_16_  variant (I couldn't figure out how to create a new variant and make it work as a separate board).

Burny08, 2024-01-13:
> Hi, I managed to make the LilyGo T3 V1.6.1 TXCO version work, by modifying the existing _tlora_v2_1_16_ variant (I couldn't figure out how to create a new variant and make it work as a separate board).

That sounds great! Can you tell me how I can get this version onto my board?
I can use the Espressif Flasher, but then my knowledge slowly stops :D

geeksville, 2024-01-13:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/t3-v1-6-tcxo-variant/9240/2

Burny08, 2024-01-13:
It works!
Thank you very much! 

For everyone who wants to imitate it:

Go to this Site: https://espressif.github.io/esptool-js/

Choose the baud rate 115200
Now hold down the RST button on the board, plug in USB, click on Connect and then release the RST button.

(It may be advisable to first delete the entire memory using Erase)

Set the address to 0x0 and select the firmware from @marclura .

Then click on program.

I'm Happy :))

mverch67, 2024-01-27:
In case the above works as expected I'd still like to propose a more generic name which improves re-usability and maintainability, because in theory there could be other devices using the same implementation, e.g.

```
#if defined(LORA_TCXO)
    pinMode(LORA_TCXO, OUTPUT);
    digitalWrite(LORA_TCXO, HIGH);
#endif 
```

rather than using device specific naming.

closed 2024-01-30 by thebentern

Burny08, 2024-04-22:
Hello again, 
I thought TCXO was now included in the official firmware? 
I just wanted to update to the current version and unfortunately I get Fault #3 again :(
Can someone tell me something about this? I would like to have the latest firmware for my TTGO LoRa V2.1 1.6 TCXO
Thanks!!

tcwed-br, 2025-02-11:
> Hi, I managed to extract the .bin firmware that you can flash directly with the [ESP Flash Tool](https://www.espressif.com/en/support/download/other-tools); check the screenshot for the settings:
> 
> <img alt="flash_download_tool_3 9 5_20240113_174857" width="457" src="https://private-user-images.githubusercontent.com/29801332/296504496-f2f9a74b-5635-47a0-8432-a98194d17a18.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzkzMTA4NDcsIm5iZiI6MTczOTMxMDU0NywicGF0aCI6Ii8yOTgwMTMzMi8yOTY1MDQ0OTYtZjJmOWE3NGItNTYzNS00N2EwLTg0MzItYTk4MTk0ZDE3YTE4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAyMTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMjExVDIxNDkwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI1ODkxMTc3OGJiOGVlN2I1NjYwMTg0OTNlMjZiZWIxZmU0ZTlmMDliNTg3NzIwZGM2YmE4Yjc3NWNiYTYxOTYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.9EWTXrgP-_PxGxJE91YP9D31kiytxv1apKhpeJ0g6MI">
> Here is the firmware file (I uploaded it on one of the boards that I've, and it worked, but I can't give you any guarantee if it doesn't work or it does damage your board): [firmware_meshtastic_2.2.17_lilygo_T3_V1_6_1_TCXO.bin.zip](https://github.com/meshtastic/firmware/files/13928912/firmware_meshtastic_2.2.17_lilygo_T3_V1_6_1_TCXO.bin.zip)
> 
> Let me know how it goes!