meshtastic/firmware #3086: [Feature Request]: NodeDB "Node Channel" should prefer channels based on user preference

Evanito, 2024-01-12:
### Platform

NRF52, ESP32

### Description

Hello, I am creating this issue to discuss something I may make a PR for, but want to get alignment from the maintainers on first.

### Problem
In the following scenario:
- Three node mesh: NodeA, NodeB, NodeD
- NodeA and NodeB have a private channel with default LongFast as secondary, NodeD has a default channel configuration
- NodeA and NodeB use the Direct Message feature often to communicate, and expect privacy
- NodeD has a setup that allows them to log all non-broadcast messages on LongFast, such as by using MQTT

In the above scenario, the following negative situation may occur:
1. NodeA receives a NodeInfo packet from NodeB on the LongFast channel
2. NodeA updates it's NodeDB entry for NodeB to note that NodeB is now on the LongFast channel
3. NodeA sends a direct message to NodeB, which is now sent over LongFast
4. NodeD intercepts and reads this message

If any assumptions made above are incorrect, and this situation would in fact not occur, please feel free to close this issue.

### Proposed Solution

I suggest implementing a priority of channels, where the channel closest to 0 will be chosen for each remote node and kept consistent.

### Proposed simple implementation
Add a channel number minimization check to:
https://github.com/meshtastic/firmware/blob/master/src/mesh/NodeDB.cpp#L851 
```c++
if (mp.decoded.portnum == meshtastic_PortNum_NODEINFO_APP && mp.channel < info->channel) {
    info->channel = mp.channel;
}
``` 

### Shortcomings with above implementation
- Channel 0 is not always the primary channel
- Remote nodes that remove the preferred channel will not be communicable until removed and re-entered in the NodeDB

What do you think?

garthvh, 2024-01-12:
Take a look at the channel set and Lora config that makes up the QR code, the channel indexes and names are really client only constructs.  Unless we make breaking changes there has to be a primary channel, and it is always 0

Evanito, 2024-01-12:
Oh, so it looks like shortcoming A is not an issue, as it seems we are already assuming channel 0 will be the channel that the user expects to be communicating on? I have crossed that out.

Evanito, 2024-01-12:
I sent out the above PR, though I'm really leaning towards not changing this behavior until we can have some way to "downgrade" to a higher index channel in the case where it is impossible to communicate with the remote node on the primary channel.

garthvh, 2024-01-12:
> I sent out the above PR, though I'm really leaning towards not changing this behavior until we can have some way to "downgrade" to a higher index channel in the case where it is impossible to communicate with the remote node on the primary channel.

Yeah if @GUVWAF or @andrekir are not on board with routing changes, we don't make them.

closed 2024-01-12 by garthvh