meshtastic/firmware #3095: 🚀 [Feature Request]: Allow adding an existing channel

dustin, 2024-01-14:
### OS

iOS, iPadOS

### Description

For an existing meshtastic user, there is currently no way to add an existing encrypted channel.

A channel requires a PSK (pre-shared key) to be able to send and receive messages, but the iOS UI does not have a means of supplying a key that was created for an existing channel.

In practice, this means that if you have a typical setup (a private primary, public LongFast, an admin channel, etc...) and wish to join a group, the only option available to join this group is to scan a QR code which will overwrite all of your channel settings.  Ideally, one would be able to communicate with one group without losing communication with another.

## Concerns

There is a concern that allowing a key to be pasted in might result in someone generating a less secure shared key.  This is certainly possible, but it's already possible via QR code.  While we should encourage our best security practices for end users, which means using a secure mechanism for generating a PSK, we shouldn't make it impossible for them to communicate with others out of concern that the key wasn't controlled well.

## Precedent

To add an existing channel today, one may use the Android client, web client, or python cli, so there is precedent (and workarounds for some).

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

This has come up in discussion a few times with users who have been limited by it.

garthvh, 2024-01-14:
Moved to firmware repo as this functionality is not supported by the current QR code structure

Siftah, 2024-01-14:
This isn't a firmware issue, a new channel can be added in all other clients with the key pasted in. This is a UX issue with the iOS client.

dustin, 2024-01-14:
What aspect of the firmware is currently preventing iOS users from adding a channel.  Why isn't this blocking Android, web, and CLI users?

ianmcorvidae, 2024-01-19:
From some discussion in discord and the comment above about key derivation functions it appears that Garth may think this request is for pasting in arbitrary strings, having them converted to AES keys via a key derivation function, and then added to channels. To my (contrary) understanding, the request is to allow already-generated AES keys (most likely provided by someone else, and hopefully generated randomly, but in any case already in use on a channel in some other node) to be used when adding a channel.

The request (by my understanding) is currently possible in non-iOS clients by pasting in a base64-encoded string, but the particular interface (copy-paste into a textbox) and encoding method (base64 string) are implementation details of those clients, not a requisite component of the request. I see how Garth's understanding would need to be done in firmware, since the key derivation function would probably best live there, not in clients, but I don't think that's a feature anyone's asking for. My implementation idea above is written based on my understanding of the request and would not require a key derivation function in firmware or otherwise.

I hope the confusion can be resolved, in any case!

dustin, 2024-01-19:
> but I don't think that's a feature anyone's asking for

Right, this is correct.  In particular, a derivation function would most likely defeat "Allow adding an existing channel" because that's not how anyones PSKs are currently defined or shared.

Users already have a (presumably securely) generated key, either through the iOS interface or using the [best practices doc from Google](https://cloud.google.com/network-connectivity/docs/vpn/how-to/generating-pre-shared-key ) I mentioned above.  If the user is using iOS, they do not have a way to join another user's channel.  If they're not iOS, they can.  I do think iOS users should also be able to.

While I do fail to understand the issue with treating this PSK like any other PSK and pasting it in, the means isn't specified in the title or the description.

edhaber, 2024-01-19:
Here is a screen recording of allowing a user to enter a base64 encoded key on iOS with a bit of basic validation that the base64 text decodes to a key of the correct length.

https://github.com/meshtastic/firmware/assets/202751/df43079b-ec23-4a45-bd3e-2bbd3204178d

Siftah, 2024-01-19:
> Here is a screen recording of allowing a user to enter a base64 encoded key on iOS with a bit of
> basic validation that the base64 text decodes to a key of the correct length.

This looks like a perfect solution!

dustin, 2024-01-19:
> The only way AES can be cracked is by using a password as your key, this mitigates this issue and is not related to a QR structure to add one channel to your channelset.

This statement is wrong at least three different ways and is similarly unrelated to this issue.

I'm very confused as to what you're reading that leads you leave comments like this one on this issue.

This is an issue about adding an existing channel to a configuration on iOS.  None of these things you mentioned have anything to do with that basic functionality.

But for whatever issue you're talking about:

1. Whatever you think "password" means is unrelated to this issue or any security aspect of adding a channel.
2. There are other ways to crack channel security.  There are at least two that I know people have done successfully on meshtastic.
3. Putting a key in a picture does not make it safer in any context than supplying it a different way.  I can't even imagine a scenario where it could.  I can think of specific scenarios where it might be slightly easier and of scenarios where it reduces security, but it doesn't enhance security.

In any case, none of these issues are remotely related to this bug and have nothing to do with firmware.  I do wish we could talk about the issue I filed.

crypto-smoke, 2024-01-19:
No one is proposing the use of password as a key. Allow import of a single channel without changing the current channel creation flow and then there is no issue with users using weak keys.

garthvh, 2024-01-19:
> No one is proposing the use of password as a key. Allow import of a single channel without changing the current channel creation flow and then there is no issue with users using weak keys.

Which is why the issue was moved here, there is no share a single channel qr structure available.

dustin, 2024-01-20:
> Which is why the issue was moved here, there is no share a single channel qr structure available.

That's not a requirement to solve this problem, as can be demonstrated by being able to do it other than iOS.  If you wanted to require a QR code for some reason, that's 100% possible today in iOS or Android without any changes because the firmware fundamentals are quite good with reasonably orthogonal feature sets.

In scalable and resilient distributed system design, you typically want smarter clients and dumber servers.  This is especially true when the clients are extremely powerful and very easy to update and the servers (nodes in this case) have very limited CPU and memory resources available and are also very difficult to update.

It's easy to argue why this shouldn't require any changes to firmware (not the least of which is demonstrating that it works today), but if there's any good argument as to why firmware changes need to be deployed in order to add a channel (a thing people have already been doing for a while now), perhaps that's worth documenting somewhere.

I don't think the firmware is perfect, but I think the system design is quite good given practical limitations.  Is there a good argument for complicating this design and deviating from best practices?

garthvh, 2024-01-20:
Open a specific issue with something actionable.

closed 2024-01-20 by garthvh

ianmcorvidae, 2024-02-26:
As an update for anyone who wanted this, I believe the stated desire (adding an existing channel that someone else created, by copying the base64-encoded PSK into a text field) has been implemented in the iOS app now, by meshtastic/Meshtastic-Apple#497 and folks in discord are reporting that it's in their version(s) of the app.