meshtastic/firmware #3097: T-Beam APX2101-V1.2 w/NEO M8N-0-10[Bug]:

BobsBlueNorth, 2024-01-15:
### Category

Hardware Compatibility

### Hardware

T-Beam

### Firmware Version

2.2.18.e9bde80

### Description

I have two T-Beams w/ the NEO M8N GNSS.  

These two devices seem to be having an issue with updating their GPS location after the initial boot. I have been watching the logs and I can see the GNSS boot, lock and report the position at start-up.  

I can also see the GNSS being woken up every two minutes looking for updates in location.   What I find interesting is it never updates the position. It always reports back the exact same Lat/Long/Alt, even when I disconnect the GNSS antenna.

When the GNSS is asleep, the LED indicator light for it is off. During the update at the 2 minutes mark, it flashes once.

I am not seeing this on my T-Beams with NEO-6M or my Heltec Wireless Trackers with UC6580 GNSS. They both update correctly.

### Relevant log output

```Shell
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
E (836) esp_core_dump_flash: No core dum���ѥѥ���found!
E (837) esp_core_dump_flash: No core dump partition found!
[    12][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   464][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
��@INFO  | ??:??:?? 0 

//\ E S H T /\ S T / C

INFO  | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=reset
DEBUG | ??:??:?? 0 Filesystem files (516096/1048576 Bytes):
DEBUG | ??:??:?? 0  /prefs/channels.proto (53 Bytes)
DEBUG | ??:??:?? 0  /prefs/config.proto (89 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (159 Bytes)
DEBUG | ??:??:?? 0  /static/.gitkeep (0 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_Black.svg.gz (602 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_White.svg.gz (610 Bytes)
DEBUG | ??:??:?? 0  /static/apple-touch-icon.png.gz (3164 Bytes)
DEBUG | ??:??:?? 0  /static/favicon.ico.gz (2270 Bytes)
DEBUG | ??:??:?? 0  /static/icon.svg.gz (852 Bytes)
DEBUG | ??:??:?? 0  /static/index-H3yz6dTg.js.gz (348 Bytes)
DEBUG | ??:??:?? 0  /static/index-eebp_vQT.js.gz (232487 Bytes)
DEBUG | ??:??:?? 0  /static/index-tjk88OUb.css.gz (16125 Bytes)
DEBUG | ??:??:?? 0  /static/index.html.gz (547 Bytes)
DEBUG | ??:??:?? 0  /static/maplibre-gl-j-ujuLPK.js.gz (208176 Bytes)
DEBUG | ??:??:?? 0  /static/robots.txt.gz (42 Bytes)
DEBUG | ??:??:?? 0  /static/site.webmanifest.gz (197 Bytes)
[   986][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
[   988][I][XPowersCommon.tpp:248] begin(): SDA:21 SCL:22
[   992][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
INFO  | ??:??:?? 0 AXP2101 PMU init succeeded, using AXP2101 PMU
[  1025][D][XPowersAXP2101.tpp:2791] setInterruptImpl(): DISABLE - HEX:0xffffffff BIN:
DEBUG | ??:??:?? 1 =======================================================================
DEBUG | ??:??:?? 1 DC1  : +   Voltage:3300 mV 
[... 206 lines cut ...]
INFO  | 01:52:03 126 [GPS] updatePosition LOCAL pos@65a48fc4, time=1705283523, latI=613091201, lonI=-1495293001, alt=175
DEBUG | 01:52:03 126 [GPS] Setting local position: latitude=613091201, longitude=-1495293001, time=1705283523
DEBUG | 01:52:03 126 [GPS] Node status update: 1 online, 1 total
DEBUG | 01:52:20 142 [Power] Battery: usbPower=1, isCharging=0, batMv=4165, batPct=100
DEBUG | 01:52:40 162 [Power] Battery: usbPower=1, isCharging=0, batMv=4166, batPct=100
DEBUG | 01:53:00 182 [Power] Battery: usbPower=1, isCharging=0, batMv=4166, batPct=100
DEBUG | 01:53:20 202 [Power] Battery: usbPower=1, isCharging=0, batMv=4165, batPct=100
DEBUG | 01:53:40 222 [Power] Battery: usbPower=1, isCharging=0, batMv=4166, batPct=100
DEBUG | 01:54:00 242 [Power] Battery: usbPower=1, isCharging=0, batMv=4165, batPct=100
DEBUG | 01:54:03 246 [GPS] WANT GPS=1
INFO  | 01:54:03 246 [GPS] Setting GPS power=1
WARN  | 01:54:03 246 [GPS] SOME data is TOO OLD: LOC 120065, TIME 120065, DATE 120065
WARN  | 01:54:04 246 [GPS] SOME data is TOO OLD: LOC 120271, TIME 120271, DATE 120271
WARN  | 01:54:04 246 [GPS] SOME data is TOO OLD: LOC 120491, TIME 120491, DATE 120491
WARN  | 01:54:04 246 [GPS] SOME data is TOO OLD: LOC 120710, TIME 120710, DATE 120710
WARN  | 01:54:04 247 [GPS] BOGUS course.value() REJECTED: 1113705842
DEBUG | 01:54:04 247 [GPS] WANT GPS=0
DEBUG | 01:54:04 247 [GPS] GPS Lock took 0, average 0
INFO  | 01:54:04 247 [GPS] Setting GPS power=0
DEBUG | 01:54:04 247 [GPS] publishing pos@65a4903d:2, hasVal=1, Sats=8, GPSlock=1
DEBUG | 01:54:04 247 [GPS] onGPSChanged() pos@65a4903d, time=1705283644, lat=613091201, lon=-1495293001, alt=175
INFO  | 01:54:04 247 [GPS] updatePosition LOCAL pos@65a4903d, time=1705283644, latI=613091201, lonI=-1495293001, alt=175
DEBUG | 01:54:04 247 [GPS] Setting local position: latitude=613091201, longitude=-1495293001, time=1705283644
DEBUG | 01:54:04 247 [GPS] Node status update: 1 online, 1 total
```

caveman99, 2024-01-15:
WARN  | 01:52:03 126 [GPS] BOGUS course.value() REJECTED: 1113705842

this is the problem.

BobsBlueNorth, 2024-01-17:
What would be causing the GNSS to report a bogus course?

BobsBlueNorth, 2024-01-23:
Rolled back to 2.2.11 to see if that would change anything. With 2.2.11 I don't even get a lock, which normally takes a couples minutes max.

//\ E S H T /\ S T / C
INFO  | ??:??:?? 1 Meshtastic hwvendor=4, swver=2.2.11.10265aa

INFO  | ??:??:?? 1 Setting GPS power=1
DEBUG | ??:??:?? 1 WANT GPS=1
INFO  | ??:??:?? 1 Setting GPS power=1
DEBUG | ??:??:?? 1 Using GPIO34 for GPS RX
DEBUG | ??:??:?? 1 Using GPIO12 for GPS TX

DEBUG | ??:??:?? 2 [GPS] NMEA GPS time 2024-01-23 04:43:42
DEBUG | ??:??:?? 2 [GPS] Upgrading time to quality 4
DEBUG | 04:43:42 2 [GPS] Read RTC time as 1705985022
WARN  | 04:43:42 2 [GPS] Warning, 2 new GPS checksum failures, for a total of 2.
DEBUG | 04:43:42 2 [GPS] WANT GPS=0
DEBUG | 04:43:42 2 [GPS] GPS Lock took 0, average 0
INFO  | 04:43:42 2 [GPS] Setting GPS power=0
DEBUG | 04:43:42 2 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 04:43:42 2 [GPS] No GPS lock
DEBUG | 04:43:42 2 [GPS] onGPSChanged() pos@0, time=1705985022, lat=0, lon=0, alt=0
INFO  | 04:43:42 2 [GPS] updatePosition LOCAL pos@0, time=1705985022, latI=0, lonI=0, alt=0
DEBUG | 04:43:42 2 [GPS] Setting local position: latitude=0, longitude=0, time=1705985022
DEBUG | 04:43:42 2 [GPS] Node status update: 1 online, 9 total

DEBUG | 04:45:42 122 [GPS] WANT GPS=1
INFO  | 04:45:42 122 [GPS] Setting GPS power=1
DEBUG | 04:45:42 122 [GPS] WANT GPS=0
DEBUG | 04:45:42 122 [GPS] GPS Lock took 0, average 0
INFO  | 04:45:42 122 [GPS] Setting GPS power=0
DEBUG | 04:45:42 122 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 04:45:42 122 [GPS] onGPSChanged() pos@0, time=1705985142, lat=0, lon=0, alt=0
INFO  | 04:45:42 122 [GPS] updatePosition LOCAL pos@0, time=1705985142, latI=0, lonI=0, alt=0
DEBUG | 04:45:42 122 [GPS] Setting local position: latitude=0, longitude=0, time=1705985142
DEBUG | 04:45:42 122 [GPS] Node status update: 4 online, 9 total

DEBUG | 04:47:42 242 [GPS] WANT GPS=1
INFO  | 04:47:42 242 [GPS] Setting GPS power=1
DEBUG | 04:47:42 242 [GPS] WANT GPS=0
DEBUG | 04:47:42 242 [GPS] GPS Lock took 0, average 0
INFO  | 04:47:42 242 [GPS] Setting GPS power=0
INFO  | 04:47:42 242 [GPS] From Radio onread
DEBUG | 04:47:42 242 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
INFO  | 04:47:42 242 [GPS] getFromRadio=STATE_SEND_PACKETS
DEBUG | 04:47:42 242 [GPS] onGPSChanged() pos@0, time=1705985262, lat=0, lon=0, alt=0
DEBUG | 04:47:42 242 [GPS] phone downloaded packet (id=0x4bb0cb11 fr=0x78 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 rxtime=1705985261 rxSNR=8.5 rxRSSI=-77)
INFO  | 04:47:42 242 [GPS] updatePosition LOCAL pos@0, time=1705985262, latI=0, lonI=0, alt=0
DEBUG | 04:47:42 242 [GPS] Setting local position: latitude=0, longitude=0, time=1705985262
DEBUG | 04:47:42 242 [GPS] encoding toPhone packet to phone variant=2, 64 bytes
DEBUG | 04:47:42 242 [GPS] Node status update: 6 online, 9 total

DEBUG | 04:49:42 362 [GPS] WANT GPS=1
INFO  | 04:49:42 362 [GPS] Setting GPS power=1
DEBUG | 04:49:42 362 [GPS] WANT GPS=0
DEBUG | 04:49:42 362 [GPS] GPS Lock took 0, average 0
INFO  | 04:49:42 362 [GPS] Setting GPS power=0
DEBUG | 04:49:42 362 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 04:49:42 362 [GPS] onGPSChanged() pos@0, time=1705985382, lat=0, lon=0, alt=0
INFO  | 04:49:42 362 [GPS] updatePosition LOCAL pos@0, time=1705985382, latI=0, lonI=0, alt=0
DEBUG | 04:49:42 362 [GPS] Setting local position: latitude=0, longitude=0, time=1705985382
DEBUG | 04:49:42 362 [GPS] Node status update: 6 online, 9 total

DEBUG | 04:51:42 482 [GPS] WANT GPS=1
INFO  | 04:51:42 482 [GPS] Setting GPS power=1
DEBUG | 04:51:42 482 [GPS] WANT GPS=0
DEBUG | 04:51:42 482 [GPS] GPS Lock took 0, average 0
INFO  | 04:51:42 482 [GPS] Setting GPS power=0
DEBUG | 04:51:42 482 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 04:51:42 482 [GPS] onGPSChanged() pos@0, time=1705985502, lat=0, lon=0, alt=0
INFO  | 04:51:42 482 [GPS] updatePosition LOCAL pos@0, time=1705985502, latI=0, lonI=0, alt=0
[... 10 lines cut ...]
INFO  | 04:53:42 603 [GPS] updatePosition LOCAL pos@0, time=1705985622, latI=0, lonI=0, alt=0
DEBUG | 04:53:42 603 [GPS] Setting local position: latitude=0, longitude=0, time=1705985622
DEBUG | 04:53:42 603 [GPS] Node status update: 6 online, 9 total

DEBUG | 04:55:42 723 [GPS] WANT GPS=1
INFO  | 04:55:42 723 [GPS] Setting GPS power=1
DEBUG | 04:55:42 723 [GPS] WANT GPS=0
DEBUG | 04:55:42 723 [GPS] GPS Lock took 0, average 0
INFO  | 04:55:42 723 [GPS] Setting GPS power=0
DEBUG | 04:55:42 723 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 04:55:42 723 [GPS] onGPSChanged() pos@0, time=1705985742, lat=0, lon=0, alt=0
INFO  | 04:55:42 723 [GPS] updatePosition LOCAL pos@0, time=1705985742, latI=0, lonI=0, alt=0
DEBUG | 04:55:42 723 [GPS] Setting local position: latitude=0, longitude=0, time=1705985742
DEBUG | 04:55:42 723 [GPS] Node status update: 6 online, 9 total

DEBUG | 04:57:42 843 [GPS] WANT GPS=1
INFO  | 04:57:42 843 [GPS] Setting GPS power=1
DEBUG | 04:57:42 843 [GPS] WANT GPS=0
DEBUG | 04:57:42 843 [GPS] GPS Lock took 0, average 0
INFO  | 04:57:42 843 [GPS] Setting GPS power=0
DEBUG | 04:57:42 843 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 04:57:42 843 [GPS] onGPSChanged() pos@0, time=1705985862, lat=0, lon=0, alt=0
INFO  | 04:57:42 843 [GPS] updatePosition LOCAL pos@0, time=1705985862, latI=0, lonI=0, alt=0
DEBUG | 04:57:42 843 [GPS] Setting local position: latitude=0, longitude=0, time=1705985862
DEBUG | 04:57:42 843 [GPS] Node status update: 6 online, 9 total

BobsBlueNorth, 2024-01-23:
-------------------------------------
**RESTART AFTER UPDATING FIX ATTEMPT DURATION FROM 0 SECONDS TO 900 SECONDS**
-------------------------------------

DEBUG | ??:??:?? 2 [GPS] Upgrading time to quality 4
DEBUG | 05:00:30 2 [GPS] Read RTC time as 1705986030
WARN  | 05:00:30 2 [GPS] Warning, 1 new GPS checksum failures, for a total of 1.
DEBUG | 05:00:30 2 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 05:00:30 2 [GPS] No GPS lock
DEBUG | 05:00:30 2 [GPS] onGPSChanged() pos@0, time=1705986030, lat=0, lon=0, alt=0
INFO  | 05:00:30 2 [GPS] updatePosition LOCAL pos@0, time=1705986030, latI=0, lonI=0, alt=0
DEBUG | 05:00:30 2 [GPS] Setting local position: latitude=0, longitude=0, time=1705986030
DEBUG | 05:00:30 2 [GPS] Node status update: 6 online, 9 total

WARN  | 05:01:00 32 [GPS] BOGUS course.value() REJECTED: 841841379
DEBUG | 05:01:00 32 [GPS] hasValidLocation RISING EDGE
DEBUG | 05:01:00 32 [GPS] WANT GPS=0
DEBUG | 05:01:00 32 [GPS] GPS Lock took 31, average 0
INFO  | 05:01:00 32 [GPS] Setting GPS power=0
DEBUG | 05:01:00 32 [GPS] publishing pos@65af480d:2, hasVal=1, Sats=7, GPSlock=1
DEBUG | 05:01:00 32 [GPS] New GPS pos@65af480d:3 lat=61.309102, lon=-149.529373, alt=178, pdop=1.72, track=0.00, speed=0.00, sats=7
DEBUG | 05:01:00 32 [GPS] onGPSChanged() pos@65af480d, time=1705986060, lat=613091023, lon=-1495293731, alt=178
INFO  | 05:01:00 32 [GPS] updatePosition LOCAL pos@65af480d, time=1705986060, latI=613091023, lonI=-1495293731, alt=178
DEBUG | 05:01:00 32 [GPS] Setting local position: latitude=613091023, longitude=-1495293731, time=1705986060
DEBUG | 05:01:00 32 [GPS] Node status update: 6 online, 9 total
INFO  | 05:01:00 32 [GPS] Sending smart pos@65af480d:6 to mesh (distanceTraveled=12715397.000000m, minDistanceThreshold=100m, timeElapsed=32980ms)
INFO  | 05:01:00 32 [GPS] Providing time to mesh 1705986060
INFO  | 05:01:00 32 [GPS] Position reply: time=1705986060, latI=613091023, lonI=--1495293731
DEBUG | 05:01:00 33 [GPS] Update DB node 0x1c50630c, rx_time=1705986060, channel=0
DEBUG | 05:01:00 33 [GPS] handleReceived(LOCAL) (id=0x3bbf843d fr=0x0c to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 WANTRESP rxtime=1705986060 priority=10)
DEBUG | 05:01:00 33 [GPS] No modules interested in portnum=3, src=LOCAL
DEBUG | 05:01:00 33 [GPS] localSend to channel 0
DEBUG | 05:01:00 33 [GPS] Add packet record (id=0x3bbf843d fr=0x0c to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 WANTRESP rxtime=1705986060 priority=10)
DEBUG | 05:01:00 33 [GPS] Using AES256 key!
DEBUG | 05:01:00 33 [GPS] ESP32 crypt fr=1c50630c, num=3bbf843d, numBytes=30!
DEBUG | 05:01:00 33 [GPS] enqueuing for send (id=0x3bbf843d fr=0x0c to=0xff, WantAck=0, HopLim=3 Ch=0xa4 encrypted rxtime=1705986060 priority=10)
DEBUG | 05:01:00 33 [GPS] txGood=1,rxGood=0,rxBad=0
DEBUG | 05:01:00 33 [GPS] Using channel 0 (hash 0xa4)
DEBUG | 05:01:00 33 [GPS] Using AES256 key!
DEBUG | 05:01:00 33 [GPS] ESP32 crypt fr=1c50630c, num=3bbf843d, numBytes=30!
DEBUG | 05:01:00 33 [GPS] decoded message (id=0x3bbf843d fr=0x0c to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 WANTRESP rxtime=1705986060 priority=10)

DEBUG | 05:03:00 153 [GPS] WANT GPS=1
INFO  | 05:03:00 153 [GPS] Setting GPS power=1
WARN  | 05:03:00 153 [GPS] SOME data is TOO OLD: LOC 120211, TIME 120211, DATE 120418
WARN  | 05:03:00 153 [GPS] SOME data is TOO OLD: LOC 120418, TIME 120418, DATE 120625
WARN  | 05:03:01 153 [GPS] SOME data is TOO OLD: LOC 120636, TIME 120636, DATE 120843
WARN  | 05:03:01 153 [GPS] SOME data is TOO OLD: LOC 120854, TIME 120854, DATE 121061
WARN  | 05:03:01 153 [GPS] BOGUS course.value() REJECTED: 841841379
DEBUG | 05:03:01 153 [GPS] WANT GPS=0
DEBUG | 05:03:01 153 [GPS] GPS Lock took 0, average 0
INFO  | 05:03:01 153 [GPS] Setting GPS power=0
DEBUG | 05:03:01 153 [GPS] publishing pos@65af4885:2, hasVal=1, Sats=7, GPSlock=1
DEBUG | 05:03:01 153 [GPS] onGPSChanged() pos@65af4885, time=1705986181, lat=613091023, lon=-1495293731, alt=178
INFO  | 05:03:01 153 [GPS] updatePosition LOCAL pos@65af4885, time=1705986181, latI=613091023, lonI=-1495293731, alt=178
DEBUG | 05:03:01 154 [GPS] Setting local position: latitude=613091023, longitude=-1495293731, time=1705986181
DEBUG | 05:03:01 154 [GPS] Node status update: 6 online, 9 total

DEBUG | 05:05:01 274 [GPS] WANT GPS=1
INFO  | 05:05:01 274 [GPS] Setting GPS power=1
WARN  | 05:05:01 274 [GPS] SOME data is TOO OLD: LOC 120064, TIME 120064, DATE 120064
WARN  | 05:05:01 274 [GPS] SOME data is TOO OLD: LOC 120271, TIME 120271, DATE 120271
WARN  | 05:05:02 274 [GPS] SOME data is TOO OLD: LOC 120489, TIME 120489, DATE 120489
WARN  | 05:05:02 274 [GPS] SOME data is TOO OLD: LOC 120707, TIME 120707, DATE 120707
WARN  | 05:05:02 274 [GPS] BOGUS course.value() REJECTED: 841841379
DEBUG | 05:05:02 274 [GPS] WANT GPS=0
DEBUG | 05:05:02 274 [GPS] GPS Lock took 0, average 0
INFO  | 05:05:02 274 [GPS] Setting GPS power=0
DEBUG | 05:05:02 274 [GPS] publishing pos@65af48fe:2, hasVal=1, Sats=7, GPSlock=1
DEBUG | 05:05:02 274 [GPS] onGPSChanged() pos@65af48fe, time=1705986302, lat=613091023, lon=-1495293731, alt=178
INFO  | 05:05:02 274 [GPS] updatePosition LOCAL pos@65af48fe, time=1705986302, latI=613091023, lonI=-1495293731, alt=178
DEBUG | 05:05:02 274 [GPS] Setting local position: latitude=613091023, longitude=-1495293731, time=1705986302
DEBUG | 05:05:02 274 [GPS] Node status update: 6 online, 9 total

DEBUG | 05:07:02 394 [GPS] WANT GPS=1
INFO  | 05:07:02 394 [GPS] Setting GPS power=1
[... cut ...]
INFO  | 05:07:03 395 [GPS] Setting GPS power=0
DEBUG | 05:07:03 395 [GPS] publishing pos@65af4977:2, hasVal=1, Sats=7, GPSlock=1
DEBUG | 05:07:03 395 [GPS] onGPSChanged() pos@65af4977, time=1705986423, lat=613091023, lon=-1495293731, alt=178
INFO  | 05:07:03 395 [GPS] updatePosition LOCAL pos@65af4977, time=1705986423, latI=613091023, lonI=-1495293731, alt=178
DEBUG | 05:07:03 395 [GPS] Setting local position: latitude=613091023, longitude=-1495293731, time=1705986423
DEBUG | 05:07:03 395 [GPS] Node status update: 6 online, 9 total

BobsBlueNorth, 2024-01-23:
Here for comparison is an older T-Beam 1.1 with a NEO-6M. This unit has no problems finding a signal. You will also see it calculates an updated position every 3 minutes.  This is even with the **"BOGUS course.value"**

//\ E S H T /\ S T / C
INFO  | ??:??:?? 0 Meshtastic hwvendor=4, swver=2.2.17.dbac2b1

INFO  | ??:??:?? 1 Setting GPS power=1
DEBUG | ??:??:?? 1 WANT GPS=1
INFO  | ??:??:?? 1 Setting GPS power=1
DEBUG | ??:??:?? 1 Using GPIO34 for GPS RX
DEBUG | ??:??:?? 1 Using GPIO12 for GPS TX

DEBUG | ??:??:?? 1 [GPS] Probing for GPS at 9600 
INFO  | ??:??:?? 2 [GPS] Found a UBlox Module using baudrate 9600
DEBUG | ??:??:?? 2 [GPS] Module Info : 
DEBUG | ??:??:?? 2 [GPS] Soft version: 7.03 (45969)
DEBUG | ??:??:?? 2 [GPS] Hard version: 00040007
DEBUG | ??:??:?? 2 [GPS] Extensions:0
WARN  | ??:??:?? 3 [GPS] Unable to save GNSS module configuration.
INFO  | ??:??:?? 3 [RangeTestModule] Range Test Module - Disabled
DEBUG | ??:??:?? 4 [GPS] NMEA GPS time 2024-01-23 05:13:23
DEBUG | ??:??:?? 4 [GPS] Upgrading time to quality 4
DEBUG | 05:13:23 4 [GPS] Read RTC time as 1705986803
DEBUG | 05:13:23 4 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 05:13:23 4 [GPS] No GPS lock
DEBUG | 05:13:23 4 [GPS] onGPSChanged() pos@0, time=1705986803, lat=0, lon=0, alt=0
INFO  | 05:13:23 4 [GPS] updatePosition LOCAL pos@0, time=1705986803, latI=0, lonI=0, alt=0
DEBUG | 05:13:23 4 [GPS] Setting local position: latitude=0, longitude=0, time=1705986803
DEBUG | 05:13:23 4 [GPS] Node status update: 1 online, 5 total

WARN  | 05:14:34 75 [GPS] BOGUS course.value() REJECTED: -250057272
DEBUG | 05:14:34 75 [GPS] hasValidLocation RISING EDGE
DEBUG | 05:14:34 75 [GPS] WANT GPS=0
DEBUG | 05:14:34 75 [GPS] GPS Lock took 74, average 0
INFO  | 05:14:34 75 [GPS] Setting GPS power=0
DEBUG | 05:14:34 75 [GPS] publishing pos@65af4b3a:2, hasVal=1, Sats=5, GPSlock=1
DEBUG | 05:14:34 75 [GPS] New GPS pos@65af4b3a:3 lat=61.309114, lon=-149.529366, alt=184, pdop=2.36, track=0.00, speed=0.00, sats=5
DEBUG | 05:14:34 75 [GPS] onGPSChanged() pos@65af4b3a, time=1705986874, lat=613091141, lon=-1495293663, alt=184
INFO  | 05:14:34 75 [GPS] updatePosition LOCAL pos@65af4b3a, time=1705986874, latI=613091141, lonI=-1495293663, alt=184
DEBUG | 05:14:34 75 [GPS] Setting local position: latitude=613091141, longitude=-1495293663, time=1705986874
DEBUG | 05:14:34 75 [GPS] Node status update: 5 online, 6 total
INFO  | 05:14:34 75 [GPS] Sending smart pos@65af4b3a:6 to mesh (distanceTraveled=12715396.000000m, minDistanceThreshold=100m, timeElapsed=75335ms)
INFO  | 05:14:34 75 [GPS] Providing time to mesh 1705986874
INFO  | 05:14:34 75 [GPS] Position reply: time=1705986874, latI=613091141, lonI=--1495293663
DEBUG | 05:14:34 75 [GPS] Update DB node 0x84889654, rx_time=1705986874, channel=0
DEBUG | 05:14:34 75 [GPS] handleReceived(LOCAL) (id=0x301dd338 fr=0x54 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 WANTRESP rxtime=1705986874 priority=10)
DEBUG | 05:14:34 75 [GPS] No modules interested in portnum=3, src=LOCAL
DEBUG | 05:14:34 75 [GPS] localSend to channel 0
DEBUG | 05:14:34 75 [GPS] Add packet record (id=0x301dd338 fr=0x54 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 WANTRESP rxtime=1705986874 priority=10)
DEBUG | 05:14:34 75 [GPS] Using AES256 key!
DEBUG | 05:14:34 75 [GPS] ESP32 crypt fr=84889654, num=301dd338, numBytes=30!
DEBUG | 05:14:34 75 [GPS] enqueuing for send (id=0x301dd338 fr=0x54 to=0xff, WantAck=0, HopLim=3 Ch=0xa4 encrypted rxtime=1705986874 priority=10)
DEBUG | 05:14:34 75 [GPS] txGood=3,rxGood=15,rxBad=0
DEBUG | 05:14:34 75 [GPS] Using channel 0 (hash 0xa4)
DEBUG | 05:14:34 75 [GPS] Using AES256 key!
DEBUG | 05:14:34 75 [GPS] ESP32 crypt fr=84889654, num=301dd338, numBytes=30!
DEBUG | 05:14:34 75 [GPS] decoded message (id=0x301dd338 fr=0x54 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 WANTRESP rxtime=1705986874 priority=10)

DEBUG | 05:17:34 255 [GPS] WANT GPS=1
INFO  | 05:17:34 255 [GPS] Setting GPS power=1
WARN  | 05:17:34 255 [GPS] SOME data is TOO OLD: LOC 180249, TIME 180249, DATE 180454
WARN  | 05:17:34 255 [GPS] SOME data is TOO OLD: LOC 180465, TIME 180465, DATE 180670
WARN  | 05:17:37 258 [GPS] BOGUS course.value() REJECTED: -250057272
DEBUG | 05:17:37 258 [GPS] WANT GPS=0
DEBUG | 05:17:37 258 [GPS] GPS Lock took 2, average 2
INFO  | 05:17:37 258 [GPS] Setting GPS power=0
DEBUG | 05:17:37 258 [GPS] publishing pos@65af4bf1:2, hasVal=1, Sats=4, GPSlock=1
DEBUG | 05:17:37 258 [GPS] New GPS pos@65af4bf1:3 lat=61.309219, lon=-149.529531, alt=178, pdop=3.94, track=0.00, speed=0.02, sats=4
DEBUG | 05:17:37 258 [GPS] onGPSChanged() pos@65af4bf1, time=1705987057, lat=613092188, lon=-1495295313, alt=178
INFO  | 05:17:37 258 [GPS] updatePosition LOCAL pos@65af4bf1, time=1705987057, latI=613092188, lonI=-1495295313, alt=178
DEBUG | 05:17:37 258 [GPS] Setting local position: latitude=613092188, longitude=-1495295313, time=1705987057
DEBUG | 05:17:37 258 [GPS] Node status update: 6 online, 7 total

DEBUG | 05:20:37 438 [GPS] WANT GPS=1
INFO  | 05:20:37 438 [GPS] Setting GPS power=1
WARN  | 05:20:37 438 [GPS] SOME data is TOO OLD: LOC 180287, TIME 180287, DATE 180292
WARN  | 05:20:37 438 [GPS] SOME data is TOO OLD: LOC 180503, TIME 180503, DATE 180508
DEBUG | 05:20:40 441 [Power] Battery: usbPower=1, isCharging=0, batMv=4131, batPct=98
DEBUG | 05:20:41 442 [GPS] WANT GPS=0
DEBUG | 05:20:41 442 [GPS] GPS Lock took 3, average 3
INFO  | 05:20:41 442 [GPS] Setting GPS power=0
[... cut ...]
DEBUG | 05:20:41 442 [GPS] New GPS pos@65af4ca9:3 lat=61.309082, lon=-149.529347, alt=187, pdop=2.46, track=283.94, speed=0.03, sats=5
DEBUG | 05:20:41 442 [GPS] onGPSChanged() pos@65af4ca9, time=1705987241, lat=613090825, lon=-1495293468, alt=187
INFO  | 05:20:41 442 [GPS] updatePosition LOCAL pos@65af4ca9, time=1705987241, latI=613090825, lonI=-1495293468, alt=187
DEBUG | 05:20:41 442 [GPS] Setting local position: latitude=613090825, longitude=-1495293468, time=1705987241
DEBUG | 05:20:41 442 [GPS] Node status update: 7 online, 8 total

geeksville, 2024-01-23:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/t-beam-v1-2-persists-saying-no-sats/9114/8

jp-bennett, 2024-01-23:
@BobsBlueNorth working on it. Fixes are landing, and @GPSFan in particular has some work that may help.

BobsBlueNorth, 2024-01-23:
I have been going nuts thinking I am the only one...until I saw the Meshtastic thread at about the same time geeksville posted it.

GPSFan, 2024-01-23:
I've been severely sidetracked by local events and trying to get a PR out to support the Chatter 2. Meshtastic is a fantastic project, but moves very rapidly and it's hard to keep up with all the changes on GitHub.
There are 3 basic issues with GPSs in general, module detection/identification, managing the power, and initialization. Of all the modules supported, by far u-blox has the most generations. Neo-6, Neo-7, Max-7, M8N, M10 and others. Different generations have improved capabilities, performance and power consumption. While documentation is available, it is not the easiest to understand and can contain errors/omissions/conflicts. Same goes for the boards on which the modules are mounted on.
Any help is appreciated.

BobsBlueNorth, 2024-01-31:
I just uploaded 2.2.20. Initial indications are the bug is fixed. So far I saw the GNSS come online and report a position. 2 minutes later the GNSS was woken up and I saw it found a new position and wasn't just re-reporting the old.

GPSFan, 2024-01-31:
The fixes for the M8 series are not complete yet, but those that got into 2.2.20 will certainly help.
As always, a good view of the sky is important especially with the tiny antennas that come with the t-beams.

BobsBlueNorth, 2024-02-01:
It definitely is working. Both my malfunctioning T-Beams work as they should. For the past few hours they are updating their positions quickly and flawlessly as I drive around town.

closed 2024-02-01 by thebentern