meshtastic/firmware #3100: [Bug]: Lilygo LORA V2 1.6 does not save region or name

lollo03, 2024-01-15:
### Category

Other

### Hardware

T-Lora v2 1.6

### Firmware Version

2.2.17

### Description

The device does not save settings regarding region or name. I tried with CLI, Android, and iOS apps.

https://github.com/meshtastic/firmware/assets/23106263/14bb33ac-71e5-4dff-bc75-81f545526a4a

Firmware version 2.2.16 does not have the same bug

GUVWAF, 2024-01-17:
How did you flash it? I would advise to try flashing it again using the [CLI](https://meshtastic.org/docs/getting-started/flashing-firmware/esp32/cli-script).

thebentern, 2024-01-19:
Cannot repro. This is likely an issue where the device needs a full install and erase instead of an update/

closed 2024-01-19 by thebentern