meshtastic/firmware #3104: Circuitmess Chatter 2.0 & Mestastic. (Low Priority)

JasonBurchell76, 2024-01-17:
### Platform

ESP32

### Description

I’m a 47 yo leaning how to code. Am I allowed to attempt to make a work around so Meshtastic will work on Chatter 2 board? I noticed a closed issue that seemed to be asking the developers to do the work.  I’ve been finding these Chatter boards second hand every, though it didn’t seem like many were made. They are of good quality and seems a waste not to find some use for them. That being said, I think it’s a waste of time for true developers to work on soon to be obsolete tech.  I apologize for any breaches of protocol. I look forward to hearing back.  Any info, input or advise will be greatly appreciated. Thanks for your time and consideration. 

-Jason Burchell

jp-bennett, 2024-01-17:
> Am I allowed to attempt to make a work around so Meshtastic will work on Chatter 2 board?

Absolutely! That's what's so cool about an open source project: You don't need anybody's permission to work on it. That said, I think @GPSFan already has working code for one of the chatter devices. There's been enough demand, I suspect we'll get support merged into the main project before long.

GPSFan, 2024-01-17:
I already have the code working on the Chatter 2. both with the original Lora module and with  a Heltec RA-62 replacement.
See:
https://github.com/meshtastic/firmware/issues/2896
for additional details.
I'm using the Chatter 2 right now as a test platform to debug GPS issues.

jp-bennett, 2024-01-17:
I'll go ahead and close this issue as a dupe.

closed 2024-01-17 by jp-bennett

GPSFan, 2024-01-17:
@jp-bennett could someone review my PR to protobufs to add the ID for the Chatter 2?
Thanks in advance.