meshtastic/firmware #3109: [Bug]: Heltec Traker running V2.2.17 reports incorrect altitudes

kovar-ursa, 2024-01-18:
### Category

Hardware Compatibility, Other

### Hardware

Heltec Wireless Tracker

### Firmware Version

2.2.17.dbac2b1

### Description

Flashed Heltek Wireless Traker with v2.2.17 and configured it as a TRACKER. In some cases, the position appears as an extremely large value. On an identical device sitting right next to the first it reports the expected value.

U-T019 │ T019  │ !e2e5b820 │ 45.9692°   │ -72.9999°   │ 3 m    
U-T017 │ T017  │ !e2e5b84c │ 45.9692°   │ -72.9999°   │ 42949659 m
U-T016 │ T016  │ !e2e5b62c │ 45.9692°   │ -72.9999°   │ 5 m

(The .9999 is accurate, BTW)

Those are three identical devices running the same firmware and identical configuration files.

LilyGo T-Beams with the same configuration do not exhibit this problem.        

### Relevant log output

```Shell
mesh/2/json/xxxx/!e2e5b5b0 0 b'{"channel":0,"from":3806705740,"id":662610717,"payload":{"altitude":42949666,"latitude_i":429692293,"longitude_i":-709998908,"sats_in_view":19,"time":1705587881,"timestamp":1705587878},"rssi":-24,"sender":"!e2e5b5b0","snr":12,"timestamp":1705587881,"to":4294967295,"type":"position"}'
```

caveman99, 2024-01-18:
This is a unsigned/signed problem and does not have anything to do with the specific device. problem is either in TinyGPS++ or in our backend code. The device thinks it's 1400m below sea level.

kovar-ursa, 2024-01-18:
1) Is there any way for us to correct this downstream?
2) I don't think the device thinks it is 1,400m below sea level. It fluctates between these incorrect values and legitimate values. I don't think the GPS data is fluctuating between 0-10m and -1,400m.
3) 
32 bit signed int. 
2^31-1 values for positive numbers and 2^31 values for negative numbers and 1 value for zero.
4294967295 is -1.
42949659, the value I am seeing, is positive, correct? 

I'm not sure how you are getting -1400m out of 42949659. Unless you multiply it by 100 in which case you get -1396

jp-bennett, 2024-01-18:
I bet the GPS chip is actually sending an *empty* altitude value. TinyGPS doesn't have a code path for this, and ends up dereferencing an unset pointer. Working on a fix for a related issue, that may fix this too.

kovar-ursa, 2024-01-18:
Huh. Why would it sometimes send a populated value and sometimes an empty value? And why would that empty value manifest with various false values?

> 42949670 m 
> 42949672 m
> 42949668 m

Three devices all sitting near each other.

jp-bennett, 2024-01-18:
Given those values, @caveman99 is probably right. And the raw value is in 1/100 of a meter, so that checks out. Any chance you're 14 meters below sea level?

kovar-ursa, 2024-01-18:
I'm very close to sea level. My phone shows me at 7m above MSL. These devices started at between 1m and 5m above MSL. It'd take 14m of fluctuation but I suppose it is possible, particularly with what we're seeing.

dchauran, 2024-03-04:
@kovar-ursa I was having the same issue, and after updating to 2.2.24 alpha (which incorporates some position fixes) my device is reporting more reasonable altitudes (within 100ft at any rate). You might want to check that version.

dchauran, 2024-03-04:
Nevermind, mine went back to reporting 140874865.12ft lol. Impressively high.

BjornB2, 2024-03-07:
> @kovar-ursa I was having the same issue, and after updating to 2.2.24 alpha (which incorporates some position fixes) my device is reporting more reasonable altitudes (within 100ft at any rate). You might want to check that version.

I am having this issue on .24 as well.

DooMMasteR, 2024-03-11:
yepp, it's still persistent

BjornB2, 2024-03-15:
[image]

New T-Echo alternating between 1m (correct) and bogus 42950m. 2.30. I see this on all my nodes.

BjornB2, 2024-03-25:
The GPS alt value is stored in a int32_t, which has a max value of 4,294,967,295.
GPS coordinates are usually in a 10^5 format to avoid a decimal which makes the error of 42950  4,295,000,000 with the 5 zeros added. 42950 would be the rounded value of 42949.67295 so I guess we can conclude that this is an overflow error. 

Several users earlier in this thread seem to be around sea level, so there might be a relation between altitude 0 or near zero and this overflow problem. Maybe a leading zero's problem?

For me it's above my skill level to actually debug the code in this case. Could be a TinyGPS++ problem also of course. 

By the way, in my screenshot above it alternates between correct alt and 42950. I think the correct values might come from my phone since GPS sharing was enabled.  Not sure though.

I hope someone picks up this error 😊

thebentern, 2024-04-12:
Closed by #3605

closed 2024-04-12 by thebentern