meshtastic/firmware #3110: [Bug]: Can only set MSL as the desired altitude, cannot set HAE

kovar-ursa, 2024-01-18:
### Category

Other

### Hardware

T-Beam S3, Heltec Wireless Tracker

### Firmware Version

All.

### Description

Two related issues:

1) You need to set both ALTITUDE and ALTITUDE_MSL in the Position bitmask to get the device to send altitude data.
2) You can only set MSL. There is no bit to set for HAE. 

`
    if (pos_flags & meshtastic_Config_PositionConfig_PositionFlags_ALTITUDE) {
        if (pos_flags & meshtastic_Config_PositionConfig_PositionFlags_ALTITUDE_MSL)
            p.altitude = localPosition.altitude;
        else
            p.altitude_hae = localPosition.altitude_hae;
        if (pos_flags & meshtastic_Config_PositionConfig_PositionFlags_GEOIDAL_SEPARATION)
            p.altitude_geoidal_separation = localPosition.altitude_geoidal_separation;
    }
`
    
 I suspect the intention was to add a bit for HAE but it never got done.

garthvh, 2024-01-18:
It is one or the other, there is a single altitude value on the packet and the value changes based on your position flag.  It is possible some devices do not support HAE

kovar-ursa, 2024-01-18:
If I set just ALTITUDE no altitude data is transmitted. I must set both ALTITUDE and ALTITUDE_MSL for any altitude data to be sent. This seems to force me to use MSL for altitude values.

The code is setting either `p_altitude` or `p.altitude_hae` depending on ALTITUDE_MSL. Maybe p.altitude is being sent out even if p.altitude_hae is the one that is set? (I'm not sure where in the code this is handled.)

garthvh, 2024-01-18:
if you have MSL checked (which is the default) it outputs MSL for altitude, otherwise HAE

if (pos_flags & meshtastic_Config_PositionConfig_PositionFlags_ALTITUDE_MSL) p.altitude = localPosition.altitude; else p.altitude_hae = localPosition.altitude_hae;

kovar-ursa, 2024-01-18:
With bitmask 163 (ALTITUDE, ALTITUDE_MSL, SATINVIEW, TIMESTAMP) 
> !e2e5b84c │ 42.9693°   │ -70.9999°   │ 42949665 m. (ignore the bad value for altitude)

With bitmask 161 (ALTITUDE, SATINVIEW, TIMESTAMP):
> !e2e5b84c │ 42.9693°   │ -70.9999°   │ N/A   

If I turn off ALTITUDE_MSL but still have ALTITUDE set, no altitude data is transmitted.

I just did the above and the only change to the config file was 163 -> 161. Same firmware, hardware, etc.

garthvh, 2024-01-19:
@jp-bennett have you seen a HAE altitude value ever?  The code looks to be adding it to the packet and sending it out if the position flag is set but there never seems to be a value.

jp-bennett, 2024-01-19:
> have you seen a HAE altitude value ever?

I'm not even entirely sure what an HAE altitude is. I'll look into it when I get back to my work on checking for empty fields.

garthvh, 2024-01-19:
co

> > have you seen a HAE altitude value ever?
> 
> I'm not even entirely sure what an HAE altitude is. I'll look into it when I get back to my work on checking for empty fields.

Altitude value for aircraft

kovar-ursa, 2024-01-19:
HAE is height above ellipsoid. I believe this is the "normal" GPS altitude value, i.e., what all GPS receivers provide.

Some modern GPS receivers have added the ability to calculate height above mean sea level, MSL. This is traditionally done with barometric pressure sensors.

So, I believe, HAE data is what you get from all GPS receivers and MSL data only is produced by GPS receivers with the ability to calculate height above MSL.

kovar-ursa, 2024-01-19:
https://avionicswest.com/Articles/GPSalt.html

closed 2024-11-08 by github-actions[bot]