meshtastic/firmware #3135: [Feature Request]: Add Hong Kong to region list.

moresecode, 2024-01-27:
### Platform

NRF52

### Description

Please add Hong Kong to the region list.
The radio frequency band of the LoRa infrastructure used is 923 MHz in Hong Kong (the unlicensed band also called AS923)

code8buster, 2024-01-30:
According to this document which is the guideline for license-exempt uses of the 920-925mhz band in HK, the device must use frequency hopping spread spectrum, which Meshtastic does not do.
https://www.elegislation.gov.hk/hk/cap106Z!en-zh-Hant-HK

moresecode, 2024-01-30:
Thank you so much for your answer, Code8buster.
I have 2 more questions hope you can answer, thanks in advance.
1- By "Meshtastic doesn't do FHSS", is it fundamentally impossible for Meshtastic, so I can just give the HK region all at once?
2- I can use my WisBlock RAK4631(AS923) in the US by setting to the US region?

andrekir, 2024-01-30:
good catch @code8buster. even if all requirements were met, Hong Kong wouldn't require a new region as the existing `TW` region could be used.

---

(...)

4. TECHNICAL REQUIREMENTS _(HKCA1078 ISSUE 1 PAGE 1 DECEMBER 2017)_
4.1 The equipment shall operate in the frequency band 920 – 925 MHz and meet the technical requirement according to the Code of Federal Regulations (USA); Title 47 Telecommunication; Chapter 1 Federal Communications Commission, Part 15 Radio Frequency Devices; Section 15.247.
4.2 The equipment shall use frequency hopping spread spectrum modulation.
4.3 The maximum allowed 20 dB bandwidth of the hopping channel is 500 kHz.
4.4 The peak transmitter power shall not exceed 1W and the equivalent isotropically radiated power (EIRP) from the equipment shall not exceed 4W.

(...)

---

[hkca1078.pdf](https://github.com/meshtastic/firmware/files/14072260/hkca1078.pdf) (source: https://www.ofca.gov.hk/filemanager/ofca/en/content_401/hkca1078.pdf)

closed 2024-01-30 by andrekir