meshtastic/firmware #3138: [Bug]: MQTT uplink message does not have correct type / payload / source ID

milkpirate, 2024-01-28:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

v2.2.17.dbac2b1

### Description

My minimal setup looks like so:

```mermaid
graph LR
    PC <-->|Serial| NodeA
    <--> |LORA| NodeB
    <--> |Bluetooth| Android

    NodeA <--> |MQTT| mosquitto
```

the `mosquitto` log looks like so (converted to YAML for better readability):

```yaml
---
# lora/2/json/home-prim/!NodeA
channel: 0
from: 145556388
id: 1057405219
payload:
  air_util_tx: 0.125111117959023
  battery_level: 101
  channel_utilization: 0
  voltage: 4.35400009155273
sender: '!NodeA'
timestamp: 1706448997
to: 4294967295
type: telemetry
```
Messages sent from `Android` to `NodeB` in the prim. channel, look like so:
```yaml
---
# lora/2/json/home-secd/!NodeA
channel: 1
from: 3806671628
id: 1421562879
payload:
  text: test-message
rssi: -60
sender: '!NodeA'
snr: 13.5
timestamp: 1706449977
to: 4294967295
type: text
```
**1. Problem**: The message sent from `NodeB` does not appear as beeing sent from `NodeB` instead from `NodeA`.

When I send a message in the "private" channel from `NodeB` to `NodeA`, the message appears as:
```yaml
---
# lora/2/json/home-prim/!NodeA
channel: 0
from: 145556388
id: 1057405208
sender: '!NodeA'
timestamp: 1706448349
to: 3806671628
type: ""
```
**2. Problem**:
- The message type is empty
- There is no text at all
- And the source (sender) is not `NodeB` 

Is that as intended? Since to me it does not seem to be desirable.

GUVWAF, 2024-01-28:
The "sender" field is always the node connected to MQTT (gateway). You'll need to look at the "from" field, which is the unique node number, which you can map to a name received in NodeInfo messages, see also: https://meshtastic.org/docs/software/mqtt/#json-topic

For 2: do you have encryption enabled? I believe if you disable that you should see the type and text.

milkpirate, 2024-01-28:
@GUVWAF do you mean encryption at all or the MQTT encryption:
[image]
since this one is off.

GUVWAF, 2024-01-28:
Yes, I meant MQTT encryption. OK, yes indeed it doesn't actually work like this, it's a bit confusing. It only uplinks messages to MQTT that it is going to send out via LoRa. For messages on a channel, it will rebroadcast it for other nodes to receive. However, for a direct message, it only sends back an acknowledgment. This is probably the packet with empty type you're seeing (because the ACK has no JSON output yet).

Yes, enabling downlink indeed also means it rebroadcasts the packets over LoRa. Currently you can't configure that separately (except when completely disabling LoRa Tx). What you _can_ do with firmware 2.2.19 is letting other LoRa nodes ignore messages that came via MQTT.

You made good points and I think it's worth to revisit the MQTT-related settings.

milkpirate, 2024-01-28:
Great, I guess it'll be good to be able to set a list of topics to subscribe to / forward.

GUVWAF, 2024-02-03:
Hope you don't mind that I'm renaming and re-purposing it as a feature request (since the `sender`/`from` fields are correct) for more granular MQTT settings and uplinking DMs to the MQTT node.

GUVWAF, 2024-02-21:
Uplinking DMs towards the MQTT node is implemented with #3245.

closed 2024-05-05 by thebentern