meshtastic/firmware #3139: [Feature Request]: Startup message

milkpirate, 2024-01-28:
### Platform

ESP32

### Description

I would like to send a startup message as soon as the device comes online. Is that possible?

Background: I am using the Heltec V3. I want to put it in my mailbox. When the lid is opened the wakeup button is triggered/shorted and the device wakes up. Then I want to send a message to the mesh.

thebentern, 2024-01-28:
I think this is a good use case for building some functionality around the detection sensor module using a combo of the sensor device role w/ is_power_saving of true. We have precedence for this type of power saving behavior in the sensor role w/ environment telemetry and the tracker role w/ position. Namely the device wakes up, gathers information it needs, sends to the mesh, and goes to sleep on an interval. We could perhaps do something similar but with a GPIO interrupt instead of a time based interrupt.

milkpirate, 2024-01-28:
That would be really nice too, but I guess my use case is even simpler... since the device could be woken up, upon the interesting events.

So with the current setup I could set it to sensor, also enable telemetry for e.g. position?
Although it does not have a GPS module, it would not report anything meaningful but for me any message would suffice as long as I can relate it to the corresponding device