meshtastic/firmware #3149: [Board]: CanaryOne

hdngr, 2024-01-30:
### SOC

NRF52

### Lora IC

SX1262

### Product Link

https://canaryradio.io/products/canary-lora

### Description

CanaryOne is meant to be a complete out of the box solution for existing Meshtastic users, as well as individuals and organizations new to Meshtastic.  First 100 radios will be ready to ship February 2nd, 2023.

Our team plans to contribute to Mesthastic as CanaryOne is adopted.

Thank you for everything that you do for the open source community, and those of us that rely on off grid communications.  Developing inexpensive hardware would not be possible without the significant Open Source community contributions.

Feel free to reach out for any questions: team@canaryradio.io.

thebentern, 2024-01-30:
Very cool! Is this device based off of existing hardware variants or is it a completely custom board?
Also, would you be willing to send a couple of these to our development team for testing?

hdngr, 2024-01-30:
> Very cool! Is this device based off of existing hardware variants or is it a completely custom board? Also, would you be willing to send a couple of these to our development team for testing?

@thebentern this is a completely custom board.  Of course inspired by other boards out there and the great work that so many folks have done already...  One of the things we are exploring is how to pull folks into the meshtastic community with something that works completely out of the box...  So - hackable to the core, but for non technical users it just works out of the box.

Of course.  We have some more field testing to do ourselves.  What is the best way to get you guys radios?  team@canaryradio.io

thebentern, 2024-01-31:
> @thebentern this is a completely custom board. Of course inspired by other boards out there and the great work that so many folks have done already... One of the things we are exploring is how to pull folks into the meshtastic community with something that works completely out of the box... So - hackable to the core, but for non technical users it just works out of the box.
> 
> Of course. We have some more field testing to do ourselves. What is the best way to get you guys radios? [team@canaryradio.io](mailto:team@canaryradio.io)

Sounds good. I'll reach out via email.

thebentern, 2024-02-17:
Closing, as this has been integrated.

closed 2024-02-17 by thebentern