meshtastic/firmware #3158: [Board]: rak4631-no-buttons

guthip, 2024-02-01:
### SOC

NRF52

### Lora IC

RAK4631

### Product Link

https://store.rakwireless.com/products/wisblock-kit-2-lora-based-gps-tracker-with-solar-panel

### Description

Standard rak4631 environment, except that there are no buttons other than reset. If one uses the 
    default_envs=rak4631 
the board will reset as it interprets the non-existing pin_button as a long press.
This 
    default_envs=rak4631_no_buttons 
removes the button definitions, and runs as intended.

guthip, 2024-02-01:
Alternatively, make the lack of user buttons a setting in the app

garthvh, 2024-02-01:
A rak4631 already come with no button, use the regular uf2 to glash

closed 2024-02-01 by garthvh

guthip, 2024-02-01:
> A rak4631 already comes with no button, use the regular uf2 to glash

Thanks, Garth, for the quick response.

Totally possible I’m wrong. At the same time, the standard rak4631 image causes a continuous reset cycle, with the message of detection of a long press power down request. Not defining the button_pins helps solve that issue.

Let me know if any log files would help. 

Regards, Hans

garthvh, 2024-02-01:
You might try erasing the flash, this is a standard base board and core, I have flashed hundreds of these without button issues.

guthip, 2024-02-01:
# You might try erasing the flash
Done that quite a few times. Could it be a hardware issue, i.e. depending on which slot the GPS and sensor model is in?