meshtastic/firmware #3170: [Bug]: Heltec V3 won’t connect to access point on channel 1

gblades, 2024-02-04:
### Category

WiFi

### Hardware

Heltec V3

### Firmware Version

2.2.16

### Description

I have a ubiquity setup with multiple access points and I was experiencing poor WiFi signal strength due primarily to the foil backed loft insulation.
I installed a new AP in the loft a couple of meters away from the Meshtastic device but it would not automatically switch to the stronger signal. I set it to be locked to the loft AP and rebooted but it then failed to connect.
After a lot of troubleshooting I reconfigured the loft AP to force it to use UK channel 11 which the working APs were using at which point the Heltec connected and started working. Other devices were able to connect to the loft AP prior to this change without any issues.
I suspect there is a setting for the WiFi region which might be set incorrectly or require a config option so that it supports all the WiFi channels in every country.

GUVWAF, 2024-02-05:
Was the new AP locked to channel 1 or did it happen to use that one? ESP32 only supports 2.4GHz Wi-Fi, so if your AP was on a hybrid 2.4/5GHz mode, it might not work.
Wi-Fi channels are the same almost over the whole world (https://en.wikipedia.org/wiki/List_of_WLAN_channels). Channel 1 or 11 shouldn't make a technical difference, but if channel 1 is too congested in your place, that might also be a problem.

gblades, 2024-02-05:
Loft AP was in hybrid mode but so was the upstairs AP which is has been connecting to fine for a month.
Loft AP had automatically set itself to channel 1 and the upstairs AP to channel 11 as the ubiquiti kit automatically does a nightly channel scan to optimise which channel they use.
So it can connect to the AP when set to channel 11 but if it is set to channel 1 it will not connect.
My ipad and pixel phone did work on the loft AP when set to channel 1 so it wasn't an issue with the AP.
Channel 1 is not congested.

GUVWAF, 2024-02-05:
Did you try locking it to channel 1 instead of 11? 
If that also doesn’t work, I would recommend to open an issue on the Arduino ESP32 core (https://github.com/espressif/arduino-esp32), because it is their WiFi library Meshtastic uses for the Heltec V3, and we don't specify which channels it may or may not use. Nor do we set the country, but I don't find anywhere stating that is required.

closed 2024-02-05 by garthvh