meshtastic/firmware #3171: [Feature Request]: Current Time/Clock Functionality

Cantir2068, 2024-02-05:
### Platform

ESP32

### Description

While this request is mainly in response t othe Lilygo T-Watch 2020 line of products, I feel there would also be value for anything else that has a RTC component. Short of it is, 'This is a smart watch that once Meshtastic is flashed onto the firmware can't act as a watch.'

gala-vine, 2024-03-07:
I agree with this.. the T watch s3 has its own firmware that can been installed updated via Arduino, can't we just add Meshtastic to its library and boot it as a separate app on the watch?.

djkazic, 2024-03-10:
There's actually a clock already on the page with ChUtil and battery! Swipe left from the default page. It can be hard to spot, but I made some modifications in Screen.cpp to increase the font size.

gala-vine, 2024-03-11:
> There's actually a clock already on the page with ChUtil and battery! Swipe left from the default page. It can be hard to spot, but I made some modifications in Screen.cpp to increase the font size.

yes i know the GPS clock, but only sets time once its got a GPS lock else the time is incorrect, yeah I had thought about editing the  .cpp to make it a bigger font size myself.. I think we need to play around with the settings some more to see if we can optimise battery life as well.

djkazic, 2024-03-11:
My watch sets it's time without GPS. I know this because the t-watch does not have GPS and yet the time is set. The Android app at least auto configures it.

gala-vine, 2024-03-11:
Yes that's what i meant, it gets it from the App/gps,

djkazic, 2024-03-11:
You also only need one device with time, it should send the time across the mesh to the other devices and they'll sync up

fifieldt, 2024-09-16:
The new GUI https://github.com/meshtastic/device-ui could address this

thebentern, 2024-10-07:
This exists now on T-Watch S3

closed 2024-10-07 by thebentern