meshtastic/firmware #3172: [Bug]: Heltec Tracker can't turn off GPS location.

Edgs, 2024-02-05:
### Category

Hardware Compatibility

### Hardware

Heltec Wireless Tracker

### Firmware Version

2.2.19

### Description

Tracker transmits live GPS location, even though it appears disabled in settings.
'Smart Position' = disabled
'Use fixed position' = disabled
'GPS enabled' = disabled
All the update and broadcast intervals are at their default (non zero) values.

Gabrielerusso, 2024-02-05:
i don't remember having this problem, both disabling it via settings and via triple button push.

Edgs, 2024-02-06:
It does appear to have a 'last known location' stored in the fixed position values, but it continues to transmit this as a current location, even though 'Use fixed position' is disabled.
I would expect it to transmit no location with the above settings.

Gabrielerusso, 2024-02-06:
> It does appear to have a 'last known location' stored in the fixed position values, but it continues to transmit this as a current location, even though 'Use fixed position' is disabled. I would expect it to transmit no location with the above settings.

"Use fixed position" it's for the fixed position value you set, that you see it's the last known one, this is very usefull for a tracker as not every time the gps fix is available. This is the same for all meshtastic devices, if you want to reset every past GPS data you can simply press the reset button and this will erase the last known GPS location.

It's possible to edit the firmware to reset the location data when the GPS is disabled, but i don't know if this would be an issue for some users that activate the GPS one time to get a position and then deactivate it.

Example: i'm using a Heltec Tracker to track my dog, the dog goes inside an house where the location data isn't available, the tracker keeps giving the last known position (outside the house) .

Edgs, 2024-02-06:
I agree with all the posibilities you list there, except:
> ...if you want to reset every past GPS data you can simply press the reset button and this will erase the last known GPS location.
This does not happen - the 'last location' is always there, and always transmitted, even though both Location options are disabled.

The user would want the last location to transmit if GPS was enabled, but unable to fix it's location.
The user would want the 'fixed location' to be transmitted if 'Transmit fixed location' was enabled.
But if GPS is disabled, and 'Transmit fixed location' is disabled, then it should not transmit any location.

I'm setting these options using the Android app. incidentally.
The location in 'Fixed location' came from a GPS position on initial power on after installation - I didn't enter anything.
If I do try to set the Location to 0.0, 0.0, after reboot the original Location is still set, and still being transmitted.

Gabrielerusso, 2024-02-06:
I just tried it, it does happen on .22 and from what i remember it does happen on .19 too. Are you sure you are not setting the location via the App ?

Edgs, 2024-02-06:
Maybe things have changed in .22 - I might try that later.
For reference, I also have a Heltec V3, which does have a fixed position enabled.  This position is never cleared, even through a firmware upgrade, and I just now turned off 'Use fixed position' in that, and after restarting I can still request and receive it's same fixed position.
(I cleared the nodedb on the requesting node, to be sure it didn't have an old position stored.)

I'm not using any location on the Android device, and the fixed locations are at known locations away from the Android device.

Gabrielerusso, 2024-02-06:
> Maybe things have changed in .22 - I might try that later. For reference, I also have a Heltec V3, which does have a fixed position enabled. This position is never cleared, even through a firmware [...]

Fixed position, like all the other settings, is stored in flash memory.

Edgs, 2024-02-06:
Just now, on my Tracker, I enabled GPS, and allowed a few position updates.  I then disabled GPS.
Following the reset, the last known position seen then appeared in the 'Fixed Location', changing the values from before the GPS was enabled.  I didn't manually enter any values in there.
Once again, with GPS _disabled_, and 'Use fixed position' _disabled_, the node transmits that 'last known position' which it stored in 'Fixed location' as a current position.

Gabrielerusso, 2024-02-06:
> Just now, on my Tracker, I enabled GPS, and allowed a few position updates. I then disabled GPS. Following the reset, the last known position seen then appeared in the 'Fixed Location', changing the [...]

Just tested, what you say is present in .19, doesn't apper to be a problem in .22 .

Gabrielerusso, 2024-02-09:
i found some combination that could still lead to a position leak, i resolved that in #3163 with commit [3cce370](https://github.com/meshtastic/firmware/pull/3163/commits/3cce370faf5f7f32c4150eed8b1a79f5cf0a8635) . Now when you disable the location every past position data is deleted if "fixed position" is disabled.

thebentern, 2024-03-13:
Setting position broadcast secs to max int or better yet, using the new position precision feature is the solution to this issue. 
#3263

closed 2024-03-13 by thebentern